calimero-project/calimero-server #1: Group monitor works but Bus monitor doesn't with USB interface

alesf76, 2015-11-04:
My setup is Windows 8.1 32 bit, calimero-server from the Java 8 branch, a KNX USB interface.
Connecting from ETS3 and activating the group monitor works, but if I activate the bus monitor I get this exception

[main] INFO calimero.server - Discovery service network interfaces:
[main] INFO calimero.server -     listen on all
[main] INFO calimero.server -     outgoing all
[main] INFO calimero.server - Service container 147b:5120: 
[main] INFO calimero.server -     IPv4 UDP host 192.168.1.133 port 3671 routing false
[main] INFO calimero.server -     usb connection, TP1 medium, device 1.0.248
[main] INFO calimero.server -     GrpAddrFilter []
[main] INFO calimero.server - connect to 147b:5120
[main] INFO calimero.usb - Found KNX devices:
 |--Bus 001 Device 003: ID 147b:5120
[main] INFO calimero.usb.147b:5120 - Bus 001 Device 003: ID 147b:5120
[main] INFO calimero.link.147b:5120 - Device Mask Version 0012 - System 1 (BCU 1)
[main] INFO calimero.link.147b:5120 - PEI type 1
[main] INFO calimero.link.147b:5120 - KNX individual address 1.0.253
[main] INFO calimero.link.147b:5120 - hop count set to 6
[main] INFO Calimero KNX server gateway - main-line group address forward setting set to 1
[main] INFO Calimero KNX server gateway - sub-line group address forward setting set to 1
[Calimero KNX server gateway] INFO knx-server - launch KNXnet/IP server My KNXnet/IP Server
[knx-server control endpoint 147b:5120] INFO knx-server - knx-server control endpoint 147b:5120 is up and running
[knx-server control endpoint 147b:5120] WARN knx-server - received disconnect request with unknown channel id 1 - ignored
[knx-server control endpoint 147b:5120] INFO knx-server - 147b:5120: setup data endpoint (channel 1) for connection request from /192.168.1.134:63229
[knx-server control endpoint 147b:5120] INFO knx-server - assigning additional individual address 1.0.249
[knx-server control endpoint 147b:5120] WARN calimero.usb.147b:5120 - close connection
javax.usb.UsbPlatformException: USB error 5: Unable to release interface: Entity not found
	at org.usb4java.javax.ExceptionUtils.createPlatformException(ExceptionUtils.java:39)
	at org.usb4java.javax.AbstractDevice.releaseInterface(AbstractDevice.java:455)
	at org.usb4java.javax.Interface.release(Interface.java:113)
	at tuwien.auto.calimero.serial.usb.UsbConnection.close(UsbConnection.java:647)
	at tuwien.auto.calimero.serial.usb.UsbConnection.close(UsbConnection.java:554)
	at tuwien.auto.calimero.link.AbstractLink.close(AbstractLink.java:293)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway$KNXnetIPServerListener.closeLink(KnxServerGateway.java:322)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway$KNXnetIPServerListener.acceptDataConnection(KnxServerGateway.java:189)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.lambda$acceptConnection$5(KNXnetIPServer.java:2411)
	at java.util.stream.MatchOps$1MatchSink.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.tryAdvance(Unknown Source)
	at java.util.stream.ReferencePipeline.forEachWithCancel(Unknown Source)
	at java.util.stream.AbstractPipeline.copyIntoWithCancel(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(Unknown Source)
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.allMatch(Unknown Source)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.acceptConnection(KNXnetIPServer.java:2411)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.initNewConnection(KNXnetIPServer.java:2389)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.handleServiceType(KNXnetIPServer.java:2142)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ServiceLoop.onReceive(KNXnetIPServer.java:1476)
	at tuwien.auto.calimero.internal.UdpSocketLooper.loop(UdpSocketLooper.java:134)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ServiceLoop.run(KNXnetIPServer.java:1447)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$LooperThread.run(KNXnetIPServer.java:1665)
[knx-server control endpoint 147b:5120] INFO calimero.link.147b:5120 - link closed
[knx-server control endpoint 147b:5120] INFO calimero.usb.147b:5120 - Bus 001 Device 003: ID 147b:5120
[knx-server control endpoint 147b:5120] ERROR calimero.link - initial connection attempt
tuwien.auto.calimero.KNXException: open USB connection
	at tuwien.auto.calimero.serial.usb.UsbConnection.<init>(UsbConnection.java:405)
	at tuwien.auto.calimero.serial.usb.UsbConnection.<init>(UsbConnection.java:375)
	at tuwien.auto.calimero.link.KNXNetworkMonitorUsb.<init>(KNXNetworkMonitorUsb.java:98)
	at tuwien.auto.calimero.server.gateway.SubnetConnector.lambda$openMonitorLink$16(SubnetConnector.java:283)
	at tuwien.auto.calimero.link.Connector$Link.connect(Connector.java:446)
	at tuwien.auto.calimero.link.Connector$Link.<init>(Connector.java:219)
	at tuwien.auto.calimero.link.Connector$Link.<init>(Connector.java:176)
	at tuwien.auto.calimero.link.Connector.newMonitor(Connector.java:172)
	at tuwien.auto.calimero.server.gateway.SubnetConnector.openMonitorLink(SubnetConnector.java:297)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway$KNXnetIPServerListener.acceptDataConnection(KnxServerGateway.java:190)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.lambda$acceptConnection$5(KNXnetIPServer.java:2411)
	at java.util.stream.MatchOps$1MatchSink.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.tryAdvance(Unknown Source)
	at java.util.stream.ReferencePipeline.forEachWithCancel(Unknown Source)
	at java.util.stream.AbstractPipeline.copyIntoWithCancel(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(Unknown Source)
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.allMatch(Unknown Source)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.acceptConnection(KNXnetIPServer.java:2411)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.initNewConnection(KNXnetIPServer.java:2389)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ControlEndpointService.handleServiceType(KNXnetIPServer.java:2142)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ServiceLoop.onReceive(KNXnetIPServer.java:1476)
	at tuwien.auto.calimero.internal.UdpSocketLooper.loop(UdpSocketLooper.java:134)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$ServiceLoop.run(KNXnetIPServer.java:1447)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer$LooperThread.run(KNXnetIPServer.java:1665)
Caused by: javax.usb.UsbClaimException: An interface is already claimed
	at org.usb4java.javax.AbstractDevice.claimInterface(AbstractDevice.java:404)
	at org.usb4java.javax.Interface.claim(Interface.java:102)
	at tuwien.auto.calimero.serial.usb.UsbConnection.open(UsbConnection.java:608)
	at tuwien.auto.calimero.serial.usb.UsbConnection.<init>(UsbConnection.java:389)
	... 27 more
[main] INFO Calimero KNX server gateway - knx-server: user request for shutdown

Also other functions doesn't work:
- trying to download a program make the server continuosly print
[Calimero KNX server gateway] WARN Calimero KNX server gateway - no active KNXnet/IP connection for destination 1.0.253
- trying to download an address the server never detects the pressed button
- diagnotics/individual addresses doesn't work, no error or warning but it doesn't detect nothing.

bmalinowsky, 2015-11-06:
There are several issues:

1) Busmonitor does not open, because usb4java thinks the interface is still claimed. This is a known windows-only issue (at least on w7 and w8); the returned libusb error code screws with the IF release. I actually thought that got fixed, but I'm not using windows that often.
-> Do this: while calimero-server is _running_ and open busmonitor failed, unplug and plug the usb cable. Then the device is detected again and the server can claim the IF. Then start the ETS busmonitor. Note that this happens on every release/claim cycle, e.g., also when switching from groupmon to busmon.

2) "no active KNXnet/IP connection for destination 1.0.253": problem with the address assignment: device 1.0.253 is the USB IF, your client is assigned 1.0.249. Incoming frames use destination 253. This is caused by the additional individual address assignement to connections, and obviously not handled correctly by the server.
-> Put 1.0.253 as the only additional individual address, so your client uses that address.

3) diagnostics etc: In general, some mgmt the ETS is capable of when directly communicating with an IF, calimero currently cannot do. Checking programming mode and reading device info etc. should work.
Anyway, I need a little more specifics: i.e., run the server with trace logging 
`-Dorg.slf4j.simpleLogger.defaultLogLevel=trace` to see if a device on the bus actually answers. E.g., an answer of a device in programming mode should show up as "T_Broadcast, A_IndAddr.response".

bmalinowsky, 2015-11-06:
For 1) I committed a workaround, https://github.com/calimero-project/calimero-core/commit/4bd9917201341f43a92cdaeb2bda191169a4d9e7.
Tested with usb4java-javax v1.2.0.

Wrt 2) commit 4aa47a9de832e262aea7d827c6779ae758497e84 should mitigate this problem of forwarding subnet messages to some extent. It's not a solution, though.

alesf76, 2015-11-09:
1) seems to work
2) busmonitor now works! also programming a simple device with an already assigned address, but checking programming mode and assigning an address still doesn't

This is the stack trace, 1.4.109 is the address assigned to the device in programming mode
[Gateway IP to subnet dispatcher] TRACE Calimero KNX server gateway - server-sid
e KNXnet/IP Tunneling 192.168.1.102:49428 channel 2: 1.0.253->0/0/0 L_Data.req,
system priority hop count 6 repeat tpdu 01 00
[Gateway IP to subnet dispatcher] TRACE Calimero KNX server gateway - 1.0.253->0
/0/0: T_Broadcast, A_IndAddr.read
[Gateway IP to subnet dispatcher] TRACE Calimero KNX server gateway - send posit
ive cEMI L_Data.con
[Gateway IP to subnet dispatcher] TRACE KNXnet/IP Tunneling 192.168.1.102:49428
- sending cEMI frame seq 11, wait for ack, attempt 1 (channel 2)
[Gateway IP to subnet dispatcher] DEBUG calimero.link.147b:5120 - send message t
o 0/0/0, blocking
[knx-server control endpoint 147b:5120] TRACE KNXnet/IP Tunneling 192.168.1.102:
49428 - received service ack 11 from /192.168.1.102:49428 (channel 2)
[Gateway IP to subnet dispatcher] TRACE calimero.link.147b:5120 - EMI 11 00 00 0
0 00 00 d1 01 00
[Gateway IP to subnet dispatcher] TRACE calimero.usb.147b:5120 - sending I/O req
uest 0113110008000901010000110000000000d1010000000000000000000000000000000000000
00000000000000000000000000000000000000000000000000000
[Gateway IP to subnet dispatcher] TRACE calimero.link.147b:5120 - send to 0/0/0
succeeded
[usb4java IRP Queue Processor] TRACE calimero.usb.147b:5120 - EP 1 IN I/O reques
t 01131100080009010100004eb010fd0000d1010000000000000000000000000000000000000000
00000000000000000000000000000000000000000000000000
[usb4java IRP Queue Processor] DEBUG calimero.usb.147b:5120 - received Emi1 fram
e 4eb010fd0000d10100
[usb4java IRP Queue Processor] DEBUG calimero.link.147b:5120 - confirmation of 0
/0/0
[usb4java IRP Queue Processor] TRACE calimero.usb.147b:5120 - EP 1 IN I/O reques
t 011311000800090101000049b0146d0000d1014000000000000000000000000000000000000000
00000000000000000000000000000000000000000000000000
[usb4java IRP Queue Processor] DEBUG calimero.usb.147b:5120 - received Emi1 fram
e 49b0146d0000d10140
[usb4java IRP Queue Processor] DEBUG calimero.link.147b:5120 - indication from 1
.4.109
[Calimero KNX server gateway] TRACE Calimero KNX server gateway - KNX subnet 147
b:5120: 1.4.109->0/0/0 L_Data.ind, system priority hop count 5 repeat tpdu 01 40

[Calimero KNX server gateway] TRACE Calimero KNX server gateway - 1.4.109->0/0/0
: T_Broadcast, A_IndAddr.response

bmalinowsky, 2015-11-10:
OK, I found the problem why the devices in programming mode weren't listed in ETS. It's committed in calimero-core, so pulling it should fix that. I ran tests using PL110 and not TP1, but that shouldn't make a difference.

alesf76, 2015-11-11:
Great! It works, both individual addresses check and address programming.
One particular combination not working: if I start the bus monitor, and then open the individual address form, this last one doesn't work, it doesn't show devices in programming mode, and on the console many of these errors are outputted

[knx-server control endpoint 147b:5120] WARN KNXnet/IP Monitor 192.168.1.104:544
59 - client is not allowed to send cEMI messages in busmonitor mode - ignored
[Calimero KNX server gateway] ERROR Calimero KNX server gateway - on dispatching KNX message
tuwien.auto.calimero.KNXIllegalArgumentException: bus monitor uses cEMI bus monitor frame type        at tuwien.auto.calimero.server.knxnetip.DataEndpointServiceHandler.check
FrameType(DataEndpointServiceHandler.java:398)

bmalinowsky, 2015-11-12:
Perfect, thanks for your support!

The warning "client is not allowed to send cEMI messages in busmonitor mode - ignored" is normal and expected:  busmonitor switches your USB interface from link layer to busmonitor layer. This is a completely passive mode, sending messages is not allowed. Therefore, with a "KNXnet/IP Monitor" connection, the server will warn about any received message from clients (and ignore them).

"bus monitor uses cEMI bus monitor frame typ" is the same story, just the other direction: in busmonitor mode you won't receive cEMI L_Data (they are ignored), only Busmon.ind.

Sending messages while monitoring only works with group monitor, then the connection type you see will be a "KNXnet/IP Tunneling", i.e., link layer.

alesf76, 2015-11-14:
OK, I realized that it's ETS5 that have problems. When using it directly connected to an USB interface, if the bus monitor is open it doesn't allow me to use the group addresses check function saying that the USB interface is already in use. But if I use it to connect to calimero, it says nothing and goes on.