calimero-project/calimero-server #3: No such method error: tuwien.auto.calimero.mgmt.PropertyClient$Property.readLevel()I

nanosonde, 2018-08-13:
Hi!

I have manually downloaded the JARs (Version 2.4-rc1) from here:
https://mvnrepository.com/artifact/com.github.calimero

Then I have added the dependencies and the dir looks like this:
```
user@lxsrv:~/calimero$ ls -R
.:
calimero-core-2.4-rc1.jar    calimero-server-2.4-rc1.jar  commons-lang3-3.2.1.jar             resources  server-config.xml      usb4java-1.2.0.jar        usb-api-1.0.2.jar
calimero-device-2.4-rc1.jar  calimero-tools-2.4-rc1.jar   libusb4java-1.2.0-linux-x86_64.jar  run.sh     server-config.xml.new  usb4java-javax-1.2.0.jar

./resources:
javax.usb.properties  properties.xml
user@lxsrv:~/calimero$
```

So everthing should be in place. 
However, when running it, I get:

```
user@lxsrv:~/calimero$ java -cp "./*:/usr/share/java/slf4j-simple.jar:/usr/share/java/slf4j-api.jar" tuwien.auto.calimero.server.Launcher server-config.xml
Exception in thread "main" java.lang.NoSuchMethodError: tuwien.auto.calimero.mgmt.PropertyClient$Property.readLevel()I
        at tuwien.auto.calimero.device.ios.InterfaceObjectServer$IosAdapter.createNewDescription(InterfaceObjectServer.java:931)
        at tuwien.auto.calimero.device.ios.InterfaceObjectServer$IosAdapter.access$100(InterfaceObjectServer.java:679)
        at tuwien.auto.calimero.device.ios.InterfaceObjectServer.initIoProperties(InterfaceObjectServer.java:610)
        at tuwien.auto.calimero.device.ios.InterfaceObjectServer.addInterfaceObject(InterfaceObjectServer.java:294)
        at tuwien.auto.calimero.device.ios.InterfaceObjectServer.<init>(InterfaceObjectServer.java:162)
        at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer.initBasicServerProperties(KNXnetIPServer.java:720)
        at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer.<init>(KNXnetIPServer.java:316)
        at tuwien.auto.calimero.server.Launcher.<init>(Launcher.java:455)
        at tuwien.auto.calimero.server.Launcher.main(Launcher.java:435)
user@lxsrv:~/calimero$
```

Any idea, what's wrong?

calimero-project, 2018-08-13:
Please use the core jar from jcenter: http://jcenter.bintray.com/com/github/calimero/calimero-core/2.4-rc1/
which is correct (the maven repo does not allow to edit/replace the jar, so this will be the case until rc2).

I  also think you are missing the _slf4j-api_ jar (and maybe any actual logger if you want to see output, e.g., _slf4j-simple_)?

nanosonde, 2018-08-19:
Thanks!
I will try the artifacts from bintray after my vacation and report back here.

nanosonde, 2018-08-27:
With the core jar from bintray it works.
Thanks!

closed 2018-08-27 by nanosonde