calimero-project/calimero-server #4: ETS5 can not program

songzh96, 2019-09-22:
Hi,

I am using calimero made a KNX IP gatway, I can monitor knx bus message，but I can't program KNX device.

The log shows the device not response. 

**Hardware:** Raspberry 4B，Siemens Bustranceiver Module BTM UP117 5WG1117-2AB12
**Software:** calimero-server

Best Regrads
songzh

bmalinowsky, 2019-09-22:
I just tried with ETS 5.7.2, server master HEAD, programming a venetian blinds actor on tp1.
(download/overwrite IA, unload device, partial download, full download). Which does work. 
I don't have an instabus coupler, though. 

To understand in detail what does not work for you (and why), I would need a server log with trace level enabled.

songzh96, 2019-09-24:
this is knx.service status,when I choose partial download
[image]

ETS 5.5.2
[image]

I was use tpuart module link Pi(GPIO14,GPIO15),  /dev/ttyS0

bmalinowsky, 2019-09-27:
Can you successfully read something else, like clicking on device info in ETS for a device? Does that work? I tried several scenarios to reproduce your issue, but couldn't (also with v2.4). 

I was probably unclear with 
> [...] server log with trace level enabled.

What I meant was to adjust the server log level like this:
1. Take a `simplelogger.properties` file, e.g., https://github.com/calimero-project/calimero-tools/blob/master/resources/simplelogger.properties
2. In that file, adjust the defaultLogLevel to trace (`defaultLogLevel=trace`) 
3. copy file to  `/etc/calimero/`
4. restart server
5. program device
6. use a command like `sudo journalctl -f -u knx -n 10000`  -> there should be hundreds of lines log output from calimero.

From that log, one can see if a DD0 response was actually sent, if it got forwarded, etc...

songzh96, 2019-09-28:
I am using a script written by someone else. I looked at it and I did not install calimero-tool.

So, I used the command sudo journalctl -f -u knx -n 10000, the output is the contents of the following file

Please help me look at it, thank you

Best Regards
Songzh

[log.txt](https://github.com/calimero-project/calimero-server/files/3665053/log.txt)

bmalinowsky, 2019-09-28:
> no subnet configured for destination 1.2.45 (received T_Connect from 1.1.129)

Your script configured the server to subnet `1.1.0` but you want to address a device with address `1.2.45`. This won't work! The allowed address range for your bus is 1.1.[0..255].

You have to change this either in your script, and call the script again, or change it directly in `/etc/calimero/server-config.xml` and restart the server.

The correct configuration is for the server to use `1.2.0` and change the additional addresses to use 1.2.x, e.g., 1.2.29, 1.2.30, etc. The additional addresses should not be assigned already to other devices.

bmalinowsky, 2019-09-28:
As a side note: your tpuart gets hot quite often and pauses sending. If that happens only now and then, it's not a problem, but your log shows it frequently.

songzh96, 2019-09-28:
Do you see "
Sep 28 04:23:42 raspberrypi java[3976]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 1.1.1: no ACK for L-Data.con"

that is when i choose partial download the log will show it.

by the way,my calimero phy address is 11.0.1.I will test it tomorrow with the method you said.

songzh96, 2019-09-28:
> As a side note: your tpuart gets hot quite often and pauses sending. If that happens only now and then, it's not a problem, but your log shows it frequently.

I have noticed it before, but I have not touched it so much.

I just connected the TP-UART to the GPIO14, 15 of the Raspberry Pi.

songzh96, 2019-09-29:
When I start knx.service, the log will show this.
Is this error causing me to download the program?

------------------------------
Sep 29 14:16:08 raspberrypi systemd[1]: Stopping Calimero KNX Daemon...
Sep 29 14:16:08 raspberrypi java[3976]: [knx-server shutdown] INFO calimero.server.gateway.knx-server - knx-server: user request for shutdown
Sep 29 14:16:08 raspberrypi java[3976]: [knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint closed
Sep 29 14:16:08 raspberrypi java[3976]: [knx-server control endpoint /dev/ttyS0] INFO calimero.server.knx-server - knx-server control endpoint /dev/ttyS0 closed
Sep 29 14:16:08 raspberrypi systemd[1]: knx.service: Succeeded.
Sep 29 14:16:08 raspberrypi systemd[1]: Stopped Calimero KNX Daemon.
Sep 29 14:16:08 raspberrypi systemd[1]: Started Calimero KNX Daemon.
Sep 29 14:16:10 raspberrypi java[23799]: [main] INFO calimero.server.knx-server - Calimero KNXnet/IP server (v2.4) 'Calimero KNXnet/IP Server'
Sep 29 14:16:11 raspberrypi java[23799]: [main] INFO calimero.server.knx-server - KNXnet/IP discovery network interfaces: listen on [all], send on [all]
Sep 29 14:16:11 raspberrypi java[23799]: [main] INFO calimero.server.knx-server - Service container '/dev/ttyS0':
Sep 29 14:16:11 raspberrypi java[23799]:         listen on eth0, KNXnet/IP routing disabled
Sep 29 14:16:11 raspberrypi java[23799]:         tpuart connection: TP1 medium, device 1.1.0
Sep 29 14:16:11 raspberrypi java[23799]: [main] INFO calimero.server.knx-server - setup tpuart subnet '/dev/ttyS0'
Sep 29 14:16:11 raspberrypi java[23799]: [main] ERROR calimero.serial.tpuart:/dev/ttyS0 - set parity failed
Sep 29 14:16:11 raspberrypi java[23799]: java.io.IOException: Invalid argument
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.serial.SerialComAdapter.setControl(Native Method)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.serial.SerialComAdapter.setParity(SerialComAdapter.java:231)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.serial.LibraryAdapter.open(LibraryAdapter.java:145)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:173)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.link.KNXNetworkLinkTpuart.<init>(KNXNetworkLinkTpuart.java:80)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$5(SubnetConnector.java:259)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.link.Connector$Link.connect(Connector.java:426)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.link.Connector$Link.<init>(Connector.java:209)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.link.Connector$Link.<init>(Connector.java:167)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.link.Connector.newLink(Connector.java:149)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:305)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.server.Launcher.connect(Launcher.java:583)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.server.Launcher.run(Launcher.java:511)
Sep 29 14:16:11 raspberrypi java[23799]:         at tuwien.auto.calimero.server.Launcher.main(Launcher.java:447)
Sep 29 14:16:11 raspberrypi java[23799]: [knx-server discovery endpoint] INFO calimero.server.knx-server - KNXnet/IP discovery listens on interface eth0
Sep 29 14:16:11 raspberrypi java[23799]: [knx-server discovery endpoint] INFO calimero.server.knx-server - KNXnet/IP discovery listens on interface lo
Sep 29 14:16:11 raspberrypi java[23799]: [knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint is up and running
Sep 29 14:16:11 raspberrypi java[23799]: [knx-server control endpoint /dev/ttyS0] INFO calimero.server.knx-server - knx-server control endpoint /dev/ttyS0 is up and running
Sep 29 14:26:14 raspberrypi systemd[1]: Stopping Calimero KNX Daemon...

bmalinowsky, 2019-09-29:
It seems unrelated. Your serial device did correctly initialize before, no? There is probably a different cause for that. 
The reason why you can't download a program is that you configured the wrong knx subnet in the server.

bmalinowsky, 2019-09-29:
> calimero.server.gateway.knx-server - timeout sending to 1.1.1: no ACK for L-Data.con"

This means that the message was sent to the knx bus, but the confirmation from the device was not received.

songzh96, 2019-09-30:
Hi @bmalinowsky 

First of all, thank you very much for your patience.

I tried to test it in various ways today，Still unable to download.
But I can use the gateway I bought perform ETS programming.  or calimero use ip connection,it will can use ETS progarming.

this is my config.xml. if configured wrong,I think it wiil be /dev/ttyS0 problem.
but I can write and read. I can not use ETS programing,

now I Turn /dev/ttyS0 into /dev/ttyAMA0, ---same problem.

Physical connection
TP-UART - TX  -> Pi Rx(GPIO 15)
TP-UART - RX -> Pi Tx(GPIO 14)
TP-UART - 5V -> Pi 5V
TP-UART - GND -> Pi Gnd

If my serial port is directly connected (Tx-Tx), I can't get the KNX Bus msg.

Best Regards
songzh
-----------------------------------------------------------------
<?xml version="1.0" encoding="UTF-8"?>

<knxServer name="knx-server" friendlyName="Calimero KNXnet/IP Server">
  <propertyDefinitions ref="/etc/calimero/properties.xml"/>
  
  <discovery listenNetIf="all" outgoingNetIf="all" activate="true"/>
  
  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="eth0">
    <knxAddress type="individual">1.1.0</knxAddress>
    
    
    
    
    
    
    
    
    
    
    <knxSubnet type="tpuart" medium="tp1">/dev/ttyS0</knxSubnet>
    
    
    
    
    
    
    
    <additionalAddresses>
      <knxAddress type="individual">1.1.129</knxAddress>
      <knxAddress type="individual">1.1.130</knxAddress>
      <knxAddress type="individual">1.1.131</knxAddress>
      <knxAddress type="individual">1.1.132</knxAddress>
      <knxAddress type="individual">1.1.133</knxAddress>
      <knxAddress type="individual">1.1.134</knxAddress>
      <knxAddress type="individual">1.1.135</knxAddress>
      <knxAddress type="individual">1.1.136</knxAddress>
    </additionalAddresses> 

    <groupAddressFilter/>
  </serviceContainer>
  
</knxServer>

bmalinowsky, 2019-09-30:
If you can read/write group values, then your physical connection is ok, no need to change that.

Your server-config is good for programming devices in the range `1.1.x`. 
But to see what goes wrong with, e.g., device 1.1.1, I would need more output from the log. Your log contains only info and warnings, please set the log config to `trace` as described in my comment above.

songzh96, 2019-10-05:
Hi @bmalinowsky 

this is my log.

Best Regards
songzh

[log.txt](https://github.com/calimero-project/calimero-server/files/3693104/log.txt)

bmalinowsky, 2019-10-10:
The inter-byte delay while receiving a frame is too long (it exceeds 4000 us). The important lines are

> reset input buffer, discard partial frame ...

> 4 partial frames discarded, increase max. inter-byte delay to 7500 us

It essentially means that several frames got discarded. And then programming won't work of course.
Which I find surprising if you connect the tpuart link over GPIO. I don't know why that happens in your particular case. I only observed those delayed timings when using a usb -> serial adapter.

One quick thing you can try is to run the 2.5 line of calimero (you use 2.4) which contains several changes:
* re-download the install script you use (it got updated recently)
* change the git branches in the script file to "master", i.e.:
    - export GIT_BRANCH="master"
    - export GIT_BRANCH_TOOLS="master"
* run the script

songzh96, 2019-10-12:
Hi， @bmalinowsky .

Thank you for your answer again.

I have upgraded the branches to the master,it worked. Unfortunately, dowland very slowly,and it will interrupt.

this is my log,can you help me check again.
[log.txt](https://github.com/calimero-project/calimero-server/files/3720074/log.txt)

Best Regards 
songzh

songzh96, 2019-11-09:
Hi

Just tested, this issue has been fixed
solution：
pi tx  --->  tpuart rx
pi rx ----> tpuart tx
gnd ----> gnd

it can download now, but very slow, It costs about 20s. if i use another gateway,it will costs 5s

I found an interesting phenomenon，Whether I cross or connect directly, it can listen to KNX bus messages.

Thank you very much for your help.

Best Regards
songzh

beireken, 2019-11-10:
I thought the Siemens tpuart module has 5v where the pi uart gpio are max 3.3v?

songzh96, 2019-11-10:
> 3.3v

But I used the knxd library to test it perfectly. I also measured the voltage, it is 3.3v

beireken, 2019-11-10:
I did measure the voltage on the Siemens uart and i had 5v. That's strange. I'll check again later.

songzh96, 2019-11-10:
> again

Yes， you are right, Siemens tpuart output is 5v and 20v. But I have made a tpuart myself. it is output is 3.3v also can use calimero library and use ETS program.

I think this has nothing to do with the voltage of the module because when I use the knxd library with Siemens tpuart, it works fine. (It may be that my wire is wrong.)

beireken, 2019-11-10:
Ow, ok. Sorry for the mistake then :)

bmalinowsky, 2019-12-18:
@songzh96 regarding downloading: I tested with the latest version of calimero-core, which shows an increase in download speed compared to the version when we had the discussion. So some downloads which seemed to "stuck" in ets, are now some seconds faster for me.

songzh96, 2019-12-19:
Hi @bmalinowsky 

When I was preparing to test, I encountered a problem.

I used calimero_2.sh script to install and run on my Raspberry Pi 4b, but this script did not add knx.service to me. How do I add it manually or how to run calimero-server.

By the way, if I have ever installed calimero, how do I upgrade, using the git command?

Best Regards
Songzh

bmalinowsky, 2019-12-19:
@songzh96 

I believe all the changes from calimero_2.sh got merged into the main script [here](https://github.com/Race666/calimero-server/blob/master/calimero.sh). 
And you can simply rerun the script, it will automatically fetch the latest changes.

But note that before you run the script, inside the script you have to adjust your knx setup if necessary, and also switch to master branch (by default the script uses v2.4, not v2.5):

`export GIT_BRANCH="release/2.4"` -> `export GIT_BRANCH="master"`
`export GIT_BRANCH_TOOLS="release/2.4"` -> `export GIT_BRANCH_TOOLS="master"`

You can also make a clean install by first running `.calimero.sh clean`

songzh96, 2019-12-19:
> @songzh96
>
> I believe all the changes from calimero_2.sh got merged into the main script [here](https://github.com/Race666/calimero-server/blob/master/calimero.sh).
> And you can simply rerun the [...]

Thank you

But I just used this script today with some errors，Could not find the download link for JDK8

Hardware: Raspberry Pi 4B
OS: Raspbian Buster

Can you fix it?

Best Regards
Songzh

bmalinowsky, 2019-12-19:
The default jdk for raspberry buster is OpenJDK 11, not jdk 8. I think there is something wrong in the script you use.

songzh96, 2019-12-19:
probably is. But today I used "git clone https://github.com/Race666/calimero-server.git"
Then run, the script will report an error (JDK version problem).

I'll try again tomorrow. If there are any questions, I will post them here on time.
Thank you for your answer.

Best Regards
Songzh

closed 2019-12-19 by songzh96

songzh96, 2019-12-23:
Hi @bmalinowsky 

Today I tested calimero on my Raspberry Pi and it didn't work very well.
I used the calimero installed by calimero.sh

- it can send and receive information, but the connection will be lost halfway
- it also can't download programs for knx devices

**Hardware:** Pi 4B  &  tpuart module(Siemens)
**interface:** /dev/ttyAMA0 or /dev/ttyS0(the interface will show high temperature warning)
**software:** calimero.sh(master)

#### log
```
Dec 23 12:03:39 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:41 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:44 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:47 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:50 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:53 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:56 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:03:59 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:02 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:05 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:08 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:11 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:14 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 0/0/0: no ACK for L-Data.con
Dec 23 12:04:31 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
Dec 23 12:04:31 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
Dec 23 12:04:42 raspberrypi java[1435]: [/dev/ttyAMA0 data endpoint /192.168.8.199:62358] INFO calimero.server.knx-server - /dev/ttyAMA0 data endpoint /192.168.8.199:62358 closed
Dec 23 12:04:43 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
Dec 23 12:04:45 raspberrypi java[1435]: [knx-server control endpoint /dev/ttyAMA0] WARN calimero.server.knx-server - received disconnect request with unknown channel id 9 - ignored
Dec 23 12:04:47 raspberrypi java[1435]: [knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - /dev/ttyAMA0: setup data endpoint (channel 10) for connection request from /192.168.8.199:63240 (udp)
Dec 23 12:04:47 raspberrypi java[1435]: [knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - assign additional individual address 1.1.129 to channel 10
Dec 23 12:04:47 raspberrypi java[1435]: [/dev/ttyAMA0 data endpoint /192.168.8.199:63240] INFO calimero.server.knx-server - /dev/ttyAMA0 data endpoint /192.168.8.199:63240 is up and running
Dec 23 12:04:47 raspberrypi java[1435]: [knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
```

Best Regards
Songzh

bmalinowsky, 2019-12-23:
Thank you. That's unfortunate, so the changes made it worse on your device. 
Can you again enable `trace` mode for logging, like I wrote in my reply here: https://github.com/calimero-project/calimero-server/issues/4#issuecomment-536029334 , and show the log file.

songzh96, 2019-12-24:
some logs

```
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:959 TRACE calimero.server.gateway.knx-server - server-side KNX IP Tunneling 192.168.8.199:64091 (channel 3, 1.1.129): 1.1.129->1.1.40 L_Data.req, system priority hop count 6 repeat, tpdu 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:959 TRACE calimero.server.gateway.knx-server - 1.1.129->1.1.40: T_Disconnect
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:960 TRACE calimero.server.gateway.knx-server - send positive cEMI L_Data.con
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:960 DEBUG calimero.server.gateway.knx-server - dispatch 1.1.129->1.1.40 using KNXnet/IP routing service 224.0.23.12
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:961 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.8.199:64091 - sending cEMI frame seq 5, WaitForAck, attempt 1 (channel 3) 06 10 04 20 00 14 04 03 05 00 2e 00 b0 60 11 81 11 28 00 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:961 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - add to multicast loopback frame buffer: 1.1.129->1.1.40 L_Data.ind, system priority hop count 6, tpdu 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:962 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 10 29 00 b0 60 11 81 11 28 00 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:963 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - discard multicast loopback cEMI frame: 1.1.129->1.1.40 L_Data.ind, system priority hop count 6, tpdu 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:963 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.8.199:64091 - received service ack 5 from /192.168.8.199:64091 (channel 3)
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:964 TRACE calimero.server.gateway.knx-server - dispatch to KNX subnet 1.1.0 (tpuart:/dev/ttyAMA0 in service container '/dev/ttyAMA0')
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:965 DEBUG calimero.link.tpuart:/dev/ttyAMA0 - send (wait for confirmation) 1.1.129->1.1.40 L_Data.req, system priority hop count 5 repeat, tpdu 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:966 TRACE calimero.link.tpuart:/dev/ttyAMA0 - cEMI 11 00 b0 50 11 81 11 28 00 81
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:966 TRACE calimero.serial.tpuart:/dev/ttyAMA0 - create UART services 80 b0 81 11 82 81 83 11 84 28 85 50 86 81 47 37
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:967 TRACE calimero.serial.tpuart:/dev/ttyAMA0 - UART ready for sending after 4 us
Dec 24 02:52:19 raspberrypi java[3518]: 02:52:19:967 DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - write UART services, waiting for .con
Dec 24 02:52:20 raspberrypi java[3518]: 02:52:20:054 WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:477 TRACE calimero.server.gateway.knx-server - server-side KNX IP Tunneling 192.168.8.199:64091 (channel 3, 1.1.129): 1.1.129->1.1.40 L_Data.req, system priority hop count 6 repeat, tpdu 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:478 TRACE calimero.server.gateway.knx-server - 1.1.129->1.1.40: T_Connect
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:478 TRACE calimero.server.gateway.knx-server - send positive cEMI L_Data.con
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:479 DEBUG calimero.server.gateway.knx-server - dispatch 1.1.129->1.1.40 using KNXnet/IP routing service 224.0.23.12
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:479 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.8.199:64091 - sending cEMI frame seq 6, WaitForAck, attempt 1 (channel 3) 06 10 04 20 00 14 04 03 06 00 2e 00 b0 60 11 81 11 28 00 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:479 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - add to multicast loopback frame buffer: 1.1.129->1.1.40 L_Data.ind, system priority hop count 6, tpdu 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:480 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 10 29 00 b0 60 11 81 11 28 00 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:481 TRACE calimero.server.knxnetip.KNXnet/IP routing service 224.0.23.12 - discard multicast loopback cEMI frame: 1.1.129->1.1.40 L_Data.ind, system priority hop count 6, tpdu 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:482 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.8.199:64091 - received service ack 6 from /192.168.8.199:64091 (channel 3)
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:483 TRACE calimero.server.gateway.knx-server - dispatch to KNX subnet 1.1.0 (tpuart:/dev/ttyAMA0 in service container '/dev/ttyAMA0')
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:484 DEBUG calimero.link.tpuart:/dev/ttyAMA0 - send (wait for confirmation) 1.1.129->1.1.40 L_Data.req, system priority hop count 5 repeat, tpdu 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:484 TRACE calimero.link.tpuart:/dev/ttyAMA0 - cEMI 11 00 b0 50 11 81 11 28 00 80
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:485 TRACE calimero.serial.tpuart:/dev/ttyAMA0 - create UART services 80 b0 81 11 82 81 83 11 84 28 85 50 86 80 47 36
Dec 24 02:52:21 raspberrypi java[3518]: 02:52:21:486 TRACE calimero.serial.tpuart:/dev/ttyAMA0 - UART ready for sending after 5 us
[... cut ...]
Dec 24 02:52:33 raspberrypi java[3518]: 02:52:33:638 DEBUG calimero.server.knxnetip.KNX IP Tunneling 192.168.8.199:64091 - close connection for channel 3 - client request
Dec 24 02:52:33 raspberrypi java[3518]: 02:52:33:639 DEBUG calimero.server.gateway.knx-server - removed connection KNX IP Tunneling 192.168.8.199:64091 (client request)
Dec 24 02:52:33 raspberrypi java[3518]: 02:52:33:639 INFO calimero.server.knx-server - /dev/ttyAMA0 data endpoint /192.168.8.199:64091 closed
Dec 24 02:52:33 raspberrypi java[3518]: 02:52:33:704 WARN calimero.server.gateway.knx-server - timeout sending to 1.1.40: no ACK for L-Data.con
```