calimero-project/calimero-server #6: USB interface with "ERROR calimero.link.connector"

silviopen, 2020-04-17:
Hello,
i am trying to try openhab on Raspberry 3B to manage a knx system.
I discovered the calimero project via the blog www.michlstechblog.info of which I followed the instructions to install calimero and the installation seems to have been successful, but it doesn't seem to connect to the USB interface.

ZENNIO KNX USB INTERFACE ZN1SY-USBP
Raspberry 3B

I updated the file "/etc/udev/rules.d/90-knxusb-devices.rules" with the ID of my interface (28c2 - 0002) but using the command "journalctl -xu knx" I noticed what I think is an error (I apologize if the term is incorrect, but I'm a beginner).

I attach what appears on the terminal after the command, hoping that someone can help me, because I have already given up using KNXD as I was unable to use/configure it.

thankyou

[journalctl log.txt](https://github.com/calimero-project/calimero-server/files/4491914/journalctl.log.txt)

bmalinowsky, 2020-04-17:
The default setting of Michael's script relies on calimero finding all "known" knx interfaces plugged to your computer by their `vendor:product` id. Your usb device is found but not recognized as knx interface (therefore, it is ignored). 

You can easily specify your exact interface in the server config, located in _/etc/calimero/server-config.xml_.

Look for the line `<knxSubnet type="usb" medium="tp1"/>`.

Since your usb device's ID is `28c2:0002` and is named `zennio`, you can either modify the line to
`<knxSubnet type="usb" medium="tp1">28c2:0002</knxSubnet>`
or to
`<knxSubnet type="usb" medium="tp1">zennio</knxSubnet>`.

Also (not necessary), in the script, in lines 74 and 76, you can change 
"release/2.4" to "master". I will add your knx interface to the known interfaces in the master branch, so next time you run the script it will be detected automatically like other products.

Thanks for telling me!

silviopen, 2020-04-23:
As I have already written by email,
Thanks for the tip, it worked.

In Michael's script I edited lines 75 and 77 (because lines 74 and 76 are just comments on the code). 

Did you mean so ?:
```javascript
# Branch to use 
export GIT_BRANCH="master"
# Build for Tools with master branch failed at 20190612 -> 2.4 release
export GIT_BRANCH_TOOLS="master"
```
Now I have only one doubt. Like the other user, I also noticed a difference in the response times of the system according to the IP address used for the command
```
time knxtools write 1/1/5 switch on 192.168.178.111
12:31:20:800 INFO calimero.knxnetip.KNXnet/IP Tunneling 192.168.178.111:3671 - establish connection from /192.168.178.111:48117 to /192.168.178.111:3671
12:31:20:829 INFO calimero.knxnetip.KNXnet/IP Tunneling 192.168.178.111:3671 - connection established (channel 7)
12:31:20:836 INFO calimero.knxnetip.KNXnet/IP DevMgmt 192.168.178.111:3671 - establish connection from /192.168.178.111:50991 to /192.168.178.111:3671
12:31:22:871 INFO calimero.knxnetip.KNXnet/IP DevMgmt 192.168.178.111:3671 - connection established (channel 8)
write to 1/1/5 successful
12:31:25:505 INFO calimero.process.communication 192.168.178.111:3671 - detached from link 192.168.178.111:3671
12:31:25:512 INFO calimero.link.192.168.178.111:3671 - link closed

real    0m5.662s
user    0m1.730s
sys    0m0.160s
```
```
[12:31:25] openhabian@openhab:~$ time knxtools write 1/1/5 switch off 224.0.23.12
12:31:30:726 INFO calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671 - multicast loopback mode enabled
write to 1/1/5 successful
12:31:31:242 INFO calimero.process.communication 224.0.23.12:3671 - detached from link 224.0.23.12:3671
12:31:31:251 INFO calimero.link.224.0.23.12:3671 - link closed

real    0m1.424s
user    0m1.552s
sys    0m0.123s
```
I don't know if this is important, but I still report it.

For use with openhab, I am not sure which IP address to use for my KNX/USB interface in the configuration file and which type to use TUNNEL or ROUTER.

bmalinowsky, 2020-04-25:
The change to branch master is OK. 

In one of the recent commits I added the product ID of your usb device, so in the future when you reinstall the server, your usb interface should be recognized automatically (no server-config adjustment necessary).

Wrt to the delay in response times: unfortunately I did only rarely observe this so far on my rpi, so I am not sure what causes this. However, I removed 2 usb read accesses during the connection setup, which should improve timings. Maybe you can try it out.

The IP address for tunneling is the same as the IP of your rpi (192.168.178.111). You can also figure it out by calling the discovery service:
```
knxtools discover
```

For routing, you always use the multicast address `224.0.23.12`.

Whether you use routing or tunneling does not really matter, choose your preference or what works better.

NB. if you want to reinstall the server, you would do this:
```
sudo ./calimero.sh clean
sudo ./calimero.sh usb
sudo rm /etc/calimero-server/properties.xml
```

silviopen, 2020-04-26:
Hi, thanks for support.
I tried the command
> knxtools discover

but there are no files in the `/opt/calimero` folder 
```
Error: Unable to access jarfile /opt/calimero-tools/calimero-tools-2.*.jar
```

I also tried updating calimero but after the command `sudo ./calimero.sh usb` signals me a problem
```
Your branch is up to date with 'origin/master'.
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details
> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Could not target platform: 'Java SE 11' using tool chain: 'JDK 8 (1.8)'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 31s
1 actionable task: 1 executed
```

I don't have enough skills to figure out what to do

bmalinowsky, 2020-04-26:
The cause is an old jdk. The script actually checks that before compiling, so I wonder how you got there. Java 11 should be the default, not Java 8.

Check the output of ```java -version```, does it show java version 11?

Do
```
apt-cache show openjdk-11-jdk-headless
```

or 
```
sudo update-alternatives --config java
```

You can also simply try to install jdk 11 by typing

```
sudo apt-get -i -y openjdk-11-jdk-headless
```

silviopen, 2020-04-26:
Nowthere are still problems and sorry for the long message.
To avoid making further messes, I try to describe everything I have done.

- In the past few days I have fixed the IP address by setting a static address by setting the router, which is now the following. 192.168.178,10
- I've reactivated the cloud connector, which was no longer online, following a guide on the forum and I hope I have not changed anything that had to do with calimero
- I've 

`java -version` says:
```
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (Zulu8.40.0.178-CA-linux_aarch32hf) (build 1.8.0_222-b178)
OpenJDK Client VM (Zulu8.40.0.178-CA-linux_aarch32hf) (build 25.222-b178, mixed mode, evaluation)
```

- I changed java version with the command

`sudo update-alternatives --config java` and now is:
```
openjdk version "11.0.7" 2020-04-14
OpenJDK Runtime Environment (build 11.0.7+10-post-Raspbian-3deb10u1)
OpenJDK Server VM (build 11.0.7+10-post-Raspbian-3deb10u1, mixed mode)
```

- I used the following commands preceded by `sudo su -` otherwise they didn't work
> ./calimero.sh clean
> ./calimero.sh usb

- I restarted the raspberry as recommended at the end of the long installation process

- the last command `sudo rm /etc/calimero-server/properties.xml` give me error 
`rm: cannot remove '/etc/calimero-server/properties.xml': No such file or directory`

now `knxtools discover` answers:
` INFO calimero.knxnetip.Discoverer - search stopped after 3 seconds with 0 responses`

I can't no more command the light. The following commands do nothing
` knxtools write 1/1/5 switch on 224.0.23.12`
```
15:54:42:247 INFO calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671 - multicast loopback mode disabled
write to 1/1/5 successful
15:54:43:562 INFO calimero.link.224.0.23.12:3671 - link closed
```

`knxtools write 1/1/5 switch on 192.168.178.10`
```
15:55:26:925 ERROR calimero.knxnetip.KNXnet/IP Tunneling 192.168.178.10:3671 - establishing connection failed, timeout connecting to control endpoint /192.168.178.10:3671
15:55:26:943 ERROR calimero.tools.ProcComm - completed with error
tuwien.auto.calimero.KNXTimeoutException: timeout connecting to control endpoint /192.168.178.10:3671
        at tuwien.auto.calimero.knxnetip.ClientConnection.connect(ClientConnection.java:233)
        at tuwien.auto.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:169)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.newConnection(KNXNetworkLinkIP.java:430)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.<init>(KNXNetworkLinkIP.java:266)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.newTunnelingLink(KNXNetworkLinkIP.java:149)
        at tuwien.auto.calimero.tools.Main.newLink(Main.java:435)
        at tuwien.auto.calimero.tools.ProcComm.createLink(ProcComm.java:447)
        at tuwien.auto.calimero.tools.ProcComm.start(ProcComm.java:311)
        at tuwien.auto.calimero.tools.ProcComm.run(ProcComm.java:271)
        at tuwien.auto.calimero.tools.ProcComm.main(ProcComm.java:257)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at tuwien.auto.calimero.tools.Main.main(Main.java:159)
```
the openhabian log also reports a problem
```
2020-04-26 16:01:37.458 [ERROR] [net/IP Tunneling 192.168.178.10:3671] - establishing connection failed, timeout connecting to control endpoint /192.168.178.10:3671
```

If it helps I paste here the result of `journalctl -xu knx -b`
[... 20 lines cut ...]
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.device.BaseKnxDevice.initTableProperties(BaseKnxDevice.java:500)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.device.BaseKnxDevice.initIos(BaseKnxDevice.java:470)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.device.BaseKnxDevice.<init>(BaseKnxDevice.java:179)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.device.BaseKnxDevice.<init>(BaseKnxDevice.java:243)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer.<init>(KNXnetIPServer.java:345)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.server.Launcher.<init>(Launcher.java:602)
Apr 26 14:48:58 openhab java[535]:         at tuwien.auto.calimero.server.Launcher.main(Launcher.java:581)
Apr 26 14:48:58 openhab systemd[1]: knx.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit knx.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Apr 26 14:48:58 openhab systemd[1]: knx.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit knx.service has entered the 'failed' state with result 'exit-code'.
lines 2-40/40 (END)
```

Do you need more information or are you already able to tell me what happened?

bmalinowsky, 2020-04-26:
Perfect, the jdk part is solved. (The problem was likely because you already had installed Zulu and manually overrode the config to stay on java 8.)

Discover does not show anything because the server is not running.

The server start problem is exactly because there is the `properties.xml` file leftover.
--> I typed the folder wrong in my answer: the folder is _/etc/calimero_, not /etc/_calimero-server_. Sorry for that!

So the command would be
```
sudo rm /etc/calimero/properties.xml
```

Then restart the server:
```
sudo systemctl restart knx
```

bmalinowsky, 2020-04-26:
As a side note: you can of course run any jdk distribution you want, like zulu 11 or zulu 14 etc. It does not need to be the default openjdk.

silviopen, 2020-04-26:
Another step towards the solution :-)
now the command `knxtools discover`

gives this result (which I didn't remember before was so long):
```
Using 192.168.178.10 at eth0
----------------------------
"Calimero KNXnet/IP Server" endpoint 192.168.178.10:3671 (IPv4 UDP & TCP)
KNX address 1.1.0
KNX medium TP1
Installation 0 - Project 0 (ID 0)
KNX IP multicast address 224.0.23.12
MAC address b8:27:eb:34:9a:3d
Supported services: Core (v2), Device Management (v2), Tunneling (v2), Routing (v2)
DD0 091A - KNXnet/IP Router communication possible
max local APDU length 1400
maximum APDU length: 15
addresses 1.1.129 (occupied, authorized, usable) 1.1.130 (free, authorized, usable) 1.1.131 (free, authorized, usable) 1.1.132 (free, authorized, usable) 1.1.133 (free, authorized, usable) 1.1.134 (free, authorized, usable) 1.1.135 (free, authorized, usable) 1.1.136 (free, authorized, usable)

Using 192.168.178.10 at eth0
----------------------------
"Calimero KNXnet/IP Server" endpoint 192.168.178.10:3671 (IPv4 UDP & TCP)
KNX address 1.1.0
KNX medium TP1
Installation 0 - Project 0 (ID 0)
KNX IP multicast address 224.0.23.12
MAC address b8:27:eb:34:9a:3d
Supported services: Core (v2), Device Management (v2), Tunneling (v2), Routing (v2)

Using 127.0.0.1 at lo
---------------------
"Calimero KNXnet/IP Server" endpoint 192.168.178.10:3671 (IPv4 UDP & TCP)
KNX address 1.1.0
KNX medium TP1
Installation 0 - Project 0 (ID 0)
KNX IP multicast address 224.0.23.12
MAC address b8:27:eb:34:9a:3d
Supported services: Core (v2), Device Management (v2), Tunneling (v2), Routing (v2)
DD0 091A - KNXnet/IP Router communication possible
max local APDU length 1400
maximum APDU length: 15
addresses 1.1.129 (occupied, authorized, usable) 1.1.130 (free, authorized, usable) 1.1.131 (free, authorized, usable) 1.1.132 (free, authorized, usable) 1.1.133 (free, authorized, usable) 1.1.134 (free, authorized, usable) 1.1.135 (free, authorized, usable) 1.1.136 (free, authorized, usable)

Using 127.0.0.1 at lo
---------------------
"Calimero KNXnet/IP Server" endpoint 192.168.178.10:3671 (IPv4 UDP & TCP)
KNX address 1.1.0
KNX medium TP1
Installation 0 - Project 0 (ID 0)
KNX IP multicast address 224.0.23.12
MAC address b8:27:eb:34:9a:3d
Supported services: Core (v2), Device Management (v2), Tunneling (v2), Routing (v2)
17:58:38:152 INFO calimero.knxnetip.Discoverer - search stopped after 3 seconds with 4 responses
```

But `knxtools write 1/1/5 switch on 224.0.23.12` does not produce any results yet even if the messages would indicate otherwise :-(
```
18:15:09:997 INFO calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671 - multicast loopback mode disabled
write to 1/1/5 successful
18:15:11:307 INFO calimero.link.224.0.23.12:3671 - link closed
```

silviopen, 2020-04-26:
> As a side note: you can of course run any jdk distribution you want, like zulu 11 or zulu 14 etc. It does not need to be the default openjdk.

Here I demonstrate all my inexperience.
Can I delete the other versions of java that are not needed (and above all: how can I verify what the other versions are for)?

bmalinowsky, 2020-04-26:
The discovery result shows two things: 
1. the server is discoverable from a remote host (via the server IP `192.168.178.10`) and locally using the loopback (`127.0.0.1`), that's why there is for example "Using 192.168.178.10 at eth0"
2. the server supports KNXnet/IP v2 (that's the longer search response, it shows a little bit more info) because that is useful for newer protocol versions

That's why there are 4 entries in total, but it's always the same server (I might condense the output in the future, it's somewhat verbose).

Re knxtools: the output says that the client is OK with it. So I would check what the server says. For example, if you do in one terminal this: 
```
journalctl -f -u knx -n 5000 -o short-precise
```
and then do the knxtools write, what does the server say (not the old log output, just the new one)?

Re older Java versions: in general you can always delete older versions (unless you have a specific application that requires a specific java version to run; for java 8 that might be stuff that is > 2 years old and was not updated yet). Just using Java 11/12/13 should be quite safe.

silviopen, 2020-04-26:
with `journalctl -f -u knx -n 5000 -o short-precise`
I get this looped continuously (I stopped it with ctrl + c)
```
Apr 26 19:25:21.707398 openhab java[5277]: 19:25:21:707 DEBUG calimero.link.connector - execute scheduled connect 81 (no max)
Apr 26 19:25:21.723530 openhab java[5277]: 19:25:21:723 DEBUG calimero.usb.28c2:0002 - extracting USB device strings, javax.usb.UsbPlatformException: USB error 3: Can't open device Bus 001 Device 004: ID 28c2:0002: Access denied (insufficient permissions)
Apr 26 19:25:21.724833 openhab java[5277]: 19:25:21:723 INFO calimero.usb.28c2:0002 - Bus 001 Device 004: ID 28c2:0002
Apr 26 19:25:21.724833 openhab java[5277]: 19:25:21:724 TRACE calimero.usb.28c2:0002 - Interface USB interface 00, setting 0
Apr 26 19:25:21.724833 openhab java[5277]: 19:25:21:724 TRACE calimero.usb.28c2:0002 - EP 1 OUT
Apr 26 19:25:21.727583 openhab java[5277]: 19:25:21:724 TRACE calimero.usb.28c2:0002 - EP 2 IN
Apr 26 19:25:21.727583 openhab java[5277]: 19:25:21:725 DEBUG calimero.usb.28c2:0002 - Found USB device endpoint addresses OUT 0x1, IN 0x82
Apr 26 19:25:21.729738 openhab java[5277]: 19:25:21:728 WARN calimero.link.connector - connection attempt 81: open USB connection '28c2:0002' (USB error 3: Can't open device Bus 001 Device 004: ID 28c2:0002: Access denied (insufficient permissions))
```

bmalinowsky, 2020-04-26:
This:
```
open USB connection '28c2:0002' (USB error 3: Can't open device Bus 001 Device 004: ID 28c2:0002: Access denied (insufficient permissions)
```
is Linux access permissions. The user you run the server with does not have permission to open the usb interface.

silviopen, 2020-04-26:
I confess that I have no idea how to solve this other problem...

bmalinowsky, 2020-04-26:
If you look at udev rules at `/etc/udev/rules.d/90-knxusb-devices.rules` there should already be some knx rules for usb interfaces. Check if your usb interface is listed (the `vendor:product` stuff). Otherwise add a line with it (then log out/in, maybe restart server).

silviopen, 2020-04-26:
Okay, I updated the `90-knxusb-devices.rules` file even though I had already done it a couple of times when I first installed Calimero. 

Now I can see telegrams on bus (the output don't stop, I've used an other ssh connection to give the command) and at the center you can see the light at 1/1/5) but the command `knxtools write 1/1/5 switch on 224.0.23.12` it has no effect on real light.
```
Apr 26 20:00:08.432597 openhab java[543]: 20:00:08:431 DEBUG calimero.server.gateway.knx-server - dispatch 2.2.201->1/3/57 to all server-side connections
Apr 26 20:00:08.432597 openhab java[543]: 20:00:08:432 TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 11 29 00 bc d0 22 c9 0b 39 01 00 80
Apr 26 20:00:08.434309 openhab java[543]: 20:00:08:433 TRACE calimero.server.gateway.knx-server - server-side KNX IP routing service 224.0.23.12: 2.2.201->1/3/57 L_Data.ind, low priority hop count 5, tpdu 00 80
Apr 26 20:00:08.434309 openhab java[543]: 20:00:08:433 TRACE calimero.server.gateway.knx-server - 2.2.201->1/3/57: T_Group, A_Group.write
Apr 26 20:00:08.434309 openhab java[543]: 20:00:08:433 WARN calimero.server.gateway.knx-server - cannot dispatch to KNX subnet 1.1.0, no network link (null)
Apr 26 20:00:08.435604 openhab java[543]: 20:00:08:434 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.178.10:50258 - sending cEMI frame seq 235, WaitForAck, attempt 1 (channel 1) 06 10 04 20 00 15 04 01 eb 00 29 00 bc d0 22 c9 0b 39 01 00 80
Apr 26 20:00:08.436280 openhab java[543]: 20:00:08:436 TRACE calimero.server.knxnetip.KNX IP Tunneling 192.168.178.10:50258 - received service ack 235 from /192.168.178.10:50258 (channel 1)
Apr 26 20:00:08.436832 openhab java[543]: 20:00:08:436 TRACE calimero.server.gateway.knx-server - dispatching to KNX subnets: exclude subnet usb-1.1.0

Apr 26 20:00:08.612783 openhab java[543]: 20:00:08:612 TRACE calimero.server.gateway.knx-server - server-side KNX IP routing service 224.0.23.12: 0.0.0->1/1/5 L_Data.ind, low priority hop count 6, tpdu 00 81
Apr 26 20:00:08.613360 openhab java[543]: 20:00:08:612 TRACE calimero.server.gateway.knx-server - 0.0.0->1/1/5: T_Group, A_Group.write

Apr 26 20:00:08.613360 openhab java[543]: 20:00:08:612 WARN calimero.server.gateway.knx-server - cannot dispatch to KNX subnet 1.1.0, no network link (null)
Apr 26 20:00:09.106453 openhab java[543]: 20:00:09:106 TRACE calimero.usb.28c2:0002 - EP 2 IN I/O request 0113150008000d010300002900bce022c90b3303008007bc
Apr 26 20:00:09.106453 openhab java[543]: 20:00:09:106 DEBUG calimero.usb.28c2:0002 - received cEMI frame 2900bce022c90b3303008007bc
Apr 26 20:00:09.107619 openhab java[543]: 20:00:09:106 DEBUG calimero.link.28c2:0002 - indication 2.2.201->1/3/51 L_Data.ind, low priority hop count 6, tpdu 00 80 07 bc
Apr 26 20:00:09.107619 openhab java[543]: 20:00:09:107 TRACE calimero.server.gateway.knx-server - KNX subnet usb-1.1.0: 2.2.201->1/3/51 L_Data.ind, low priority hop count 6, tpdu 00 80 07 bc
Apr 26 20:00:09.107619 openhab java[543]: 20:00:09:107 TRACE calimero.server.gateway.knx-server - 2.2.201->1/3/51: T_Group, A_Group.write
Apr 26 20:00:09.108675 openhab java[543]: 20:00:09:107 DEBUG calimero.server.gateway.knx-server - dispatch 2.2.201->1/3/51 to all server-side connections
Apr 26 20:00:09.108675 openhab java[543]: 20:00:09:108 TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 13 29 00 bc d0 22 c9 0b 33 03 00 80 07 bc

bmalinowsky, 2020-04-26:
OK, now the access permissions are good. I see in the log that the link is not initialized.

What if you try the tunneling mode, like this:
```
knxtools write 1/1/5 switch on 192.168.178.10
```

If that does not work, what does the log show if you restart the server. There has to be a warning or error.

silviopen, 2020-04-26:
I'm not sure what happened, but after giving `sudo systemctl restart knx`it's working! 
Yet I am convinced that I have restarted the raspberry to be sure that everything started again.

I have verified that with **IP 192.168.178.10** it takes a little more time (4/5 sec.), 
instead with **224.0.23.12** it is faster (about 1 sec.).
It also works perfectly from UI (via browser and APP).

Now I'll do another backup of the SD in order to have the most recent restore point :-)

At this point I am satisfied, because my goal was to have a functioning graphical interface to manage the system. I need help to remove useless version of Java but it's off topic.

If you think you need to do anything else, just tell me. I can only thank you for the extreme willingness to help a noob.

bmalinowsky, 2020-04-26:
Thank you for the kind words!

You can remove a jdk with a command starting like this:
```
sudo apt remove zulu...
```
Try auto-complete on the zulu part, I don't know the exact name of your distribution. I guess it starts with `zulu8`?

Wrt the script, I will look at it and see if some of the quirks you observed can be smoothed out in the repo of the author, so that anyone benefits.

bmalinowsky, 2020-04-26:
Btw. that unicast connections (i.e., IP 192.168.178.10) take a little longer is normal, because the protocol is actually establishing a connection (the delay is the initial setup). Multicast (224.0.23.12) is using a connection-less protocol.

With a remote gui or web-app you probably won't notice the difference that much (in comparison to a single `knxtools write`) because the connection is usually established and kept open.

The improvement I was hoping to see was the reduction of the quite obscure initial delay (2 seconds) as you observed in

```
12:31:20:836 INFO calimero.knxnetip.KNXnet/IP DevMgmt 192.168.178.111:3671 - establish connection from /192.168.178.111:50991 to /192.168.178.111:3671
12:31:22:871 INFO calimero.knxnetip.KNXnet/IP DevMgmt 192.168.178.111:3671 - connection established (channel 8)
```

silviopen, 2020-04-26:
> You can remove a jdk with a command starting like this:
> 
> ```
> sudo apt remove zulu...
> ```
> 
> Try auto-complete on the zulu part, I don't know the exact name of your distribution. I guess it starts with `zulu8`?

Yes It is. Tomorrow morning I will try ;-)

silviopen, 2020-04-26:
> With a remote gui or web-app you probably won't notice the difference that much (in comparison to a single knxtools write) because the connection is usually established and kept open.

I don't understand the technical details but I can tell you that from the UI it's practically immediate, compared to the few seconds of the command from the terminal and for now it's enough for me.
If I have to check any parameters, please tell me

closed 2020-06-16 by calimero-project