calimero-project/calimero-server #8: unknown service type 0x205 - ignored

tpetkovich, 2021-02-02:
Greetings!

I am using calimero-server on a Windows box, to tunnel an USB KNX connection for an OpenHAB3 server. (As OpenHAB KNX binding unfortunately does not have an USB implementation). Both calimero-server and the OpenHAB3 is on the same Windows box, using a loopback interface to communicate. Due to some reason the calimero-server does not accept the 0x205 CONNECTION_REQUEST KNX/IP service type, which is sent by the OpenHAB binding (which is based on calimero as well). ETS5 discovery and connection works perfectly. Is this working as intended? Or is there some kind of problem with the KNX/IP protocol used between the binding and claimero-server?

My server-config.xml:
```
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />

	<serviceContainer reuseCtrlEP="false" activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="any">
		<knxAddress type="individual">1.1.80</knxAddress>
		<knxSubnet type="usb" medium="tp1">USB Interface (MDRC)</knxSubnet>
	</serviceContainer>

</knxServer>
```

The log of the calimero-server:
```
[main] INFO calimero.device.knx-server - could not open file:///c:/Adat/test/bin/knx-server-ios.xml, create resource on closing device (c:\Adat\test\bin\knx-server-ios.xml (A rendszer nem találja a megadott fájlt))
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.5-dev
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - USB Interface (MDRC):
        listen on any (port 3671), KNX IP routing disabled
        usb USB Interface (MDRC) connection: TP1 medium, device 1.1.80
[main] INFO calimero.server.knx-server - setup usb subnet 'USB Interface (MDRC)'
type 'stop' to stop the gateway and shutdown the server
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net0
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth0
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth1
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net1
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface ppp0
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net2
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth3
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net3
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth4
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net4
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net5
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface net6
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth5
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth6
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth7
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth8
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth9
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth10
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth11
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth12
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth13
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth14
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth15
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth16
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth17
[knx-server discovery endpoint] WARN calimero.server.knx-server - KNXnet/IP discovery join fails with no IP address bound to interface eth18
[knx-server discovery endpoint] INFO calimero.server.knx-server - KNXnet/IP discovery listens on interfaces [lo, eth2]
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint is up and running
[knx-server control endpoint USB Interface (MDRC)] INFO calimero.server.knx-server - control endpoint 'USB Interface (MDRC)' secure mgmt/tunneling connections: optional/optional
[knx-server tcp service USB Interface (MDRC)] INFO calimero.server.knx-server - knx-server tcp service USB Interface (MDRC) is up and running
[knx-server control endpoint USB Interface (MDRC)] INFO calimero.server.knx-server - knx-server control endpoint USB Interface (MDRC) is up and running
[knx-server] INFO calimero.usb - Found 1 KNX USB devices:
 |--Bus 001 Device 004: ID 147b:5120
[knx-server] INFO calimero.usb.USB Interface (MDRC) - Bus 001 Device 004: ID 147b:5120
[knx-server] INFO calimero.link.USB Interface (MDRC) - Device Descriptor (Mask Version) 0012 - System 1 (BCU 1)
[knx-server] INFO calimero.link.USB Interface (MDRC) - PEI type 1
enforce transmission interframe spacing, wait 14 ms
enforce transmission interframe spacing, wait 14 ms
enforce transmission interframe spacing, wait 9 ms
enforce transmission interframe spacing, wait 30 ms
enforce transmission interframe spacing, wait 14 ms
enforce transmission interframe spacing, wait 30 ms
[knx-server] INFO calimero.link.USB Interface (MDRC) - KNX individual address 1.1.255
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /127.0.0.1:58893 with unknown service type 0x205 - ignored
```

bmalinowsky, 2021-02-02:
The log line is an info from the discovery endpoint, not the control endpoint. You can ignore that. (I could actually omit that log entry as it might cause confusion.)
ETS likely connects via TCP, that's why you don't see the output there.

If you don't reuse the control endpoint address ```reuseCtrlEP="false"```, you have to leave the ```additionalAddresses``` element in there and add some addresses you want to use.

tpetkovich, 2021-02-02:
Thank You!

Now I understand that the discovery-endpoint was not relevant in this case, so I deactivated it. The control endpoint was not receiving the connection. I experimented a bit, and it seemed, that it was not using the right interface with listenNetIf="any", so I have changed it to listenNetIf="lo", and turned on reuseCtrlEP. Now it seems to be working fine, the connection is established and KNX telegrams are sent to OpenHAB.

Here is my working server-config.xml: 

```
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
	<discovery listenNetIf="all" outgoingNetIf="all" activate="false" />

	<serviceContainer reuseCtrlEP="true" activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="lo">
		<knxAddress type="individual">1.1.80</knxAddress>
		<knxSubnet type="usb" medium="tp1">USB Interface (MDRC)</knxSubnet>
	</serviceContainer>

</knxServer>
```
Thank You again for the swift response and help!

bmalinowsky, 2021-02-06:
FYI, the one downside with a single knx address is that you are not able to parallel connect with ETS, and if you disrupt the tunneling connection (no proper close), you will have to wait the connection timeout to reconnect.

Anyway, good it works now. I will close this issue.

closed 2021-02-06 by bmalinowsky