calimero-project/calimero-server #9: Can't get calimero server to run on Alpine Docker

aschamberger, 2021-03-27:
I installed calimero server on an Alpine Docker on Home Assistant OS via this Dockerfile: https://github.com/aschamberger/hassio-addons/blob/master/calimero-server/Dockerfile

The serial lib is compiled on Alpine to support musl and the Java runtime is reduced to java.base via jlink.

There is an exception thrown during startup:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
<?xml version="1.0" encoding="UTF-8"?>
<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="Calimero KNXnet/IP Server (on Home Assistant OS)">
	<!-- KNXnet/IP search & discovery -->
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />
	<!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
	<serviceContainer activate="true" routing="false" networkMonitoring="true" 
		udpPort="3671" listenNetIf="any">
		<knxAddress type="individual">1.1.100</knxAddress>
		<knxSubnet type="ft12-cemi" medium="tp1">/dev/ttyAMA0</knxSubnet>
		<!-- KNX group address filter applied by the server for this service container (optional) -->
		<groupAddressFilter>
		</groupAddressFilter>
		<!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
		<additionalAddresses>
			<knxAddress type="individual">1.1.101</knxAddress>
			<knxAddress type="individual">1.1.102</knxAddress>
			<knxAddress type="individual">1.1.103</knxAddress>
			<knxAddress type="individual">1.1.104</knxAddress>
			<knxAddress type="individual">1.1.105</knxAddress>
		</additionalAddresses>
	</serviceContainer>
	<!-- Add next service container (optional) -->
</knxServer>
[main] TRACE calimero.xml - resolve /etc/server-config.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] TRACE calimero.xml - resolve jar:file:/opt/calimero/calimero-core-2.5-M1.jar!/properties.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///knx-server-ios.xml
[main] INFO calimero.device.knx-server - could not open file:///knx-server-ios.xml, create resource on closing device (/knx-server-ios.xml (No such file or directory))
[main] INFO calimero.server.knx-server - Calimero KNXnet/IP Server (on Home Assistant OS) v2.5-M1
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNXnet/IP Server (on Home Assistant OS)' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - /dev/ttyAMA0:
	listen on any (port 3671), KNX IP routing disabled
	ft12-cemi /dev/ttyAMA0 connection: TP1 medium, device 1.1.100
[main] DEBUG calimero.server.knx-server - setup ft12-cemi subnet '/dev/ttyAMA0'
type 'stop' to stop the gateway and shutdown the server
request to stop server
[knx-server discovery endpoint] TRACE calimero.server.knx-server - network stack uses IPv4 addresses: unknown
[main] INFO calimero.server.gateway.knx-server - knx-server: user request for shutdown
[knx-server discovery endpoint] TRACE calimero.server.knx-server - found network interfaces veth6c65d7f, vethe7827eb, vethfcf3684, veth0003e0c, vethf61de3e, veth5fed532, hassio [172.30.32.1], docker0 [172.17.0.1], eth0 [192.168.178.72], lo [127.0.0.1]
[knx-server] TRACE calimero.serial - check Java library path /opt/jdk/lib/server:/opt/jdk/lib:/opt/jdk/../lib:/usr/java/packages/lib:/lib:/usr/lib
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint [hassio, docker0, eth0, lo] is up and running
[knx-server] INFO calimero.serial - no serialcom in java.library.path: [/opt/jdk/lib/server, /opt/jdk/lib, /opt/jdk/../lib, /usr/java/packages/lib, /lib, /usr/lib]
[main] INFO calimero.server.knx-server - knx-server discovery endpoint closed
[main] INFO calimero.server.knx-server - knx-server control endpoint /dev/ttyAMA0 closed
[main] DEBUG calimero.device.knx-server - saving interface object server to file:///knx-server-ios.xml
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - using rxtx library for serial port access
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
[knx-server control endpoint /dev/ttyAMA0] TRACE calimero.server.knx-server - /dev/ttyAMA0 control endpoint bound to 172.30.32.1:3671
[knx-server] INFO calimero.serial.ft12:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
[knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - /dev/ttyAMA0 secure mgmt/tunneling connections: optional/optional
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
[knx-server control endpoint /dev/ttyAMA0] WARN calimero.server.knx-server - initialization of knx-server control endpoint /dev/ttyAMA0 failed, retry in 10 seconds
tuwien.auto.calimero.KnxRuntimeException
	at java.base/java.util.concurrent.locks.AbstractQueuedSynchronizer.acquireInterruptibly(Unknown Source)
	at java.base/java.util.concurrent.locks.ReentrantLock.lockInterruptibly(Unknown Source)
	at java.base/java.util.concurrent.ArrayBlockingQueue.poll(Unknown Source)
	at tuwien.auto.calimero.server.knxnetip.TcpLooper.start(TcpLooper.java:87)
	at tuwien.auto.calimero.server.knxnetip.ControlEndpointService.<init>(ControlEndpointService.java:165)
	at tuwien.auto.calimero.server.knxnetip.KNXnetIPServer.lambda$addServiceContainer$0(KNXnetIPServer.java:409)
	at tuwien.auto.calimero.server.knxnetip.LooperTask.run(LooperTask.java:102)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
Caused by: java.lang.InterruptedException
	... 13 more
[knx-server] TRACE calimero.serial.ft12:/dev/ttyAMA0 - send reset to BCU
[knx-server /dev/ttyAMA0 tcp service] INFO calimero.server.knx-server - knx-server /dev/ttyAMA0 tcp service (hassio 172.30.32.1:3671) is up and running
[knx-server] TRACE calimero.serial.ft12:/dev/ttyAMA0 - sending FT1.2 frame, blocking, attempt 1
[knx-server] DEBUG calimero.link./dev/ttyAMA0 - hop count set to 6
[knx-server] DEBUG calimero.server.gateway.knx-server - set maximum APDU length of '/dev/ttyAMA0' to 15
[main] DEBUG calimero.device.knx-server - saving device memory to /knx-server-ios.mem
```

bmalinowsky, 2021-03-27:
Quite likely because you leave stdin attached without a tty? Use `--no-stdin`.
As in `tuwien.auto.calimero.server.Launcher --no-stdin /etc/server-config.xml`

aschamberger, 2021-03-27:
Thanks for the quick reply. That fixed the exception :) 

However I'm still not able to connect from the ETS. ETS says the device rejected the connection.
Also the discovery is not working. Did you run calimero server with Docker before? What do I need to expose network wise besides 3671/udp?
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
<?xml version="1.0" encoding="UTF-8"?>
<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="Hass OS KNXnet/IP Server">
	<!-- KNXnet/IP search & discovery -->
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />
	<!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
	<serviceContainer activate="true" routing="true" networkMonitoring="true" 
		udpPort="3671" listenNetIf="any">
		<knxAddress type="individual">1.1.0</knxAddress>
		<knxSubnet type="ft12-cemi" medium="tp1">/dev/ttyAMA0</knxSubnet>
		<!-- KNX group address filter applied by the server for this service container (optional) -->
		<groupAddressFilter>
		</groupAddressFilter>
		<!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
		<additionalAddresses>
			<knxAddress type="individual">1.1.101</knxAddress>
			<knxAddress type="individual">1.1.102</knxAddress>
			<knxAddress type="individual">1.1.103</knxAddress>
			<knxAddress type="individual">1.1.104</knxAddress>
			<knxAddress type="individual">1.1.105</knxAddress>
		</additionalAddresses>
	</serviceContainer>
	<!-- Add next service container (optional) -->
</knxServer>
[main] TRACE calimero.xml - resolve /etc/server-config.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] TRACE calimero.xml - resolve jar:file:/opt/calimero/calimero-core-2.5-M1.jar!/properties.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///knx-server-ios.xml
[main] INFO calimero.device.knx-server - could not open file:///knx-server-ios.xml, create resource on closing device (/knx-server-ios.xml (No such file or directory))
[main] INFO calimero.server.knx-server - Hass OS KNXnet/IP Server v2.5-M1
[main] INFO calimero.server.knx-server - knx-server 'Hass OS KNXnet/IP Server' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - /dev/ttyAMA0:
	listen on any (port 3671), KNX IP routing multicast group 224.0.23.12
	ft12-cemi /dev/ttyAMA0 connection: TP1 medium, device 1.1.0
[main] DEBUG calimero.server.knx-server - setup ft12-cemi subnet '/dev/ttyAMA0'
[knx-server discovery endpoint] TRACE calimero.server.knx-server - network stack uses IPv4 addresses: unknown
[knx-server discovery endpoint] TRACE calimero.server.knx-server - found network interfaces eth0 [172.30.33.0], lo [127.0.0.1]
[knx-server] TRACE calimero.serial - check Java library path /opt/jdk/lib/server:/opt/jdk/lib:/opt/jdk/../lib:/usr/java/packages/lib:/lib:/usr/lib
[knx-server] INFO calimero.serial - no serialcom in java.library.path: [/opt/jdk/lib/server, /opt/jdk/lib, /opt/jdk/../lib, /usr/java/packages/lib, /lib, /usr/lib]
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint [eth0, lo] is up and running
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - using rxtx library for serial port access
[knx-server routing service 224.0.23.12] DEBUG calimero.knxnetip.KNX IP routing service 224.0.23.12 - join multicast group 224.0.23.12 on 0.0.0.0
[knx-server routing service 224.0.23.12] INFO calimero.knxnetip.KNX IP routing service 224.0.23.12 - multicast loopback mode enabled
[knx-server control endpoint /dev/ttyAMA0] TRACE calimero.server.knx-server - /dev/ttyAMA0 control endpoint bound to 172.30.33.0:3671
[knx-server routing service 224.0.23.12] INFO calimero.server.knx-server - knx-server /dev/ttyAMA0 KNX IP routing service 224.0.23.12 is up and running
[knx-server] INFO calimero.serial.ft12:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
[knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - /dev/ttyAMA0 secure mgmt/tunneling connections: optional/optional
[knx-server] TRACE calimero.serial.ft12:/dev/ttyAMA0 - send reset to BCU
[knx-server] TRACE calimero.serial.ft12:/dev/ttyAMA0 - sending FT1.2 frame, blocking, attempt 1
[knx-server] DEBUG calimero.link./dev/ttyAMA0 - hop count set to 6
[knx-server /dev/ttyAMA0 tcp service] INFO calimero.server.knx-server - knx-server /dev/ttyAMA0 tcp service (eth0 172.30.33.0:3671) is up and running
[knx-server] DEBUG calimero.server.gateway.knx-server - set maximum APDU length of '/dev/ttyAMA0' to 15
[... cut ...]
[knx-server control endpoint /dev/ttyAMA0] DEBUG calimero.server.knx-server - responses use route back control endpoint /192.168.178.29:51971
[knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - send KNXnet/IP description to /192.168.178.29:51971: "Hass OS KNXnet/IP Server", KNX address 1.1.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address 02:42:ac:1e:21:00, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 1.1.101, 1.1.102, 1.1.103, 1.1.104, 1.1.105, KNX manufacturer ID 0x0, data 0x626d323031312020
```

bmalinowsky, 2021-03-27:
KNX discovery uses multicast by default, docker has some difficulties with ipv4 multicasting concerning external networks (from the view of the default docker network). Discovery from other _docker nodes_ works without problems. 

You can see that sending udp/tcp unicast requests outside docker will be answered, as will description requests. 

Running the server container will have to use `--net host` for the docker bridge to forward mcast stuff. 
If the server actually receives a discovery request, you will see a line like 
`DEBUG calimero.server.knx-server - KNXnet/IP discovery: identify as blah blah` 
in the log.

I do see description requests from 192.168.178.29:51971 in your log, so some tool can actually send datagrams to the server and they are answered. The same applies to connection requests.

aschamberger, 2021-03-27:
With host networking it worked and calimero server showed up in ETS :)

However the ETS still didn't connect. The log lines above were also from ETS requests, but they didn't get answered. With the working discovery I saw the server is advertised with the local docker network IPs. This gave me the idea to change `listenNetIf` to `eth0` instead of `any` in the serviceContainer and I got a working connection.

After this I wanted to make a read request in group monitor. The connection dropped immediately. Hardware is a Weinzierl kberry module.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
<?xml version="1.0" encoding="UTF-8"?>
<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="Hass OS KNXnet/IP Server">
	<!-- KNXnet/IP search & discovery -->
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />
	<!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
	<serviceContainer activate="true" routing="true" networkMonitoring="true" 
		udpPort="3671" listenNetIf="eth0">
		<knxAddress type="individual">1.1.0</knxAddress>
		<knxSubnet type="ft12-cemi" medium="tp1">/dev/ttyAMA0</knxSubnet>
		<!-- KNX group address filter applied by the server for this service container (optional) -->
		<groupAddressFilter>
		</groupAddressFilter>
		<!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
		<additionalAddresses>
			<knxAddress type="individual">1.1.101</knxAddress>
			<knxAddress type="individual">1.1.102</knxAddress>
			<knxAddress type="individual">1.1.103</knxAddress>
			<knxAddress type="individual">1.1.104</knxAddress>
			<knxAddress type="individual">1.1.105</knxAddress>
		</additionalAddresses>
	</serviceContainer>
	<!-- Add next service container (optional) -->
</knxServer>
[main] TRACE calimero.xml - resolve /etc/server-config.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] TRACE calimero.xml - resolve jar:file:/opt/calimero/calimero-core-2.5-M1.jar!/properties.xml
[main] TRACE calimero.xml - no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///knx-server-ios.xml
[main] INFO calimero.device.knx-server - could not open file:///knx-server-ios.xml, create resource on closing device (/knx-server-ios.xml (No such file or directory))
[main] INFO calimero.server.knx-server - Hass OS KNXnet/IP Server v2.5-M1
[main] INFO calimero.server.knx-server - knx-server 'Hass OS KNXnet/IP Server' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - /dev/ttyAMA0:
	listen on eth0 (port 3671), KNX IP routing multicast group 224.0.23.12
	ft12-cemi /dev/ttyAMA0 connection: TP1 medium, device 1.1.0
[main] DEBUG calimero.server.knx-server - setup ft12-cemi subnet '/dev/ttyAMA0'
[knx-server discovery endpoint] TRACE calimero.server.knx-server - network stack uses IPv4 addresses: unknown
[knx-server discovery endpoint] TRACE calimero.server.knx-server - found network interfaces veth6c65d7f, vethe7827eb, vethfcf3684, veth0003e0c, vethf61de3e, veth5fed532, hassio [172.30.32.1], docker0 [172.17.0.1], eth0 [192.168.178.72], lo [127.0.0.1]
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint [hassio, docker0, eth0, lo] is up and running
[knx-server] TRACE calimero.serial - check Java library path /opt/jdk/lib/server:/opt/jdk/lib:/opt/jdk/../lib:/usr/java/packages/lib:/lib:/usr/lib
[knx-server] INFO calimero.serial - no serialcom in java.library.path: [/opt/jdk/lib/server, /opt/jdk/lib, /opt/jdk/../lib, /usr/java/packages/lib, /lib, /usr/lib]
[knx-server control endpoint /dev/ttyAMA0] TRACE calimero.server.knx-server - /dev/ttyAMA0 control endpoint bound to 192.168.178.72:3671
[knx-server routing service 224.0.23.12] DEBUG calimero.knxnetip.KNX IP routing service 224.0.23.12 - join multicast group 224.0.23.12 on eth0
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - using rxtx library for serial port access
[knx-server routing service 224.0.23.12] INFO calimero.knxnetip.KNX IP routing service 224.0.23.12 - multicast loopback mode enabled
[knx-server routing service 224.0.23.12] INFO calimero.server.knx-server - knx-server /dev/ttyAMA0 KNX IP routing service 224.0.23.12 is up and running
[knx-server] INFO calimero.serial.ft12:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
[knx-server control endpoint /dev/ttyAMA0] INFO calimero.server.knx-server - /dev/ttyAMA0 secure mgmt/tunneling connections: optional/optional
[knx-server] DEBUG calimero.serial.ft12:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
[knx-server /dev/ttyAMA0 tcp service] INFO calimero.server.knx-server - knx-server /dev/ttyAMA0 tcp service (eth0 192.168.178.72:3671) is up and running
[... cut ...]
[knx-server subnet dispatcher] TRACE calimero.serial.ft12:/dev/ttyAMA0 - sending FT1.2 frame, blocking, attempt 1
[Calimero FT1.2 receiver] WARN calimero.link./dev/ttyAMA0 - negative confirmation of 2/1/10: 2e 00 9d d0 11 66 11 0a 01 00 00
[knx-server subnet dispatcher] DEBUG calimero.server.knxnetip.KNX IP Tunneling 192.168.178.29:51441 - send tunneling-feat.info (channel 1) connection status 00
[knx-server subnet dispatcher] WARN calimero.server.gateway.knx-server - timeout sending to 2/1/10: no confirmation reply received for 2/1/10
``` 

Thanks again.

bmalinowsky, 2021-03-27:
I should add that (at least for me) mcast discovery also only works if i run both client and server on the same host, and the exposed port does not make a difference.

bmalinowsky, 2021-03-27:
Yeah, "any" will pick the default docker network, forgot that, sorry.

Anyway, now the ip connection itself is fine and uses tcp. Two problems I see are that the stick is responding with a negative .con and the server informs about a disconnected bus.

IIRC, ETS disconnects on the latter msg, so temporarily, you can avoid the disconnected info by setting an attribute `udpOnly` equal to "true" (so ETS connects via UDP only) in the serviceContainer element in the server config. The connection should be kept open.

Is the kberry using a usb to serial adapter? Could be that it needs the same workaround as the other usb sticks.

aschamberger, 2021-03-28:
The kberry is this device: https://www.weinzierl.de/index.php/de/alles-knx1/knx-module/knx-baos-modul-838 and directly sits on the raspi uart gpio.

bmalinowsky, 2021-03-28:
I don't know that device in particular, but being a baos device, it has its own device address. From some other device I know that such interface don't like the source address to be changed (or set at all) during a `.req`. This can be the cause of the negative `.con`. In the baos docs at least, the source is not set.

aschamberger, 2021-03-28:
Hi again, the udpOnly didn't keep the connection open.

However I tried to do the same request like in the log above with knxd. With knxd it works.
Here is the knxd log output. Maybe you can confirm your hypothesis:
```
Layer 8 [64:tunnel/1.1.102         803.727] CONNECTIONSTATE_REQUEST on 1
Layer 1 [ 4:server/Server          803.727] Send(002): 01 00
Layer 8 [64:tunnel/1.1.102         803.733] CONNECTIONSTATE_REQUEST on 1
Layer 1 [ 4:server/Server          803.733] Send(002): 01 00
Layer 8 [64:tunnel/1.1.102         804.120] TUNNEL_REQ
Layer 8 [63:tunnel/ConnC           804.120] found addr 1.1.102
Layer 1 [ 4:server/Server          804.120] Send(004): 04 01 01 00
Layer 6 [18:router/ConnC           804.120] sending, send_more clear
Layer 1 [ 4:server/Server          804.120] Send(011): 29 00 BC D0 11 66 11 0A 01 00 00
Layer 2 [20:router.pace_/router    804.120] out 1/2: delay for 0.017 sec
Layer 6 [11:B.interface/Conn       804.120] sending, send_more clear
Layer 6 [18:router/ConnC           804.120] still waiting
Layer 6 [ 1:main                   804.120] wait L
Layer 1 [ 4:server/Server          804.120] Send(015): 04 01 01 00 2E 00 BC E0 00 00 11 0A 01 00 00
Layer 8 [64:tunnel/1.1.102         804.124] TUNNEL_ACK
Layer 1 [14:B.interface/ft12wrap   804.133] Processing(001): E5
Layer 2 [20:router.pace_/router    804.137] delay done
Layer 6 [18:router/ConnC           804.137] sendNext called, send_more set
Layer 6 [18:router/ConnC           804.137] is OK
Layer 6 [63:tunnel/ConnC           804.137] is OK
Layer 6 [11:B.interface/Conn       804.137] still waiting
Layer 1 [14:B.interface/ft12wrap   804.233] Processing(016): 68 0C 0C 68 F3 2E 00 9D D0 FF FF 11 0A 01 00 00
Layer 1 [14:B.interface/ft12wrap   804.233] Processing: left(016): 68 0C 0C 68 F3 2E 00 9D D0 FF FF 11 0A 01 00 00
Layer 1 [14:B.interface/ft12wrap   804.236] Processing(018): 68 0C 0C 68 F3 2E 00 9D D0 FF FF 11 0A 01 00 00 A8 16
Layer 2 [17:B.interface/ft12cemi   804.236] Confirmed
Layer 6 [11:B.interface/Conn       804.236] sendNext called, send_more set
Layer 6 [18:router/ConnC           804.236] is OK
Layer 6 [63:tunnel/ConnC           804.236] is OK
Layer 6 [11:B.interface/Conn       804.236] is OK
Layer 6 [ 4:server/Server          804.236] is OK
Layer 6 [ 1:main                   804.236] OK
Layer 6 [ 2:main/L                 804.236] OK L
```

aschamberger, 2021-03-29:
I quickly read through the BAOS User Guide: https://www.weinzierl.de/images/download/development/830/KnxBAOS_Users_Guide.pdf

Chapter '5.2.2 cEMI in the Application' and '12.2.2 Direct Access to the KNX system' describe the requirement to set the device to link layer mode. I guess this needs to be done to make it work...

bmalinowsky, 2021-03-30:
Link-layer mode is always set for that connection, before any further communication.

In your last trace,  one can see that the source address got cleared and and the weinzierl device confirms with a different address (15.15.255 or FF FF).

Calimero does such address replacement currently only for certain types (usb) where it knows its necessary.
What you can try is, if you change in the server-config.xml the serviceContainer address from 1.1.100 to 15.15.100, and change the first additionalAddresses entry to 15.15.255 (i.e. the weinzierl address), and make sure that a client gets that address assigned. 
This is not a solution, because it defeats the purpose of having multiple client connections, but at least you can see if the weinzierl device accepts its own address.

aschamberger, 2021-03-30:
I tried what you suggested and it didn't work. Besides the address replacement you mentioned I think there needs to be done what is described in the link I posted obove:
```
To send self-made telegrams, the application accesses the link layer of the KNX stack. This is
done by sending the following telegram.
APP: F6 00 08 01 34 10 01 00 M_PropWrite.req
This changes the communication mode to link layer:
KNX BAOS Starter Kit
WEINZIERL ENGINEERING GmbH
Page 46/143
 0xF6 – M_PropWrite.req: Write a value to an internal property
 0x0008 – The property is located in the cEMI server object type
 0x01 – Instance of the object is 1
 0x34 – The property ID is 52, "Communication Mode"
 0x10 – Number of elements (high nibble)
 0x01 – Starting index
 0x00 – Value to write (link layer mode)
To switch the communication mode back to BAOS, the value 0xF0 must be written into the
same property.
```
I couldn't find this in the calimero code and could confirm its set in knxd: https://github.com/knxd/knxd/blob/848a5cc6cc1f3f79116fbf61c825ca9cd47057cd/src/backend/ft12.cpp#L444

bmalinowsky, 2021-03-31:
It's done [here](https://github.com/calimero-project/calimero-core/blob/baf96552fe1a4932c57af8fafa5fe3445d67c43e/src/tuwien/auto/calimero/link/KNXNetworkLinkFT12.java#L155) using the bcu switcher.

aschamberger, 2021-03-31:
This I saw already yesterday, but it sets something different [here](https://github.com/calimero-project/calimero-core/blob/baf96552fe1a4932c57af8fafa5fe3445d67c43e/src/tuwien/auto/calimero/link/BcuSwitcher.java#L374):
```
final byte[] switchLinkLayer = { (byte) peiSwitch_req, 0x00, 0x18, 0x34, 0x56, 0x78, 0x0A, };
```
vs. writing to the property from the doc for the kberry:
```
F6 00 08 01 34 10 01 00 M_PropWrite.req
```

bmalinowsky, 2021-03-31:
You're comparing the service used for EMI2 against the cEMI one, of course it is different. You are creating a link for cEMI, so the service used is created [here](https://github.com/calimero-project/calimero-core/blob/baf96552fe1a4932c57af8fafa5fe3445d67c43e/src/tuwien/auto/calimero/link/BcuSwitcher.java#L356). 

What is actually sent, is decided [here](https://github.com/calimero-project/calimero-core/blob/edfcbde9fc6bdfee6f2a18c14a9e8e95cadb440c/src/tuwien/auto/calimero/link/BcuSwitcher.java#L392).

aschamberger, 2021-04-03:
Sorry for asking stupid questions. 

> Calimero does such address replacement currently only for certain types (usb) where it knows its necessary.
> What you can try is, if you change in the server-config.xml the serviceContainer address [...]

I just tried to connect to my house KNX and it works with the configuration you suggested. I can see all messages and send read requests to unknown GAs and the tunnel stays open.

Before I just connected to a spare KNX PSU without any devices on the bus.

Now the question remains how to enable this address replacement magic that exists for some USB devices also for the kberry.

bmalinowsky, 2021-04-13:
Your question is not stupid.

The most generic way I see to adjust the behavior (without poking the device, trying to set its address etc.) is to add a setting in the config. I did exactly that, adding an attribute to the _subnet_ element _knxAddress_, which statically overrides default source address assignment.

For example, `knxAddress="0"` in your case (Or "0.0.0"). IP clients can then have any address.

You can try the docker image which contains that particular changeset at https://hub.docker.com/r/calimeroproject/knxserver .

Pass the server config via a volume param, something like
`docker run --rm -ti --net host -v /my/dir/server-config.xml:/usr/app/server-config.xml calimeroproject/knxserver server-config.xml`

If this works, I can incorporate it upstream. 
Do you also use the interface for BAOS or to program devices, or just group communication?

aschamberger, 2021-04-13:
> You can try the docker image which contains that particular changeset at https://hub.docker.com/r/calimeroproject/knxserver .

Can I download the jars somewhere? I run it on raspberry pi and would need to incorporate them into my container on arm64.

> Do you also use the interface for BAOS or to program devices, or just group communication?

I'd like to use it for programming devices and group communication. I picked it because of the form factor ready made for raspi GPIO.

bmalinowsky, 2021-04-13:
I put them in a gist:
https://gist.github.com/bmalinowsky/3269cdab752fcff2d7c1e02c68f11084

you can download it pressing the "download zip" button there (it should contain 3 calimero jars). 

(Note that you can also simply copy files directly out of a docker container.)

aschamberger, 2021-04-15:
Hi, thanks for uploading the jars, I just tried them. Works for group communication :)
I also tried to program two devices with ETS. ETS says device didn't respond in time.
Here's the last part of the log before the ETS throws the error:
```
[knx-server subnet dispatcher] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - add to multicast loopback frame buffer: 1.1.101->1.1.74 L_Data.ind, system priority hop count 6, tpdu 80
[knx-server subnet dispatcher] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 10 29 00 b0 60 11 65 11 4a 00 80
[knx-server routing service 224.0.23.12] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - discard multicast loopback cEMI frame: 1.1.101->1.1.74 L_Data.ind, system priority hop count 6, tpdu 80
[knx-server subnet dispatcher] TRACE calimero.server.gateway.knx-server - dispatch to KNX subnet 1.1.0 (/dev/ttyAMA0 in service container '/dev/ttyAMA0')
[knx-server subnet dispatcher] DEBUG calimero.link./dev/ttyAMA0 - send (wait for confirmation) 1.1.0->1.1.74 L_Data.req, system priority hop count 5 repeat, tpdu 80
[knx-server subnet dispatcher] DEBUG calimero.link./dev/ttyAMA0 - send message to 1.1.74, wait for ack
[knx-server subnet dispatcher] TRACE calimero.link./dev/ttyAMA0 - EMI 11 00 b0 50 11 00 11 4a 00 80
[knx-server subnet dispatcher] TRACE calimero.serial.ft12:/dev/ttyAMA0 - sending FT1.2 frame, blocking, attempt 1
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - confirmation of 1.1.74
[knx-server subnet dispatcher] TRACE calimero.link./dev/ttyAMA0 - send to 1.1.74 succeeded
[knx-server subnet dispatcher] TRACE calimero.server.gateway.knx-server - server-side KNX IP LinkLayer 192.168.178.29:60209 (channel 2, 1.1.101): 1.1.101->1.1.74 L_Data.req, system priority hop count 6 repeat, tpdu 43 00: T_Connected seq 0, A_Device-desc.read
[ForkJoinPool.commonPool-worker-3] TRACE calimero.server.gateway.knx-server - send positive cEMI L_Data.con
[knx-server subnet dispatcher] DEBUG calimero.server.gateway.knx-server - dispatch 1.1.101->1.1.74 using KNX IP routing service 224.0.23.12
[ForkJoinPool.commonPool-worker-3] TRACE calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:60209 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 2) 06 10 04 20 00 15 04 02 00 00 2e 00 b0 60 11 65 11 4a 01 43 00
[knx-server subnet dispatcher] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - add to multicast loopback frame buffer: 1.1.101->1.1.74 L_Data.ind, system priority hop count 6, tpdu 43 00
[knx-server subnet dispatcher] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 11 29 00 b0 60 11 65 11 4a 01 43 00
[knx-server subnet dispatcher] TRACE calimero.server.gateway.knx-server - dispatch to KNX subnet 1.1.0 (/dev/ttyAMA0 in service container '/dev/ttyAMA0')
[knx-server subnet dispatcher] DEBUG calimero.link./dev/ttyAMA0 - send (wait for confirmation) 1.1.0->1.1.74 L_Data.req, system priority hop count 5 repeat, tpdu 43 00
[knx-server subnet dispatcher] DEBUG calimero.link./dev/ttyAMA0 - send message to 1.1.74, wait for ack
[knx-server subnet dispatcher] TRACE calimero.link./dev/ttyAMA0 - EMI 11 00 b0 50 11 00 11 4a 01 43 00
[knx-server subnet dispatcher] TRACE calimero.serial.ft12:/dev/ttyAMA0 - sending FT1.2 frame, blocking, attempt 1
[knx-server routing service 224.0.23.12] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - discard multicast loopback cEMI frame: 1.1.101->1.1.74 L_Data.ind, system priority hop count 6, tpdu 43 00
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - confirmation of 1.1.74
[knx-server subnet dispatcher] TRACE calimero.link./dev/ttyAMA0 - send to 1.1.74 succeeded
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - indication 1.1.56->2/1/2 L_Data.ind, low priority hop count 6, tpdu 00 80 07 ee
[knx-server] TRACE calimero.server.gateway.knx-server - subnet /dev/ttyAMA0: 1.1.56->2/1/2 L_Data.ind, low priority hop count 6, tpdu 00 80 07 ee: T_Group, A_Group.write
[knx-server] DEBUG calimero.server.gateway.knx-server - dispatch 1.1.56->2/1/2 to all server-side connections
[knx-server] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - add to multicast loopback frame buffer: 1.1.56->2/1/2 L_Data.ind, low priority hop count 5, tpdu 00 80 07 ee
[knx-server] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 13 29 00 bc d0 11 38 11 02 03 00 80 07 ee
[knx-server routing service 224.0.23.12] TRACE calimero.server.knxnetip.KNX IP routing service 224.0.23.12 - discard multicast loopback cEMI frame: 1.1.56->2/1/2 L_Data.ind, low priority hop count 5, tpdu 00 80 07 ee
[knx-server] TRACE calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:60209 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 2) 06 10 04 20 00 17 04 02 00 00 29 00 bc d0 11 38 11 02 03 00 80 07 ee
[knx-server] TRACE calimero.server.gateway.knx-server - dispatching to KNX subnets: exclude subnet /dev/ttyAMA0
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - indication 1.1.56->2/7/3 L_Data.ind, low priority hop count 6, tpdu 00 80 14 88
[... cut ...]
[knx-server /dev/ttyAMA0 tcp connection /192.168.178.29:60209] DEBUG calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:60209 - close connection for channel 2 - client request
[knx-server /dev/ttyAMA0 tcp connection /192.168.178.29:60209] DEBUG calimero.server.gateway.knx-server - removed connection KNX IP LinkLayer 192.168.178.29:60209 (client request)
[/dev/ttyAMA0 data endpoint 192.168.178.29:60209] INFO calimero.server.knx-server - /dev/ttyAMA0 data endpoint 192.168.178.29:60209 closed
[knx-server /dev/ttyAMA0 tcp connection /192.168.178.29:60209] INFO calimero.server.knx-server - close tcp connection to /192.168.178.29:60209
```

bmalinowsky, 2021-04-16:
Thanks for also trying programming.
I found the issue I see in your log; an updated server jar is attached (github doesn't allow .jar, so I renamed .jar -> .zip).

[calimero-server-2.5-M1.zip](https://github.com/calimero-project/calimero-server/files/6327972/calimero-server-2.5-M1.zip)

aschamberger, 2021-04-17:
Thanks for looking into it. Tried the updated server jar, but it didn't work :(
Does this partial trace help already or do you need more?

```
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x43 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x35 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x16 - ignored
[Calimero FT1.2 receiver] WARN calimero.serial.ft12:/dev/ttyAMA0 - toggle frame count bit
[knx-server] DEBUG calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:62434 - send tunneling-feat.info connection status 00
[knx-server] WARN calimero.server.gateway.knx-server - timeout sending to 15.15.255: no confirmation reply received for 15.15.255
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - confirmation of 15.15.255
[Calimero link notifier] DEBUG calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:62434 - send tunneling-feat.info connection status 01
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - indication 1.1.56->2/1/2 L_Data.ind, low priority hop count 6, tpdu 00 80 07 bc
[knx-server] TRACE calimero.server.gateway.knx-server - subnet /dev/ttyAMA0: 1.1.56->2/1/2 L_Data.ind, low priority hop count 6, tpdu 00 80 07 bc: T_Group, A_Group.write
[knx-server] DEBUG calimero.server.gateway.knx-server - dispatch 1.1.56->2/1/2 to all server-side connections
[knx-server] TRACE calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:62434 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 4) 06 10 04 20 00 17 04 04 00 00 29 00 bc d0 11 38 11 02 03 00 80 07 bc
[knx-server] TRACE calimero.server.gateway.knx-server - dispatching to KNX subnets: exclude subnet /dev/ttyAMA0
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - indication 1.1.56->2/7/3 L_Data.ind, low priority hop count 6, tpdu 00 80 14 8f
[knx-server] TRACE calimero.server.gateway.knx-server - subnet /dev/ttyAMA0: 1.1.56->2/7/3 L_Data.ind, low priority hop count 6, tpdu 00 80 14 8f: T_Group, A_Group.write
[knx-server] DEBUG calimero.server.gateway.knx-server - dispatch 1.1.56->2/7/3 to all server-side connections
[knx-server] TRACE calimero.server.knxnetip.KNX IP LinkLayer 192.168.178.29:62434 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 4) 06 10 04 20 00 17 04 04 00 00 29 00 bc d0 11 38 17 03 03 00 80 14 8f
[knx-server] TRACE calimero.server.gateway.knx-server - dispatching to KNX subnets: exclude subnet /dev/ttyAMA0
[Calimero FT1.2 receiver] WARN calimero.serial.ft12:/dev/ttyAMA0 - invalid frame, discarded 14 bytes: 16 68 f3 29 00 b0 60 11 4a ff ff 0b 4b d6 00 00 00 00 00 00 00 00 00 00 00 00
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0xb - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x21 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x43 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x35 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x16 - ignored
[Calimero FT1.2 receiver] WARN calimero.serial.ft12:/dev/ttyAMA0 - invalid frame, discarded 14 bytes: 16 68 f3 29 00 b0 60 11 4a ff ff 0b 4b d6 00 00 00 00 00 00 00 00 00 00 00 00
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0xb - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x21 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x43 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x35 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x16 - ignored
[Calimero FT1.2 receiver] WARN calimero.serial.ft12:/dev/ttyAMA0 - invalid frame, discarded 14 bytes: 16 68 f3 29 00 b0 60 11 4a ff ff 0b 4b d6 00 00 00 00 00 00 00 00 00 00 00 00
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0xb - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x0 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x21 - ignored
[... cut ...]
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x43 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x15 - ignored
[Calimero FT1.2 receiver] TRACE calimero.serial.ft12:/dev/ttyAMA0 - received unexpected start byte 0x16 - ignored
[Calimero FT1.2 receiver] DEBUG calimero.link./dev/ttyAMA0 - confirmation of 1.1.74
[knx-server subnet dispatcher] TRACE calimero.link./dev/ttyAMA0 - send to 1.1.74 succeeded
```

bmalinowsky, 2021-04-18:
This is a different problem it seems. The serial input cuts up the frame. I switched to a better blocking read. As far as I can test, this always avoids the bunch of 0s (on the off-chance that it now also blocks on a garbage frame. Will need to refine that).

[calimero-core-2.5-M1.zip](https://github.com/calimero-project/calimero-server/files/6331314/calimero-core-2.5-M1.zip)

bmalinowsky, 2021-04-18:
The above + a few sanity checks to not block unnecessarily.

[calimero-core-2.5-M1.zip](https://github.com/calimero-project/calimero-server/files/6331647/calimero-core-2.5-M1.zip)

aschamberger, 2021-04-18:
Just tried the updated core jar for programming. And it works :)
Group communication also works like before.

Thanks again.

closed 2021-05-13 by bmalinowsky