calimero-project/calimero-server #13: Docker images for Raspbian

MMartinezV, 2022-01-16:
Hello,

I've seen that the published docker images on Docker Hub are only for amd64. It would be great to have docker images for arm/v7.

Would you consider to publish an official calimero-server image for that platform?

Regards,

  Manuel Martínez

calimero-project, 2022-02-12:
Hello, 
Docker Hub contains now an armv7 image for the server 2.6-SNAPSHOT, give it a try.

closed 2022-02-28 by calimero-project

MMartinezV, 2022-03-20:
Hello,
Thanks for your work. It didn't work well at first attempt and last month I've been very busy. Today I've spent some time testing the docker image and it works but there is a problem when starting the container as a daemon.

When it starts in daemon mode (docker run -d or with docker-compose), the launcher usually receives a "stop" message and it shutdowns. I don't know why.

If I start the launcher manually (docker run -it... and then run a startup script) it works fine. 
<code><pre>
[main] INFO calimero.server.knx-server - Calimero KNXnet/IP Svr v2.6-SNAPSHOT
knxd_1         | [main] INFO calimero.server.knx-server - knx-server 'Calimero KNXnet/IP Svr' - 1 service container, discovery: listen on [all] send on [all]
knxd_1         | [main] INFO calimero.server.knx-server - jung:
knxd_1         | 	listen on eth0 (port 3671), KNX IP routing multicast group 224.0.23.12
knxd_1         | 	usb jung connection: TP1 medium, device 7.1.0
knxd_1         | [main] INFO calimero.server.gateway.knx-server - activate 'jung' disruption buffer on ports [5555-5559], disruption timeout 30 s
knxd_1         | **type 'stop' to stop the gateway and shutdown the server**
knxd_1         | **request to stop server**
knxd_1         | [main] INFO calimero.server.gateway.knx-server - knx-server: **user request for shutdow**n
knxd_1         | [main] INFO calimero.server.knx-server - knx-server discovery endpoint closed
knxd_1         | [main] INFO calimero.server.knx-server - knx-server control endpoint jung closed
knxd_1         | [main] INFO calimero.server.knx-server - knx-server routing service 224.0.23.12 closed
</pre></code>

I'm using the same server-config.xml that I use with the non-docker calimero-server, and when it starts normally it works fine, so the problem doesn't seem to be there.

It's easier to reproduce the problem using docker run instead of docker-compose. This is the command I use to test it:

<pre><code>
# The one that works:
docker run -it --network host -v `pwd`/calimero-server/config/server-config.xml:/usr/app/server-config.xml -v `pwd`/calimero-server/bin/calimero-server.sh:/usr/app/calimero-server.sh --privileged -v /dev/bus/usb:/dev/bus/usb --entrypoint /bin/sh calimeroproject/knxserver:2.6-SNAPSHOT calimero-server.sh
# The one that fails:
docker run -d --network host -v `pwd`/calimero-server/config/server-config.xml:/usr/app/server-config.xml -v `pwd`/calimero-server/bin/calimero-server.sh:/usr/app/calimero-server.sh --privileged -v /dev/bus/usb:/dev/bus/usb --entrypoint /bin/sh calimeroproject/knxserver:2.6-SNAPSHOT calimero-server.sh
</code></pre> 
As you can see the only difference is start it with -d or with -it.

Calimero-server.sh is just a wrapper that execs (to ensure that there's nothing wrong on the original startup script):
<pre><code>
/usr/lib/jvm/jdk/bin/java -classpath /usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-server-2.6-SNAPSHOT.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-device-2.6-SNAPSHOT.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-rxtx-2.6-SNAPSHOT.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-core-2.6-SNAPSHOT.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/slf4j-simple-1.7.30.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/slf4j-api-1.7.30.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/usb4java-javax-1.3.0.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/nrjavaserial-5.2.1.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/usb-api-1.0.2.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/usb4java-1.3.0.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/commons-lang3-3.8.1.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/commons-net-3.3.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-linux-x86.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-linux-x86-64.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-win32-x86.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-win32-x86-64.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-darwin-x86-64.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-linux-arm.jar:/usr/app/calimero-server-2.6-SNAPSHOT/lib/libusb4java-1.3.0-linux-aarch64.jar tuwien.auto.calimero.server.Launcher server-config.xml
</code></pre>

I have done the tests on a Raspberry Pi 4 with Raspbian 10 and with the same Raspberry 4 and Raspbian 11. Is the same raspberry that I use with a non docker knxserver 2.6 and OpenHAB.

I hope it helps.

If you find time to fix the problem I would test it again.

Regards,

  Manuel Martínez

bmalinowsky, 2022-03-20:
Hello,

the server reads by default from std in (e.g., to receive _stop_). If you want to run as daemon without a terminal, you have to pass the option `--no-stdin` before the server-config. I think this should solve the problem.

MMartinezV, 2022-03-20:
Yes, it works.

Thanks lots! Can you add that flag to your official image?

Best regards,
  Manuel Martínez

bmalinowsky, 2022-03-20:
Done, thanks for the suggestion!