calimero-project/calimero-server #14: What is the latest version of calimero-server?

lukasvermeer, 2023-08-25:
Hello,

I am running the `calimeroproject/knxserver:latest-armv7` tag from [Dockerhub](https://hub.docker.com/r/calimeroproject/knxserver/tags). The logs tell me that this is "Calimero KNX IP Server v2.6-SNAPSHOT". However this Github project suggests [the latest release is actually 2.5.1](https://github.com/calimero-project/calimero-server/releases), which seems contradictory. What is the latest release? Is the Dockerhub image up to date?

I am also facing some issues getting the server up and connecting to it from ETS, but I want to make sure I am running the latest version before I open any issues about those problems. Thanks.

Best regards,
Lukas

bmalinowsky, 2023-08-28:
Hello,
2.5.1 is the latest release, but it's not on Dockerhub. The 2.6 snapshot image is newer than the 2.5 one.

What are the issues you have?

lukasvermeer, 2023-09-01:
Clear. Thanks! 👍 

I guess that means the 2.6 snapshot is ahead of the latest "official" release. When I encounter problems, should I use the snapshot or try to revert to the latest release instead?

I have opened a separate issue #15 for the specific problems I have encountered, because that is a different question and so that it is easier to find in case someone else encounters the same problems.

lukasvermeer, 2023-09-04:
Super clear. Thank you. I will keep using the Dockerhub armv7 snapshot.

closed 2023-09-04 by lukasvermeer