calimero-project/calimero-server #15: no property ID 82 in Device Object, wrong communication medium setting, and unknown service type 0x203

lukasvermeer, 2023-09-01:
I am running the armv7 2.6 snapshot inside Docker on a Raspberry Pi. I have an Albrecht Jung KNX Funk-USB Stick plugged into the device. I am running ETS6 on a (virtualised) Windows machine elsewhere in the same network.

server-config.xml:
```
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
  <discovery listenNetIf="all" outgoingNetIf="all" activate="true" />

  <serviceContainer activate="true" routing="true" networkMonitoring="true" udpPort="3671" listenNetIf="any" reuseCtrlEP="false">
    <knxAddress type="individual">1.1.0</knxAddress>
    <knxSubnet type="ip" useNat="true">192.168.188.64:3671</knxSubnet>
    <routing/>
    <!-- latencyTolerance specifies the time window for accepting secure multicasts, 
      depends on max. end-to-end network latency (usually [500 .. 8000] ms) -->
    <!-- <routing latencyTolerance="1000">224.0.23.12</routing> -->

    <!-- Enabling the disruption buffer will replay missed frames after reconnecting a KNXnet/IP client link 
      using the specific client UDP port range (if caused by a disrupted connection). -->
    <!-- <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" /> -->

    <!-- Specify one KNX subnet type, connecting the KNX network. The subnet connection type is one of 
      { "ip", "tcp", "knxip", "ft12", "usb", "tpuart", "user-supplied", "virtual", "emulate" }. 
      The KNX communication medium is one of { "tp1" (default), "pl110", "knxip", "rf" }. 
      In the follwing are some examples: -->

    <!-- <knxSubnet type="ip">192.168.10.10:3671</knxSubnet> -->
    <!-- two optional attributes are supported for tunneling: 
      * netif: specifies the server network interface to the knx subnet
      * useNat="true"|"false": use network address translation (NAT) -->
    <!-- <knxSubnet type="ip" netif="eth0" useNat="true">192.168.10.10:3671</knxSubnet> -->

    <knxSubnet type="usb" medium="rf" knxAddress="1.1.0">Jung</knxSubnet>
    <!-- <knxSubnet type="usb" medium="rf" knxAddress="1.1.0" domainAddress="00FA"></knxSubnet> -->
    <!-- domainAddress="00FA:94133492" -->

    <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
    <!--
    <additionalAddresses>
      <knxAddress type="individual">1.1.10</knxAddress>
      <knxAddress type="individual">1.1.11</knxAddress>
      <knxAddress type="individual">1.1.12</knxAddress>
      <knxAddress type="individual">1.1.13</knxAddress>
    </additionalAddresses>
    -->
  </serviceContainer>
</knxServer>
```

Docker command:
```
$> docker run --rm -ti --net host -v /mnt/Blue/docker/calimero/server-config.xml:/usr/app/server-config.xml --privileged -v /dev/bus/usb:/dev/bus/usb calimeroproject/knxserver:latest-armv7 server-config.xml 
```

Output:
```
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.6-SNAPSHOT
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - Jung:
	listen on any (port 3671), KNX IP routing multicast group 224.0.23.12
	usb Jung connection: RF medium, device 1.1.0 domain 0x000000000000 s/n 0000:00000000
[main] WARN calimero.device.knx-server - reading DoA
tuwien.auto.calimero.device.ios.KnxPropertyException: no property ID 82 in Device Object (index 0), void DP (read/write access to nonexistent property)
	at tuwien.auto.calimero.device.ios.InterfaceObject.getProperty(InterfaceObject.java:277)
	at tuwien.auto.calimero.device.ios.DeviceObject.get(DeviceObject.java:98)
	at tuwien.auto.calimero.device.ios.DeviceObject.domainAddress(DeviceObject.java:89)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.domainAddress(KnxDeviceServiceLogic.java:786)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.setDevice(KnxDeviceServiceLogic.java:147)
	at tuwien.auto.calimero.device.BaseKnxDevice.setDeviceLink(BaseKnxDevice.java:363)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway.<init>(KnxServerGateway.java:758)
	at tuwien.auto.calimero.server.Launcher.run(Launcher.java:610)
	at tuwien.auto.calimero.server.Launcher.main(Launcher.java:583)
type 'stop' to stop the gateway and shutdown the server
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint [veth7838cbf, vethae96a8f, vetha7c78bd, vethd1ac9d3, br-f520d88a05d3, docker0, eth0, lo] is up and running
[knx-server routing service 224.0.23.12] INFO calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671 - multicast loopback mode enabled
[knx-server control endpoint Jung] INFO calimero.server.knx-server - Jung secure mgmt/tunneling connections: optional/optional
[knx-server routing service 224.0.23.12] INFO calimero.server.knx-server - knx-server Jung KNX IP routing service 224.0.23.12 is up and running
[knx-server Jung tcp service] INFO calimero.server.knx-server - knx-server Jung tcp service (veth7838cbf 169.254.10.51:3671) is up and running
[knx-server control endpoint Jung] INFO calimero.server.knx-server - knx-server Jung control endpoint (veth7838cbf 169.254.10.51:3671) is up and running
[knx-server] INFO calimero.usb - Found 1 KNX USB devices:
 |--Bus 001 Device 012: ID 135e:0123
 |   Albrecht Jung KNX Funk-USB Stick (Albrecht Jung)
 |   S/N 0072FB0021DD
[knx-server] INFO calimero.usb.Jung - Bus 001 Device 012: ID 135e:0123
Albrecht Jung KNX Funk-USB Stick (Albrecht Jung)
S/N 0072FB0021DD
[knx-server] INFO calimero.link.Jung - Device Descriptor (Mask Version) 2311 - RF USB interface v2
[knx-server] WARN calimero.link.Jung - wrong communication medium setting: using TP1 to access RF medium
```

That `no property ID 82 in Device Object` looks like a warning rather than an error, so perhaps it can be ignored. That `wrong communication medium setting` is also a warning. Not sure why that appears, since the server config does not mention TP anywhere.

When I try to set up an IP Tunneling connection from ETS6, I get the following output in Docker:
```
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:37142 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:43605 with unknown service type 0x203 - ignored
```

And ETS complains that it `Failed to read device description from 192.168.188.64:3671, but Ping succeeded`.

I have also tried to activate `reuseCtrlEP` like so:
```
  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="any" reuseCtrlEP="true">
```

The output is the same as above. Unknown service type 0x203 ignored.

What am I doing wrong?

bmalinowsky, 2023-09-04:
Some observations:

* The line `<knxSubnet type="ip" useNat="true">192.168.188.64:3671</knxSubnet>` is not wanted, because it specifies a IP subnet connection. It shouldn't matter because you specify the correct one afterwards.
* If you don't use `reuseCtrlEndpoint=true`, then you should have the additionalAddresses uncommented with some KNX addresses in there which aren't used in the KNX network.
* The first exception _no property ID 82 in ..._ is expected, because it's the first start and the server checks its properties whether a domain address is in there. I could probably mute that somehow. If you restart the server, you won't get that exception (unless you run it as 'clean' Docker image, where it won't have that saved resource between starts)
* The _wrong communication medium setting ..._ is because you use `knxAddress="1.1.0"` in your settings, which I initially wrote for TP1 USB sticks, So, it's just a warning when the USB link is set up and checked, but the warning in itself makes sense. Last time I checked, RF USB sticks from Jung need the same workaround.

Now to the mysterious
`[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:37142 with unknown service type 0x203 - ignored`

This is a message from the _discovery_ endpoint, which listens to _multicast_ traffic (default 224.0.23.12), tunneling is a _unicast_ service. Service type 0x203 is the description request. Those requests should be sent to a unicast address.

Can you check if address `192.168.188.64` is the external docker address of the running docker calimero instance which listens to address `veth7838cbf 169.254.10.51:3671`?
The ping succeeds because someone is answering, but it's not the calimero server itself (it doesn't do ping).

bmalinowsky, 2023-09-04:
I now tested a virtualized ETS 6.0.6, and the armv7 docker instance with the same config file and command line as you (differing in absolute config path to config and not a Jung but Weinzierl interface):

* ETS does detect the calimero server properly
* when I use the ETS drop down in the upper right corner for server selection combo box, as long as the drop down menu is open, I get calimero warnings that the _additional addresses_ property  is not set (expected, written before)
* otherwise, when I use ETS _Device Info_ in the Diagnostics Panel with `1.1.0` I do get the server device info both with tunneling and "Ethernet" (i.e. ETS multicast)
* I can start group or bus monitor and get events.

What did you click that ETS does a _description request_?

lukasvermeer, 2023-09-04:
Thank you for your detailed responses. Very helpful.

> Can you check if address 192.168.188.64 is the external docker address of the running docker calimero instance which listens to address veth7838cbf 169.254.10.51:3671?

I'm sorry but I do not know how to check that directly, only indirectly. :-/ Before I start the docker container, there is nothing listening on 3671.

```
netstat -anp|grep "3671"
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
```

After I start it, there is.

```
netstat -anp|grep "3671"
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp6       0      0 169.254.10.51:3671      :::*                    LISTEN      -                   
udp6       0      0 169.254.10.51:3671      :::*                                -                   
udp6       0      0 :::3671                 :::*                                -  
```

I am using `--net host`, so I think this should work.

lukasvermeer, 2023-09-04:
> I now tested a virtualized ETS 6.0.6, and the armv7 docker instance with the same config file and command line as you (differing in absolute config path to config and not a Jung but Weinzierl interface)

To confirm: you changed nothing about my config file, only the absolute config path?

> ETS does detect the calimero server properly

Are these two on the same physical host machine? For me they are on different hosts. I am starting to wonder whether my home network is blocking this traffic somehow. 🤔

lukasvermeer, 2023-09-04:
> What did you click that ETS does a description request?

I manually added a IP Tunneling connection in the Manage Configured Connections screen and entered the ip of the machine running the Docker container.

bmalinowsky, 2023-09-04:
I tested now with the latest ETS 6.1.0, getting the same behavior.

> I understand now that ETS should be able to discover the Calimero instance without manually configuring it.

Yes, and when I add the tunneling connection manually I also see the description request coming in. Everytime you click the "Test" Button ETS sends a description request.

If you start the Calimero Docker instance, on the command line you can add `-vvv` after `... latest-armv7`, it should show log traces, maybe there is more.

Also, if you _don't_ uncomment the "additionalAddresses" part of the server config, ETS will start to complain it can't open new connections: the calimero server doesn't provide more addresses for tunneling.

lukasvermeer, 2023-09-04:
> No, one is a raspberry with Docker, the other is a usual desktop PC running a Windows 10 VM.

That sounds very similar to my setup.

> Yes, and when I add the tunneling connection manually I also see the description request coming in. Everytime you click the "Test" Button ETS sends a description request.

I get no output when I don't configure anything. When I configure the tunnel like so:

[image]

I get (I removed the comments for additionalAddresses, and added -vvv as you suggested):

```
docker run --rm -ti --net host -v /mnt/Blue/docker/calimero/server-config.xml:/usr/app/server-config.xml --privileged -v /dev/bus/usb:/dev/bus/usb calimeroproject/knxserver:latest-armv7 -vvv server-config.xml
[main] TRACE calimero.xml - resolve server-config.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] TRACE calimero.xml - resolve jar:file:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-core-2.6-SNAPSHOT.jar!/properties.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.6-SNAPSHOT
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [all] send on [all]
[main] INFO calimero.server.knx-server - Jung:
	listen on any (port 3671), KNX IP routing disabled
	usb Jung connection: RF medium, device 1.1.0 domain 0x000000000000 s/n 0000:00000000
[main] DEBUG calimero.server.knx-server - setup usb subnet 'Jung'
[main] WARN calimero.device.knx-server - reading DoA
tuwien.auto.calimero.device.ios.KnxPropertyException: no property ID 82 in Device Object (index 0), void DP (read/write access to nonexistent property)
	at tuwien.auto.calimero.device.ios.InterfaceObject.getProperty(InterfaceObject.java:277)
	at tuwien.auto.calimero.device.ios.DeviceObject.get(DeviceObject.java:98)
	at tuwien.auto.calimero.device.ios.DeviceObject.domainAddress(DeviceObject.java:89)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.domainAddress(KnxDeviceServiceLogic.java:786)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.setDevice(KnxDeviceServiceLogic.java:147)
	at tuwien.auto.calimero.device.BaseKnxDevice.setDeviceLink(BaseKnxDevice.java:363)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway.<init>(KnxServerGateway.java:758)
	at tuwien.auto.calimero.server.Launcher.run(Launcher.java:610)
	at tuwien.auto.calimero.server.Launcher.main(Launcher.java:583)
[main] TRACE calimero.device.knx-server - sync Addresstable Object from address 0x116
[main] TRACE calimero.device.knx-server - sync Associationtable Object from address 0x1000
[main] TRACE calimero.device.knx-server - sync Group Object Table Object from address 0x3000
type 'stop' to stop the gateway and shutdown the server
[knx-server discovery endpoint] TRACE calimero.server.knx-server - network stack uses IPv4 addresses: unknown
[knx-server discovery endpoint] TRACE calimero.server.knx-server - found network interfaces veth7838cbf [169.254.10.51], vethae96a8f [169.254.161.216], vetha7c78bd [169.254.198.156], br-f520d88a05d3 [172.18.0.1], docker0 [172.17.0.1], eth0 [192.168.188.64], lo [127.0.0.1]
[knx-server discovery endpoint] INFO calimero.server.knx-server - knx-server discovery endpoint [veth7838cbf, vethae96a8f, vetha7c78bd, br-f520d88a05d3, docker0, eth0, lo] is up and running
[knx-server control endpoint Jung] TRACE calimero.server.knx-server - Jung control endpoint bound to 169.254.10.51:3671
[knx-server control endpoint Jung] INFO calimero.server.knx-server - Jung secure mgmt/tunneling connections: optional/optional
[knx-server Jung tcp service] INFO calimero.server.knx-server - knx-server Jung tcp service (veth7838cbf 169.254.10.51:3671) is up and running
[knx-server control endpoint Jung] INFO calimero.server.knx-server - knx-server Jung control endpoint (veth7838cbf 169.254.10.51:3671) is up and running
[knx-server] DEBUG calimero.usb - Enumerate USB devices
usb4java root hub 1.0.0
 |--Bus 002 Device 001: ID 1d6b:0003
 |   xHCI Host Controller (Linux 5.10.103-v7l+ xhci-hcd)
 |   S/N 0000:01:00.0
 |--Bus 001 Device 001: ID 1d6b:0002
     xHCI Host Controller (Linux 5.10.103-v7l+ xhci-hcd)
     S/N 0000:01:00.0
      |--Bus 001 Device 002: ID 2109:3431
          USB2.0 Hub
           |--Bus 001 Device 004: ID 0bc2:ab24
           |   BUP Slim BL (Seagate)
           |   S/N NA7QR19C
           |--Bus 001 Device 013: ID 135e:0123
               Albrecht Jung KNX Funk-USB Stick (Albrecht Jung)
               S/N 0072FB0021DD
[knx-server] DEBUG calimero.usb - Enumerate USB devices using the low-level API
Bus 2 Device 1: ID 1d6b:0003
    xHCI Host Controller (Linux 5.10.103-v7l+ xhci-hcd)
    Super Speed USB
Bus 1 Device 4: ID 0bc2:ab24
[... 44 lines cut ...]
[usb4java IRP Queue Processor] TRACE calimero.usb.Jung - EP 1 IN I/O request 0113100008000801030000fb00000139100111
[usb4java IRP Queue Processor] DEBUG calimero.usb.Jung - received cEMI frame fb00000139100111
[... cut ...]
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:38716 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:34669 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:36954 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:40506 with unknown service type 0x203 - ignored
```

lukasvermeer, 2023-09-04:
Additional context, in case it is helpful: I also tried using knxd (non-Docker) on the same Pi using the same Windows 11 VM. There I am able to connect, listen to the bus, and send individual writes, but not program devices (i.e. "Download" doesn't work). I opened an issue on that project [here](https://github.com/knxd/knxd/issues/555). I would prefer to use Calimero as it supports Secure.

lukasvermeer, 2023-09-04:
> I tested now with the latest ETS 6.1.0, getting the same behavior.

Thank you for this, btw. I very much appreciate the effort you are putting in to helping me.

bmalinowsky, 2023-09-06:
I rechecked your startup log. You are using `--net host`, but I did miss the following: you are bound to a veth interface (`veth7838cbf`), which is not the one that the outside world sees, and not the one of the host.

Adjust your server config to explicitly use the net interface: `listenNetIf="docker0"`.

lukasvermeer, 2023-09-06:
Thank you for rechecking. I do suspect it's some kind of network issue with Docker. Sadly it does not seem like this particular setting made a difference.

I adjusted server-config.xml like so:

```
$> cat server-config.xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
  <discovery listenNetIf="docker0" outgoingNetIf="all" activate="true" />

  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="any" reuseCtrlEP="true">
    <knxAddress type="individual">1.1.0</knxAddress>
    <knxSubnet type="ip" useNat="true">192.168.188.64:3671</knxSubnet>
    <routing/>
    <!-- latencyTolerance specifies the time window for accepting secure multicasts, 
      depends on max. end-to-end network latency (usually [500 .. 8000] ms) -->
    <!-- <routing latencyTolerance="1000">224.0.23.12</routing> -->

    <!-- Enabling the disruption buffer will replay missed frames after reconnecting a KNXnet/IP client link 
      using the specific client UDP port range (if caused by a disrupted connection). -->
    <!-- <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" /> -->

    <!-- Specify one KNX subnet type, connecting the KNX network. The subnet connection type is one of 
      { "ip", "tcp", "knxip", "ft12", "usb", "tpuart", "user-supplied", "virtual", "emulate" }. 
      The KNX communication medium is one of { "tp1" (default), "pl110", "knxip", "rf" }. 
      In the follwing are some examples: -->

    <!-- <knxSubnet type="ip">192.168.10.10:3671</knxSubnet> -->
    <!-- two optional attributes are supported for tunneling: 
      * netif: specifies the server network interface to the knx subnet
      * useNat="true"|"false": use network address translation (NAT) -->
    <!-- <knxSubnet type="ip" netif="eth0" useNat="true">192.168.10.10:3671</knxSubnet> -->

    <knxSubnet type="usb" medium="rf" knxAddress="1.1.0">Jung</knxSubnet>
    <!-- <knxSubnet type="usb" medium="rf" knxAddress="1.1.0" domainAddress="00FA"></knxSubnet> -->
    <!-- domainAddress="00FA:94133492" -->

    <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
    <additionalAddresses>
      <knxAddress type="individual">1.1.10</knxAddress>
      <knxAddress type="individual">1.1.11</knxAddress>
      <knxAddress type="individual">1.1.12</knxAddress>
      <knxAddress type="individual">1.1.13</knxAddress>
    </additionalAddresses>
  </serviceContainer>
</knxServer>
```

And then start the docker like so:

```
 $> docker run --rm -ti --net host -v /mnt/Blue/docker/calimero/server-config.xml:/usr/app/server-config.xml --privileged -v /dev/bus/usb:/dev/bus/usb calimeroproject/knxserver:latest-armv7 -vvv server-config.xml
[main] TRACE calimero.xml - resolve server-config.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] TRACE calimero.xml - resolve jar:file:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-core-2.6-SNAPSHOT.jar!/properties.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.6-SNAPSHOT
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [docker0] send on [all]
[main] INFO calimero.server.knx-server - Jung:
	listen on any (port 3671), KNX IP routing disabled
	usb Jung connection: RF medium, device 1.1.0 domain 0x000000000000 s/n 0000:00000000
[main] DEBUG calimero.server.knx-server - setup usb subnet 'Jung'
[main] WARN calimero.device.knx-server - reading DoA
tuwien.auto.calimero.device.ios.KnxPropertyException: no property ID 82 in Device Object (index 0), void DP (read/write access to nonexistent property)
	at tuwien.auto.calimero.device.ios.InterfaceObject.getProperty(InterfaceObject.java:277)
	at tuwien.auto.calimero.device.ios.DeviceObject.get(DeviceObject.java:98)
	at tuwien.auto.calimero.device.ios.DeviceObject.domainAddress(DeviceObject.java:89)
[... 105 lines cut ...]

[image]

And here is what I see in the docker logs:

[image]

```
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:42062 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:48254 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:42844 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:44834 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:38254 with unknown service type 0x203 - ignored
[knx-server discovery endpoint] INFO calimero.server.knx-server - received packet from /192.168.188.33:36161 with unknown service type 0x203 - ignored
stop
[... cut ...]
[knx-server control endpoint Jung] DEBUG calimero.server.knx-server - cleanup knx-server control endpoint Jung
[knx-server control endpoint Jung] INFO calimero.server.knx-server - knx-server control endpoint Jung closed
[main] DEBUG calimero.device.knx-server - saving device memory to file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - saving interface object server to file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - using StaX XMLStreamWriter com.sun.xml.internal.stream.writers.XMLStreamWriterImpl
```

bmalinowsky, 2023-09-06:
Please also set the attribute in the  line `<serviceContainer activate="true" ...`. The tunneling requests go to the control endpoint, which is still using a veth interface.

lukasvermeer, 2023-09-06:
Oh apologies I missed one. Results are the same:

```
$> cat server-config.xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
  <discovery listenNetIf="docker0" outgoingNetIf="all" activate="true" />

  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="docker0" reuseCtrlEP="true">
    <knxAddress type="individual">1.1.0</knxAddress>
    <knxSubnet type="ip" useNat="true">192.168.188.64:3671</knxSubnet>
    <routing/>
    <!-- latencyTolerance specifies the time window for accepting secure multicasts, 
      depends on max. end-to-end network latency (usually [500 .. 8000] ms) -->
    <!-- <routing latencyTolerance="1000">224.0.23.12</routing> -->

    <!-- Enabling the disruption buffer will replay missed frames after reconnecting a KNXnet/IP client link 
      using the specific client UDP port range (if caused by a disrupted connection). -->
    <!-- <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" /> -->

    <!-- Specify one KNX subnet type, connecting the KNX network. The subnet connection type is one of 
      { "ip", "tcp", "knxip", "ft12", "usb", "tpuart", "user-supplied", "virtual", "emulate" }. 
      The KNX communication medium is one of { "tp1" (default), "pl110", "knxip", "rf" }. 
      In the follwing are some examples: -->

    <!-- <knxSubnet type="ip">192.168.10.10:3671</knxSubnet> -->
    <!-- two optional attributes are supported for tunneling: 
      * netif: specifies the server network interface to the knx subnet
      * useNat="true"|"false": use network address translation (NAT) -->
    <!-- <knxSubnet type="ip" netif="eth0" useNat="true">192.168.10.10:3671</knxSubnet> -->

    <knxSubnet type="usb" medium="rf" knxAddress="1.1.0">Jung</knxSubnet>
    <!-- <knxSubnet type="usb" medium="rf" knxAddress="1.1.0" domainAddress="00FA"></knxSubnet> -->
    <!-- domainAddress="00FA:94133492" -->

    <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
    <additionalAddresses>
      <knxAddress type="individual">1.1.10</knxAddress>
      <knxAddress type="individual">1.1.11</knxAddress>
      <knxAddress type="individual">1.1.12</knxAddress>
      <knxAddress type="individual">1.1.13</knxAddress>
    </additionalAddresses>
  </serviceContainer>
</knxServer>
```

```
$> docker run --rm -ti --net host -v /mnt/Blue/docker/calimero/server-config.xml:/usr/app/server-config.xml --privileged -v /dev/bus/usb:/dev/bus/usb calimeroproject/knxserver:latest-armv7 -vvv server-config.xml
[main] TRACE calimero.xml - resolve server-config.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] TRACE calimero.xml - resolve jar:file:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-core-2.6-SNAPSHOT.jar!/properties.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.6-SNAPSHOT
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [docker0] send on [all]
[main] INFO calimero.server.knx-server - Jung:
	listen on docker0 (port 3671), KNX IP routing disabled
	usb Jung connection: RF medium, device 1.1.0 domain 0x000000000000 s/n 0000:00000000
[main] DEBUG calimero.server.knx-server - setup usb subnet 'Jung'
[main] WARN calimero.device.knx-server - reading DoA
tuwien.auto.calimero.device.ios.KnxPropertyException: no property ID 82 in Device Object (index 0), void DP (read/write access to nonexistent property)
	at tuwien.auto.calimero.device.ios.InterfaceObject.getProperty(InterfaceObject.java:277)
	at tuwien.auto.calimero.device.ios.DeviceObject.get(DeviceObject.java:98)
	at tuwien.auto.calimero.device.ios.DeviceObject.domainAddress(DeviceObject.java:89)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.domainAddress(KnxDeviceServiceLogic.java:786)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.setDevice(KnxDeviceServiceLogic.java:147)
	at tuwien.auto.calimero.device.BaseKnxDevice.setDeviceLink(BaseKnxDevice.java:363)
	at tuwien.auto.calimero.server.gateway.KnxServerGateway.<init>(KnxServerGateway.java:758)
[... 84 lines cut ...]
[knx-server] TRACE calimero.usb.Jung - sending I/O request 01130f0008000701030000fc000801331001
[usb4java IRP Queue Processor] TRACE calimero.usb.Jung - EP 1 IN I/O request 0113110008000901030000fb0008013310010010
[usb4java IRP Queue Processor] DEBUG calimero.usb.Jung - received cEMI frame fb0008013310010010
[knx-server] WARN calimero.link.Jung - wrong communication medium setting: using TP1 to access RF medium
[knx-server] TRACE calimero.usb.Jung - sending I/O request 01130f0008000701030000fc000001381001
[usb4java IRP Queue Processor] TRACE calimero.usb.Jung - EP 1 IN I/O request 0113110008000901030000fb00000138100100d0
[usb4java IRP Queue Processor] DEBUG calimero.usb.Jung - received cEMI frame fb00000138100100d0
[knx-server] DEBUG calimero.link.Jung - using max. APDU length of 208
[... cut ...]
[knx-server control endpoint Jung] DEBUG calimero.server.knx-server - cleanup knx-server control endpoint Jung
[knx-server control endpoint Jung] INFO calimero.server.knx-server - knx-server control endpoint Jung closed
[main] DEBUG calimero.device.knx-server - saving device memory to file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - saving interface object server to file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - using StaX XMLStreamWriter com.sun.xml.internal.stream.writers.XMLStreamWriterImpl
```

bmalinowsky, 2023-09-06:
:/  Sorry, I just cannot read. The address you want is 192.168.188.64 of course, which is `eth0` not `docker0`.

lukasvermeer, 2023-09-06:
I think that did the trick!

**Long logs below. TL;DR everything seems to work as expected now.**

```
$> cat server-config.xml 
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
  <discovery listenNetIf="eth0" outgoingNetIf="all" activate="true" />

  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="eth0" reuseCtrlEP="true">
    <knxAddress type="individual">1.1.0</knxAddress>
    <knxSubnet type="ip" useNat="true">192.168.188.64:3671</knxSubnet>
    <routing/>
    <!-- latencyTolerance specifies the time window for accepting secure multicasts, 
      depends on max. end-to-end network latency (usually [500 .. 8000] ms) -->
    <!-- <routing latencyTolerance="1000">224.0.23.12</routing> -->

    <!-- Enabling the disruption buffer will replay missed frames after reconnecting a KNXnet/IP client link 
      using the specific client UDP port range (if caused by a disrupted connection). -->
    <!-- <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" /> -->

    <!-- Specify one KNX subnet type, connecting the KNX network. The subnet connection type is one of 
      { "ip", "tcp", "knxip", "ft12", "usb", "tpuart", "user-supplied", "virtual", "emulate" }. 
      The KNX communication medium is one of { "tp1" (default), "pl110", "knxip", "rf" }. 
      In the follwing are some examples: -->

    <!-- <knxSubnet type="ip">192.168.10.10:3671</knxSubnet> -->
    <!-- two optional attributes are supported for tunneling: 
      * netif: specifies the server network interface to the knx subnet
      * useNat="true"|"false": use network address translation (NAT) -->
    <!-- <knxSubnet type="ip" netif="eth0" useNat="true">192.168.10.10:3671</knxSubnet> -->

    <knxSubnet type="usb" medium="rf" knxAddress="1.1.0">Jung</knxSubnet>
    <!-- <knxSubnet type="usb" medium="rf" knxAddress="1.1.0" domainAddress="00FA"></knxSubnet> -->
    <!-- domainAddress="00FA:94133492" -->

    <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
    <additionalAddresses>
      <knxAddress type="individual">1.1.10</knxAddress>
      <knxAddress type="individual">1.1.11</knxAddress>
      <knxAddress type="individual">1.1.12</knxAddress>
      <knxAddress type="individual">1.1.13</knxAddress>
    </additionalAddresses>
  </serviceContainer>
</knxServer>
```

```
$>  docker run --rm -ti --net host -v /mnt/Blue/docker/calimero/server-config.xml:/usr/app/server-config.xml --privileged -v /dev/bus/usb:/dev/bus/usb calimeroproject/knxserver:latest-armv7 -vvv server-config.xml
[main] TRACE calimero.xml - resolve server-config.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] TRACE calimero.xml - resolve jar:file:/usr/app/calimero-server-2.6-SNAPSHOT/lib/calimero-core-2.6-SNAPSHOT.jar!/properties.xml
[main] TRACE calimero.xml - using StaX XMLStreamReader com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl
[main] DEBUG calimero.device.knx-server - loading interface object server from file:///usr/app/knx-server-ios.xml
[main] TRACE calimero.xml - resolve file:///usr/app/knx-server-ios.xml
[main] DEBUG calimero.device.knx-server - no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
[main] INFO calimero.server.knx-server - Calimero KNX IP Server v2.6-SNAPSHOT
[main] INFO calimero.server.knx-server - knx-server 'Calimero KNX IP Server' - 1 service container, discovery: listen on [eth0] send on [all]
[main] INFO calimero.server.knx-server - Jung:
	listen on eth0 (port 3671), KNX IP routing disabled
	usb Jung connection: RF medium, device 1.1.0 domain 0x000000000000 s/n 0000:00000000
[main] DEBUG calimero.server.knx-server - setup usb subnet 'Jung'
[main] WARN calimero.device.knx-server - reading DoA
tuwien.auto.calimero.device.ios.KnxPropertyException: no property ID 82 in Device Object (index 0), void DP (read/write access to nonexistent property)
	at tuwien.auto.calimero.device.ios.InterfaceObject.getProperty(InterfaceObject.java:277)
	at tuwien.auto.calimero.device.ios.DeviceObject.get(DeviceObject.java:98)
	at tuwien.auto.calimero.device.ios.DeviceObject.domainAddress(DeviceObject.java:89)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.domainAddress(KnxDeviceServiceLogic.java:786)
	at tuwien.auto.calimero.device.KnxDeviceServiceLogic.setDevice(KnxDeviceServiceLogic.java:147)
[... 226 lines cut ...]
[knx-server] TRACE calimero.server.gateway.knx-server - subnet Jung: Busmon.ind timestamp 34952 seq 2 (no error): 11 44 ff 02 00 fa 94 13 34 92 a0 9d 00 11 82 00 01 e3 00 80 55 6b
[knx-server] TRACE calimero.server.knxnetip.KNX IP Monitor 192.168.188.33:41016 - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 3) 06 10 04 20 00 29 04 03 00 00 2b 07 03 01 02 04 02 88 88 11 44 ff 02 00 fa 94 13 34 92 a0 9d 00 11 82 00 01 e3 00 80 55 6b
[usb4java IRP Queue Processor] TRACE calimero.usb.Jung - EP 1 IN I/O request 0113270008001f010300002b07030103040289091144ff0200fa94133492a09d0011820002e500803f0d
[usb4java IRP Queue Processor] DEBUG calimero.usb.Jung - received cEMI frame 2b07030103040289091144ff0200fa94133492a09d0011820002e500803f0d
[usb4java IRP Queue Processor] TRACE calimero.link.Jung - Busmon.ind timestamp 35081 seq 3 (no error): 11 44 ff 02 00 fa 94 13 34 92 a0 9d 00 11 82 00 02 e5 00 80 3f 0d
[... cut ...]
[knx-server Jung tcp connection 192.168.188.33:41016] DEBUG calimero.server.knxnetip.KNX IP Monitor 192.168.188.33:41016 - close connection for channel 3 - client request
[knx-server Jung tcp connection 192.168.188.33:41016] DEBUG calimero.server.gateway.knx-server - removed connection KNX IP Monitor 192.168.188.33:41016 (client request)
[knx-server Jung tcp connection 192.168.188.33:41016] INFO calimero.server.knx-server - close tcp connection to 192.168.188.33:41016
```

Then I tried to download config, and it worked too. (Log too long to post to Github.)

Huge success. Thank you very much!

bmalinowsky, 2023-09-12:
If everything works, I think this can be closed?

lukasvermeer, 2023-09-12:
Yes! Thank you for your assistance. Closing.