calimero-project/calimero-server #17: Docker Image for orange pi 5

juergenschorb, 2023-09-17:
Hello,

I'm don't find a docker image for aarch64 (orange pi 5). It would be great to have docker images for aarch64.

Would you consider to publish an official calimero-server image for that platform?

Regards,

Jürgen

bmalinowsky, 2023-09-26:
Hello,

I uploaded an aarch64 image. It would be helpful if you could try it. 
Several caveats apply:

* I don't have a native aarch64 platform, I have to cross-compile it.
* it's the latest snapshot, and I jlinked a liberica Dockerfile, so the size is 27 MB; not great not terrible.
* TP-UART and USB compiled native libs seem to work, as I run it in qemu its's hard to say. I don't know what HW you want to connect.

FYI: The aarch64 Dockerfile is in calimero-docker platform/aarch64.

HTH, Boris

juergenschorb, 2023-09-27:
Hi,

many thanks for your support. I have test your Dockerfile and get some errors (I am doing anything wrong):

- download calimero-docker branch platform/aarch64
- execute build-images.sh and get following error:
root@orangepi5:~/calimero-docker/calimero-docker# ./build-images.sh
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  18.05MB
Step 1/36 : FROM bellsoft/liberica-openjdk-alpine-musl:20 AS BUILDER
20: Pulling from bellsoft/liberica-openjdk-alpine-musl
Digest: sha256:eaa4fcb02ef21cd4d7173ee905c226bb3b00f1210a4bc43f3b106c824bdff315
Status: Downloaded newer image for bellsoft/liberica-openjdk-alpine-musl:20
 ---> 150af75807dd
Step 2/36 : ENV libname=calimero-server
 ---> Using cache
 ---> a6b8db099954
Step 3/36 : ENV libversion=3.0-SNAPSHOT
 ---> Running in 624442447e31
Removing intermediate container 624442447e31
 ---> 0268248b671a
Step 4/36 : COPY ./repos/calimero-core ./calimero-core
 ---> 45fa8726af59
Step 5/36 : COPY ./repos/calimero-device ./calimero-device
 ---> a88d5ddb177c
Step 6/36 : COPY ./repos/calimero-usb ./calimero-usb
 ---> 299c62128f4d
Step 7/36 : COPY ./repos/${libname} ./${libname}
COPY failed: file not found in build context or excluded by .dockerignore: stat repos/calimero-server: file does not exist
root@orangepi5:~/calimero-docker/calimero-docker# cd repos/
root@orangepi5:~/calimero-docker/calimero-docker/repos# ls
calimero-core  calimero-device	calimero-tools	calimero-usb
root@orangepi5:~/calimero-docker/calimero-docker/repos# git clone http://github.com/calimero-project/calimero-server.git
Cloning into 'calimero-server'...
warning: redirecting to https://github.com/calimero-project/calimero-server.git/
remote: Enumerating objects: 7989, done.
remote: Counting objects: 100% (2181/2181), done.
remote: Compressing objects: 100% (394/394), done.
remote: Total 7989 (delta 1097), reused 2170 (delta 1090), pack-reused 5808
Receiving objects: 100% (7989/7989), 1.90 MiB | 4.71 MiB/s, done.
Resolving deltas: 100% (3625/3625), done.
root@orangepi5:~/calimero-docker/calimero-docker/repos# cd ..
root@orangepi5:~/calimero-docker/calimero-docker# ./build-images.sh
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  20.91MB
Step 1/36 : FROM bellsoft/liberica-openjdk-alpine-musl:20 AS BUILDER
 ---> 150af75807dd
Step 2/36 : ENV libname=calimero-server
 ---> Using cache
 ---> a6b8db099954
Step 3/36 : ENV libversion=3.0-SNAPSHOT
 ---> Using cache
 ---> 0268248b671a
Step 4/36 : COPY ./repos/calimero-core ./calimero-core
 ---> Using cache
 ---> 45fa8726af59
Step 5/36 : COPY ./repos/calimero-device ./calimero-device
 ---> Using cache
 ---> a88d5ddb177c
Step 6/36 : COPY ./repos/calimero-usb ./calimero-usb
 ---> Using cache
 ---> 299c62128f4d
Step 7/36 : COPY ./repos/${libname} ./${libname}
 ---> 59a2935293fd
Step 8/36 : RUN cd ${libname} && ./gradlew build
 ---> Running in 05db8d8867d0
Downloading https://services.gradle.org/distributions/gradle-8.3-bin.zip
............10%............20%.............30%............40%.............50%............60%.............70%............80%.............90%............100%

Welcome to Gradle 8.3!

Here are the highlights of this release:
 - Faster Java compilation
 - Reduced memory usage
 - Support for running on Java 20

For more details see https://docs.gradle.org/8.3/release-notes.html

Starting a Gradle Daemon (subsequent builds will be faster)

FAILURE: Build failed with an exception.

* Where:
Build file '/calimero-server/build.gradle' line: 136

* What went wrong:
A problem occurred evaluating root project 'calimero-server'.
> Failed to calculate the value of task ':compileJava' property 'javaCompiler'.
   > SystemInfo is not supported on this operating system.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 56s
The command '/bin/sh -c cd ${libname} && ./gradlew build' returned a non-zero code: 1

bmalinowsky, 2023-09-27:
Hi,
the image is already on dockerhub, you can just run it, i.e. `docker run --rm -it calimeroproject/knxserver ...", you have to pass the usual other parameters like `-v ...` for the server-config etc.

I pointed you to the Dockerfile in case you are familiar with those configurations and want to have a look at it. 
The reason for your error is that Gradle doesn't recognize the JDK environment of the base image, you have to point it to the exact JDK 17. There are several pitfalls, and I will modify the Dockerfile over time. 

I am currently just not sure where to put the various workarounds without affecting builds for other platforms.

bmalinowsky, 2023-10-04:
Hi, 

I did basic invocations for Aarch64, it should basically work. If there are any issues, please reopen.

closed 2023-10-04 by bmalinowsky