calimero-project/calimero-server #18: USB is not connected to KNX network

juergenschorb, 2023-11-20:
Hello,

when I start the server I am get some errors. Is this a configuration Problem?

Regards Jürgen

Log entries (Calimero on Docker Container):

13:29:33.703 [INFO ] io.calimero.server.knx-server: knx-server siemens control endpoint (eth0 10.1.1.34:3671) is up and running
13:29:33.906 [INFO ] io.calimero.usb.provider.javax: Found 1 KNX USB devices:
|--Bus 004 Device 003: ID 0908:02dd
| N 148/12 KNX Interface (Siemens)
| S/N 0001005521D7
13:29:34.061 [INFO ] io.calimero.usb.provider.javax.siemens: Bus 004 Device 003: ID 0908:02dd
N 148/12 KNX Interface (Siemens)
S/N 0001004421D7
13:29:35.123 [ERROR ] io.calimero.link.connector: initial connection attempt
io.calimero.link.KNXLinkClosedException: USB interface is not connected to KNX network
at io.calimero.core/io.calimero.link.KNXNetworkLinkUsb.(KNXNetworkLinkUsb.java:133)
at io.calimero.core/io.calimero.link.KNXNetworkLinkUsb.(KNXNetworkLinkUsb.java:116)
at [io.calimero.server@3.0-SNAPSHOT](mailto:io.calimero.server@3.0-SNAPSHOT)/io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$9(SubnetConnector.java:334)
at io.calimero.core/io.calimero.link.Connector$Link.connect(Connector.java:417)
at io.calimero.core/io.calimero.link.Connector$Link.(Connector.java:195)
at io.calimero.core/io.calimero.link.Connector.newLink(Connector.java:151)
at [io.calimero.server@3.0-SNAPSHOT](mailto:io.calimero.server@3.0-SNAPSHOT)/io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:380)
at [io.calimero.server@3.0-SNAPSHOT](mailto:io.calimero.server@3.0-SNAPSHOT)/io.calimero.server.gateway.KnxServerGateway.run(KnxServerGateway.java:730)
at java.base/java.lang.Thread.run(Unknown Source)
13:29:35.136 [WARNING] io.calimero.server.gateway.knx-server: cannot dispatch to KNX subnet 1.1.0: no subnet connection

bmalinowsky, 2023-11-20:
The server complains that the Siemens USB device is not physically connected to the KNX network, i.e., the 2 wires (usually red and black).

juergenschorb, 2023-11-20:
The USB device is physically connected to the KNX network. I have configured it in ETS and brand a KNX address on the device.

bmalinowsky, 2023-12-15:
That's odd, I can't say why your USB stick is reporting that, without more details.
You could try running the Docker knxtools:
`docker run -it --rm --privileged calimeroproject/knxtools devinfo --usb siemens`

Maybe there is a hint.

closed 2024-02-05 by bmalinowsky