calimero-project/calimero-server #20: Errors like "invalid operation mode 0x0", "address byte array too short" and "verify write failed for address"

aleksusklim, 2024-03-29:
I am trying to use Calimero docker image with `Raspberry Pi 3 Model B Rev 1.2` to connect `KNX-USB Data Interface (GIRA 1080 00)` via onboard USB port.

I've installed Docker and followed https://hub.docker.com/r/calimeroproject/knxserver

Here is my config (with stripped comments from the default sample):

```xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
	<discovery listenNetIf="eth0" outgoingNetIf="eth0" activate="true" />
	<serviceContainer activate="true" routing="false" networkMonitoring="true"
		udpPort="3671" listenNetIf="eth0">
		<knxAddress type="individual">7.5.0</knxAddress>
		<routing latencyTolerance="1000">224.0.23.12</routing>
		<disruptionBuffer expirationTimeout="30" udpPort="5555-5559" />
		<knxSubnet type="usb" medium="tp1"></knxSubnet>
		<groupAddressFilter>
		</groupAddressFilter>
		<additionalAddresses>
			<knxAddress type="individual">7.5.10</knxAddress>
			<knxAddress type="individual">7.5.11</knxAddress>
			<knxAddress type="individual">7.5.12</knxAddress>
			<knxAddress type="individual">7.5.13</knxAddress>
		 </additionalAddresses>
	</serviceContainer>
</knxServer>
```

The docker container starts and successfully detects USB-KNX device, but then randomly throws errors like:
- `io.calimero.KNXIllegalArgumentException: invalid operation mode 0x0`
- `io.calimero.KNXIllegalArgumentException: address byte array too short`
Sometimes preceded with one or more lines like:
- `io.calimero.link.135e:0022: verify write failed for address 60: 12 vs 09`

Server is closed but apparently does not exit automatically after this (which is bad, how do I suppose to automate its restart after such state?)
Running again on the same might give a different error message. As if there were some problems with USB connection itself?

I tried to run in different modes by changing the XML config (but random errors were greatly misguiding me until I started to repeat my attempts several times after each change), but generally the device is not working.

For me it worked only 3 times (from roughly 30 or more my attempts), randomly: one shortly after Raspberry PI boot; one after power cycling USB ports (as I started to do it constantly) with `sudo uhubctl -l 1-1 -p 2 -a 0` and `sudo uhubctl -l 1-1 -p 2 -a 1`, and one just between various unsuccessful attempts without any changes in config or anything.

When the device was behaving, I was able to try ETS connect (and it showed "OK" for `Test` button after I clicked on a newly appeared IP-Tunnel device there), and also checked routing mode between TP and IP (fortunately was enabled in my config at that attempt) – it was working, at least routing some telegrams.

I suspected low USB power, so I've added a 12 V powered USB hub between Raspberry and Gira – no changes. Also I replaced the power source of Raspberry itself – again, nothing changed.

When I use this Gira KNX USB interface with Windows PC for ETS – it works flawlessly.
What could be the reason why Calimero fails? It this somehow related to hardware?

Here are full logs captured after Raspberry reboot (already with a USB hub) – two consecutive attempts that gave different error messages:

```
alpi@rpi3:~ $ sudo chmod 666 /var/run/docker.sock
alpi@rpi3:~ $ docker run --name my_knx --rm --tty --interactive --net host --privileged --volume /srv/my/dock/server-config.xml:/usr/app/server-config.xml  calimeroproject/knxserver -vvv server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
13:48:09.295 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
13:48:09.304 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:48:15.174 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
13:48:15.217 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:48:15.449 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
13:48:15.458 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
13:48:15.464 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
13:48:15.890 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
13:48:16.150 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX IP Server' with 1 service container, discovery: listen on [eth0] send on [eth0] interfaces
service container usb-7.5.0:
	listen on eth0 (port 3671), KNX IP routing disabled
	usb connection: TP1 medium, device 7.5.0
13:48:16.220 [DEBUG  ] io.calimero.server.knx-server: setup usb subnet 'usb-7.5.0'
13:48:16.303 [INFO   ] io.calimero.server.gateway.knx-server: activate 'usb-7.5.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
13:48:16.323 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
13:48:16.326 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
[... 244 lines cut ...]
	at io.calimero.core/io.calimero.KNXAddress.<init>(KNXAddress.java:80)
	at io.calimero.core/io.calimero.IndividualAddress.<init>(IndividualAddress.java:93)
	at io.calimero.core/io.calimero.link.BcuSwitcher.enter(BcuSwitcher.java:167)
	at io.calimero.core/io.calimero.link.KNXNetworkLinkUsb.linkLayerMode(KNXNetworkLinkUsb.java:230)
	at io.calimero.core/io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:156)
	at io.calimero.core/io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:115)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$9(SubnetConnector.java:334)
	at io.calimero.core/io.calimero.link.Connector$Link.connect(Connector.java:417)
	at io.calimero.core/io.calimero.link.Connector$Link.<init>(Connector.java:195)
	at io.calimero.core/io.calimero.link.Connector.newLink(Connector.java:151)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:380)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.KnxServerGateway.run(KnxServerGateway.java:730)
	at java.base/java.lang.Thread.run(Unknown Source)

13:49:10.205 [INFO   ] io.calimero.server.gateway.knx-server: knx-server: user request for shutdown
13:49:10.245 [DEBUG  ] io.calimero.server.knx-server: cleanup knx-server discovery endpoint
13:49:10.251 [DEBUG  ] io.calimero.server.knx-server: cleanup knx-server control endpoint usb-7.5.0
13:49:10.253 [INFO   ] io.calimero.server.knx-server: knx-server control endpoint usb-7.5.0 closed
13:49:10.256 [INFO   ] io.calimero.server.knx-server: knx-server discovery endpoint closed
13:49:10.263 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
13:49:10.351 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
13:49:10.364 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
stop
request to stop server
```

bmalinowsky, 2024-03-30:
> Server is closed but apparently does not exit automatically after this ...

Actually, the server should try a re-init/reconnect of the particular device (in your case the USB stick), it does do so every 10 seconds (that's currently hardcoded, cannot be configured).

In the first log,

> 13:48:17.719 [ERROR  ] io.calimero.usb.provider.javax.135e:0022: creating HID class report from 00000000000000000000024dff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

receiving such input is definitely not resembling any part of the USB protocol. The good thing is, the data event is logged as error but ignored, so it won't affect subsequent operation.

For the other more random errors you mention and that I can see in the log I think you hit a bug where frames get duplicated. The Linux you are using is 32-bit, right? So you are using the armv7 build?

aleksusklim, 2024-03-30:
> Actually, the server should try a re-init/reconnect of the particular device

Hey, you know what? Later, as messages were passing over KNX bus, I saw more errors in the log titled `received unspecified frame` with RAW bytes dump.

As if the server gets them but refused to parse because it is in "closed" state.
Also UDP ports were closed (I think?), and I saw no external connectivity. Only logging of something related to KNX activity.

I haven't noticed any reconnections attempts. Several times I left console open for quite some time (much more than 10 seconds), waiting for docker container to close on its own, but it stuck at `knx-server control endpoint usb-7.5.0 closed` and `knx-server discovery endpoint closed` being last meaningful lines in the log.

> receiving such input is definitely not resembling any part of the USB protocol.

Is USB bus really susceptible to protocol errors like that? I always thought that for USB you either have everything working as intended, or a device disconnection (I'm judging from Windows point-of-view), everything else is in fact a buggy driver that failed to handle a specific state sent by device.

Is there any "driver" for this USB device "installed" in Linux/Raspberry, or Calimero is playing the driver role on its own?
Maybe I need to install something other to better support KNX-USB devices into the system? (I don't know, have no prior experience in USB under Raspberry or any other Unix).

> the data event is logged as error but ignored, so it won't affect subsequent operation.

For me the main error looks like it is happening in the initialization stage where Calimero cannot ignore the failed read because its result is critical for proper operation afterwards.

In theory, there could be some kind of retransmission attempt to read again in case of errors, but that would only hide the actual problem and make the device slower and less reliable…

> I think you hit a bug where frames get duplicated.

I noticed in logs consequential pairs like `12 vs 09` and then `00 vs 12` – as if they were off by one position.

> The Linux you are using is 32-bit, right? So you are using the armv7 build?

How do I check the bitness of the application in a docker container?
As I said, I followed the guide that tells just `docker pull calimeroproject/knxserver`

I assumed that it would either:
- detect my architecture and download the correct build automatically; or
- throw an error telling that the architecture is different.

But it worked instead, and a few times was even functional!

Also, if you would point me to the installation instruction for Windows 10 x64 (if Calimero can work under Windows), I will test my device there. I don't mind installing any Linux in a VMware virtual machine if this might help debug the issue, just tell me what to do.

---

Here are logs of Calimero left for some time. It catches KNX telegrams issuing a warning at each one (I think the server stuck in an invalid state forever); also ETS cannot see Raspberry, and cannot connect directly by its IP either.

```
alpi@rpi3:~ $ docker run --name my_knx --rm --tty --interactive --net host --privileged --volume /srv/my/dock/server-config.xml:/usr/app/server-config.xml  calimeroproject/knxserver -vvv server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
19:23:07.742 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
19:23:07.751 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
19:23:08.741 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
19:23:08.762 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
19:23:08.995 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
19:23:09.003 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
19:23:09.009 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
19:23:09.136 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
19:23:09.380 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX IP Server' with 1 service container, discovery: listen on [eth0] send on [eth0] interfaces
service container usb-7.5.0:
	listen on eth0 (port 3671), KNX IP routing disabled
	usb connection: TP1 medium, device 7.5.0
19:23:09.446 [DEBUG  ] io.calimero.server.knx-server: setup usb subnet 'usb-7.5.0'
19:23:09.532 [INFO   ] io.calimero.server.gateway.knx-server: activate 'usb-7.5.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
19:23:09.553 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
19:23:09.557 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
19:23:09.561 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
type 'stop' to stop the gateway and shutdown the server
19:23:09.848 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
[... cut ...]
19:51:53.716 [DEBUG  ] io.calimero.usb.provider.javax.135e:0022: received EMI1 frame 49bc120f1907e20080ff
19:51:53.719 [WARNING] io.calimero.link.135e:0022: received unspecified frame 49 bc 12 0f 19 07 e2 00 80 ff
19:52:19.571 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 1 IN I/O request 011311000800090101000049bc11020001e10081
19:52:19.575 [DEBUG  ] io.calimero.usb.provider.javax.135e:0022: received EMI1 frame 49bc11020001e10081
19:52:19.579 [WARNING] io.calimero.link.135e:0022: received unspecified frame 49 bc 11 02 00 01 e1 00 81
```

bmalinowsky, 2024-03-31:
I also think that the main culprit is the failed init stage; because if I take one of the "unspecified" frames from your log, e.g., "49bc120f1907e20080ff" and parse it with Calimero, I do get a reasonable result (1.2.15->3/1/7 L_Data.ind, low priority hop count 6, tpdu 00 80 ff: T_Group, A_Group.write).

OK, so let's take a look at each individual point:

* the error "creating HID class report from 000000000000000..."; while interesting from a dev point, is also quite noisy. I pushed a commit which shortcuts this to some extent. 
* Regarding your question about USB/driver: for USB, Calimero doesn't do anything special; it uses the standard HID on every platform (where a JVM can run). Currently there are 2 workarounds
  * data event with 0 length data (duh)
  * data event with all 0 data

  Both shouldn't occur, but aren't necessarily out of the ordinary depending on the driver implementation, think spurious wake-ups, some unfiltered interrupts, polling timeouts etc. I have no idea what the lonely "024dff" is in the middle of zeros in your log. That's why previously the parsing logic tried to make sense out of it.

* The USB init hangs at some point, I see that from your last attached log

* > I noticed in logs consequential pairs like 12 vs 09 and then 00 vs 12 – as if they were off by one position.

  Yes, and I am quite sure that was because of duplicated frames. This shouldn't happen within Calimero, and I think I found the error.
* You can check the docker image with `docker image inspect calimeroproject/knxserver`. There are lines with "Architecture", "Variant", and "Os". The only reason I asked this is that I will then update the corresponding image of that particular arch, so you can give it a try.
* Calimero is written in Java, so it runs on every platform where a JVM is available (true for Windows).

aleksusklim, 2024-03-31:
I don't believe the error is because Calimero fails to ignore some zeros if they should not be there in the first place.
It is GIRA, not some random Chinese cheap USB-KNX stick that might violate the protocol!

Low power?
Buggy USB port on Raspberry? (I haven't tried them all…)
Broken USB support on 32-bit old Raspbian OS ?
Jittering in USB jack?

OH, HEY
You know what? After testing the device in the first time and thinking that Calimero is working, I went installing it (power supply + raspberry + gira + usb hub) to the network cabinet, and _maybe_ I had to cross the cable near to 220 V mains as a temporary solution…
Could that be a problem!?

> it runs on every platform where a JVM is available

Okay, I've downloaded `jdk-22_windows-x64_bin.zip` (assume I'll put /bin/ to %PATH%), what next? How do I combine `calimero-server` with `calimero-core`, `calimero-device` and `calimero-usb`? And how do I run the whole thing without Docker?
Should I download them all and run `./gradlew build` on each to get those `*-SNAPSHOT.jar` to reference for Java command-line? (https://github.com/calimero-project/calimero-server?tab=readme-ov-file#start-server)

> There are lines with "Architecture", "Variant", and "Os"

It reads `"arm"`, `"v7"`, `"linux"`

<details><summary>docker image inspect calimeroproject/knxserverserver</summary>

```json
alpi@rpi3:~ $ docker image inspect calimeroproject/knxserverserver
[
    {
        "Id": "sha256:4faf42beaf836d4c22750f84e45f74b6156e6035e8c9dbf35ed512ade16bd79e",
        "RepoTags": [
            "calimeroproject/knxserver:latest"
        ],
        "RepoDigests": [
            "calimeroproject/knxserver@sha256:7b6e46a6dac318fe3d7a675deb64aa4e7cb2472e17b72f0eded38fbde0c5f65d"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2023-10-28T13:09:03.49776673+02:00",
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "3671/tcp": {},
                "3671/udp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/lib/jvm/jdk-17/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "JAVA_HOME=/usr/lib/jvm/jdk-17"
            ],
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "/usr/app",
            "Entrypoint": [
                "calimero-server-3.0-SNAPSHOT/bin/calimero-server"
            ],
            "OnBuild": null,
            "Labels": {
                "maintainer": "Calimero Project <calimero.project@gmail.com>"
            }
        },
        "Architecture": "arm",
        "Variant": "v7",
        "Os": "linux",
        "Size": 20979032,
        "GraphDriver": {
            "Data": {
                "LowerDir": "/var/lib/docker/overlay2/0481dcfd3b38c45d80b4aa00c8f2d87bf73b784177fbfaaab3aacdf360bacde1/diff:/var/lib/docker/overlay2/3d8e6d1713ab457c8bb3269b655bdf62dfb5c7aafba425e601ea28defe4cc362/diff:/var/lib/docker/overlay2/bf7ae8efec547327d366bfc59353213ae7e553ac27a5698569a5d121cc22cb85/diff:/var/lib/docker/overlay2/735e5bfa321b548aa271b35b101c5181e23cd8e8f962a006d93a4f1419cf2374/diff:/var/lib/docker/overlay2/6db09e55c0ac80f692b89e4802f10978d23df057bb7bb793064f0b303a93454f/diff",
                "MergedDir": "/var/lib/docker/overlay2/82b2779d5b15538031275154b1f2751dada5de2303cad21a0115f8ec481f8783/merged",
                "UpperDir": "/var/lib/docker/overlay2/82b2779d5b15538031275154b1f2751dada5de2303cad21a0115f8ec481f8783/diff",
                "WorkDir": "/var/lib/docker/overlay2/82b2779d5b15538031275154b1f2751dada5de2303cad21a0115f8ec481f8783/work"
            },
            "Name": "overlay2"
        },
        "RootFS": {
            "Type": "layers",
            "Layers": [
                "sha256:315bd5088587f904ce90b0801a0c8db45648746faf106758d6946f4ef96ae5d0",
                "sha256:e08b2f35119a4737c0636d8977a03c608c507297281c0d98a540f5ec36fbeb22",
                "sha256:4d1ba503a71648b1249762d2e75595a0b419f6486c6a69101747cb562a1484de",
                "sha256:fda0ba65621b46a05a5d44f4024f2c752ebb5b3d15f75122c1f21d6286a9e4cb",
                "sha256:3c0deeed5d62d69ee0c8cfe568baea2ede583465be633297d1861f6bb7f3d756",
                "sha256:5f70bf18a086007016e948b04aed3b82103a36bea41755b6cddfaf10ace3c6ef"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```

</details>

bmalinowsky, 2024-03-31:
I uploaded a new docker image for linux/arm/v7. (I don't know how much you use docker, but you have to make sure to pull the image again.)

The 1) zeros, 2) the init, and 3) the failure to reset are not that related. But you see the 3rd because the init really got screwed up.

For 1) I cannot do much but ignore it (which is fine), the init should have really improved (having the duplicated frames was a bug).

In case you want to continue with the Windows install, I actually would use a standard JDK Installer (exe or msi), which alleviates you of the nitty-gritty parts of the setup. But otherwise you download the server repo (or use `git clone https://github.com/calimero-project/calimero-server.git`) and then `gradlew.bat build`. If that succeeds, there is no need to install anything.

aleksusklim, 2024-03-31:
Wow, it's working!!
(`docker rmi calimeroproject/knxserver:latest` + `docker pull calimeroproject/knxserver`)

Tried to run like 10 times (no reboots from the last failed attempt, nor any cable management happened yet), with zero failures!
Discovery, Tunnel and Routing – everything seems to be working.

Though, I can't see any clear lines in the log that would state "retry" or something. The only one warning type was like:
`[WARNING] io.calimero.link.135e:0022: negative confirmation of 7.5.255: 2e 00 b1 60 11 fe 75 ff 01 43 00`

How would I know that something is bad in the connection? I mean, I wanna try with a different USB cable and away from 220V mains – will that reduce ignores/disconnections or not?

Also, how do I install the previous docker image version again (to clearly test it with a good cable) now?

---

And finally, since we're here: am I right that all I need to do is to run the docker container and nothing more (no coding in Java) to freely use KNX from NodeJS with [KNXUltimate](https://github.com/Supergiovane/KNXUltimate) on the same Raspberry PI – just by pointing it to the Tunnel (or the Multicast routing group), right?

Thing is, it needs a working KNX/IP gateway (which are expensive) to communicate with the bus. That's how I found Calimero (from https://github.com/knxd/knxd/issues/369#issuecomment-682977367) when trying to understand, can a KNX-USB be used from `knxd` instead of a serial HAT? Because I found only two such HATs for Raspberry:
- https://www.tindie.com/products/cpu20/knx-raspberry-pi-hat/ (looks promising, but exists ONLY there)
- https://knx-trade.ru/weinzierl-knx-baos/706-5208.html (certified, but implements BAOS and not a RAW access, so I think it would be limiting me in many ways)
So I thought: wait, USB interfaces are much more common and cheaper (as low as $50 sticks from China), maybe it is possible to use those with Raspberry too?

Basically, Calimero turns a Raspberry PI with a KNX-USB interface to a fully-functional KNX/IP router, right?

---

Some logs from the updated build:

<details><summary>Run 1</summary>

```
alpi@rpi3:~ $ docker run --name my_knx --rm --tty --interactive --net host --privileged --volume /srv/my/dock/server-config.xml:/usr/app/server-config.xml  calimeroproject/knxserver -vvv server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
19:01:37.166 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
19:01:37.175 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
19:01:38.246 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
19:01:38.267 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
19:01:38.722 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
19:01:38.731 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
19:01:38.737 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
19:01:38.876 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
19:01:39.036 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX IP Server' with 1 service container, discovery: listen on [zt6ovuwo6l] send on [zt6ovuwo6l] interfaces
service container Usb-7.5.0:
	server: listen on zt6ovuwo6l (port 3671), KNX IP routing multicast group 224.0.23.12
	Usb connection: TP1 medium, device 7.5.0
19:01:39.113 [DEBUG  ] io.calimero.server.knx-server: setup Usb subnet {1}
19:01:39.202 [INFO   ] io.calimero.server.gateway.knx-server: activate 'Usb-7.5.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
19:01:39.224 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
19:01:39.228 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
19:01:39.239 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
type 'stop' to stop the gateway and shutdown the server
19:01:39.549 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
19:01:39.695 [TRACE  ] io.calimero.server.knx-server: found network interfaces zt6ovuwo6l [192.168.193.121]
19:01:39.700 [INFO   ] io.calimero.server.knx-server: discovery endpoint [zt6ovuwo6l] is up and running
19:01:39.719 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
19:01:39.743 [TRACE  ] io.calimero.server.knx-server: Usb-7.5.0 control endpoint bound to 192.168.193.121:3671
19:01:39.756 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on zt6ovuwo6l
19:01:39.776 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
19:01:39.808 [INFO   ] io.calimero.server.knx-server: Usb-7.5.0 KNX IP routing service (zt6ovuwo6l 224.0.23.12) is up and running
19:01:39.927 [INFO   ] io.calimero.server.knx-server: Usb-7.5.0 secure mgmt/tunneling connections: optional/optional
[... cut ...]
19:08:28.164 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
19:08:28.233 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
19:08:28.252 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
```

</details>

If you think everything is correct, this Issue can be closed.
Thank you!

bmalinowsky, 2024-04-01:
> How would I know that something is bad in the connection?

You can try it out by running the server, then disconnect the usb stick and reconnect it. There will be an error in the log, and the server will start attempts to reinit the usb stick.

If you ask for a specific notification that can be monitored (e.g., in the windows event viewer, dmesg, email notification) that's a fair point. But the server does currently not have that. So the best thing is to watch if the process terminates, or configure a specific log configuration which filters only on errors, and watch that.

> [WARNING] io.calimero.link.135e:0022: negative confirmation of 7.5.255: 2e 00 b1 60 11 fe 75 ff 01 43 00

Maybe that device does not exist?

> How would I know that something is bad in the connection? I mean, I wanna try with a different USB cable and away from 220V mains – will that reduce ignores/disconnections or not?

A standard USB cable is shielded to prevent adverse effects from an electromagnetic field (but I don't know specifics.)

> ... just by pointing it to the Tunnel (or the Multicast routing group), right?

Right. 

> Basically, Calimero turns a Raspberry PI with a KNX-USB interface to a fully-functional KNX/IP router, right?

Right.

There is one limitation in that USB devices do have their own address, which can show up with management connections (there is the config option `knxAddress` in the README). As long as you only use process communication, this doesn't matter.

In the logs, there is one other warning "skip verifying knx address of 'Usb-7.5.0' subnet interface", which shows up because the server checks the current KNX address of the USB stick (using cEMI, your devices requires EMI1). You can ignore this if you don't intend to reconfigure the USB sticks' KNX address using the server-config.xml.

Otherwise, everything looks good!

Thanks for aiding all the logs, which made finding the culprit almost easy :)

closed 2024-04-01 by bmalinowsky

aleksusklim, 2024-04-01:
I tried disconnecting USB cable (from the GIRA's end, if that matters; so the USB port on Raspberry was the same), but Calimero struggles to reconnect it without a full restart.

It loops along:
```
[DEBUG  ] io.calimero.link.135e:0022: execute scheduled connect 10 (no max)
[TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
[WARNING] io.calimero.link.135e:0022: connection attempt 10: find USB device matching '' (no USB device matching Bus 1 Device 7: ID 135e:0022
Attached at port 2 (bus:1/1/2)
Full Speed USB)
```

Is this because I have:
`<knxSubnet type="usb" medium="tp1" knxAddress="1.1.0"></knxSubnet>`
? I tried to add the ID of my usb like this:
`<knxSubnet type="usb" medium="tp1" knxAddress="1.1.0">135e:0022</knxSubnet>`

Didn't help, it cannot reconnect on its own!
My timeline:
- Run docker and wait
- Send a read request with KNXUltimate from external machine
- Disconnect USB and wait
- Connect USB and wait
- Restart KNXUltimate to send another read request without a response now
- Type "stop" and wait
- Run docker again
- Restart KNXUltimate to confirm that now Calimero works as expected and replies

(Note that I tried to read/write the bus only when USB was connected, so in logs where you see external requests – GIRA was connected and should have been already reinitialized)

**LOGS**

<details><summary>Without specifying USB id</summary>

```
alpi@rpi3:~ $ docker run --name my_knx --rm --tty --interactive --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml calimeroproject/knxserver -vvv server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
14:07:43.620 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
14:07:43.630 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
14:07:44.706 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
14:07:44.727 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
14:07:45.168 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
14:07:45.177 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
14:07:45.183 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
14:07:45.311 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
14:07:45.451 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX' with 1 service container, discovery: listen on [zt6ovuwo6l] send on [zt6ovuwo6l] interfaces
service container Usb-1.1.0:
	server: listen on zt6ovuwo6l (port 3671), KNX IP routing multicast group 224.0.23.12
	Usb connection: TP1 medium, device 1.1.0
14:07:45.530 [DEBUG  ] io.calimero.server.knx-server: setup Usb subnet {1}
14:07:45.615 [INFO   ] io.calimero.server.gateway.knx-server: activate 'Usb-1.1.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
14:07:45.637 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
14:07:45.640 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
14:07:45.646 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
type 'stop' to stop the gateway and shutdown the server
14:07:45.941 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
14:07:46.075 [TRACE  ] io.calimero.server.knx-server: found network interfaces zt6ovuwo6l [192.168.193.121]
14:07:46.080 [INFO   ] io.calimero.server.knx-server: discovery endpoint [zt6ovuwo6l] is up and running
14:07:46.112 [TRACE  ] io.calimero.server.knx-server: Usb-1.1.0 control endpoint bound to 192.168.193.121:3671
14:07:46.126 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
14:07:46.125 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on zt6ovuwo6l
14:07:46.148 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
14:07:46.173 [INFO   ] io.calimero.server.knx-server: Usb-1.1.0 KNX IP routing service (zt6ovuwo6l 224.0.23.12) is up and running
14:07:46.274 [INFO   ] io.calimero.server.knx-server: Usb-1.1.0 secure mgmt/tunneling connections: optional/optional
14:07:46.324 [INFO   ] io.calimero.server.knx-server: Usb-1.1.0 tcp service (zt6ovuwo6l 192.168.193.121:3671) is up and running
14:07:46.333 [INFO   ] io.calimero.server.knx-server: Usb-1.1.0 control endpoint (zt6ovuwo6l 192.168.193.121:3671) is up and running
14:07:46.683 [DEBUG  ] io.calimero.usb.provider.javax: Attached USB devices:
Root hub
 |--Bus 1 Device 1: ID 1d6b:0002 Linux 6.1.21-v7+ dwc_otg_hcd DWC OTG Controller S/N 3f980000.usb
     High Speed USB
      |--Bus 1 Device 2: ID 0424:9514
[... 684 lines cut ...]
[... cut ...]
14:23:15.215 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
14:23:15.303 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
14:23:15.318 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
alpi@rpi3:~ $ 
```

</details>

This is not urgent: since restarting the application works, I think I would be able to hack around it.
I've just tested the disconnection as you suggested.

bmalinowsky, 2024-04-02:
I think I found it: it's an issue when running in a containerized environment (because it does work when I try it on Windows/Linux/Mac etc. as is).

I made it work by sharing the `usb` folder by supplying the following volume to docker: `-v /dev/bus/usb/:/dev/bus/usb/` when running the docker image.

aleksusklim, 2024-04-02:
Yay, it's working! Relevant log part:

```
14:58:30.359 [TRACE  ] io.calimero.server.knxnetip.KNX IP routing service (zt6ovuwo6l 224.0.23.12): discard multicast loopback cEMI frame: 1.1.4->6/2/2 L_Data.ind, low priority hop count 5, tpdu 00 80
14:58:30.360 [TRACE  ] io.calimero.server.gateway.knx-server: dispatching to KNX subnets: exclude subnet 135e:0022
14:58:32.208 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: release USB interface 0, active=true, claimed=true
14:58:32.213 [WARNING] io.calimero.usb.provider.javax.135e:0022: close connection
javax.usb.UsbPlatformException: USB error 5: Unable to re-attach kernel driver: Entity not found
	at org.usb4java.javax.ExceptionUtils.createPlatformException(ExceptionUtils.java:39)
	at org.usb4java.javax.AbstractDevice.releaseInterface(AbstractDevice.java:464)
	at org.usb4java.javax.Interface.release(Interface.java:113)
	at io.calimero.usb.provider.javax@3.0-SNAPSHOT/io.calimero.usb.provider.javax.UsbConnection.close(UsbConnection.java:616)
	at io.calimero.usb.provider.javax@3.0-SNAPSHOT/io.calimero.usb.provider.javax.UsbConnection$UsbCallback.run(UsbConnection.java:195)
	at java.base/java.lang.Thread.run(Unknown Source)

14:58:32.230 [DEBUG  ] io.calimero.link.135e:0022: link closed by internal event (USB error 9: Transfer error on interrupt endpoint: Pipe error)
14:58:32.242 [DEBUG  ] io.calimero.server.gateway.knx-server: set maximum APDU length of '135e:0022' to 15
14:58:42.262 [DEBUG  ] io.calimero.link.135e:0022: execute scheduled connect 1 (no max)
14:58:42.265 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
14:58:42.347 [WARNING] io.calimero.link.135e:0022: connection attempt 1
14:58:52.350 [DEBUG  ] io.calimero.link.135e:0022: execute scheduled connect 2 (no max)
14:58:52.353 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
14:58:52.375 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: USB interface 00, setting 0
14:58:52.378 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 2 OUT
14:58:52.380 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 1 IN
14:58:52.382 [DEBUG  ] io.calimero.usb.provider.javax.135e:0022: Found USB device endpoint addresses OUT 0x2, IN 0x81
14:58:54.753 [DEBUG  ] io.calimero.serial.ConnectionFactory: usb4java connection provider port setup: Bus 001 Device 005: ID 135e:0022 USB interface 00
14:58:54.758 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: sending I/O request 011309000800010f01000001
14:58:54.768 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 1 IN I/O request 01130b000800030f020000010001
```

> There is one limitation in that USB devices do have their own address, which can show up with management connections

Can you explain a bit more, please?
This is what I understand (correct me if I'm wrong):
- A device that has both IP connection (blue, backbone bus) and KNX connection (green, knx line) may work as a Router/Multicast or as a Gateway/Tunnel, or both.
- If such device is acting as a Bus Coupler (with the backbone being IP), it should be added to ETS project to allow modifications to its routing/filter table (so that KNX line segment would hear only dedicated telegrams).
- Routers work by multicasting UDP datagrams "into the IP bus", as LAN packets. When a router hears KNX datagram on KNX bus – it multicasts it to IP (without specifying a single destination); when it hears any other multicasted UDP – it sends a telegram to the KNX bus.
- Gateway works by establishing tunnel connections with clients. The main client is often ETS: it connects to the gateway IP (from any its local UDP port) in direct or NAT mode; then this client is assigned a KNX address as if "it is on the green side of the coupler" – virtually from KNX bus, like if ETS was connected in the physical bus by USB-KNX.
- Still, ETS can work by utilizing Router mode too – in this case it needs to multicast from a local network interface, without knowing IP of the router.

If that is true, I have two question regarding physical addresses:
1. Why do I need to specify a list of "additional addresses" for gateway clients? When using ETS previously with real gateways, I was able to set any physical address I want, on the same page where I set gateway IP. But now with Calimero – I see ETS gives me the actual dropdown list of those addresses to pick one! Interesting…

So, any gateway that does not show a list – really supports only one connection?
I've read about `reuseCtrlEP` but that disables routing. Without this and with empty `<additionalAddresses>` ETS fails with "no more connections available" (or something like that). How do I emulate "a normally-looking gateway" then?
Or:

> There is one limitation in that USB devices do have their own address, which can show up with management connections (there is the config option `knxAddress` in the README).

I don't quite get it. Should I specify it to be the same as `the individual address of the service container`? Or set a spare one?
Why this address exists, isn't Calimero giving USB device a complete RAW telegrams to send with any contents it wants?

2. Why do I have to use a correct KNX physical address as a Bus Coupler to activate routing? The readme states about the individual address of the service container: `If routing is activated, requires a coupler/backbone address (x.y.0 or x.0.0)` For example, I think Logi [... cut ...]
Also when Calimero docker is killed forcefully, I eventually get a tunnel disconnection timeout error even for localhost connections on the Raspberry PI itself, which is very good.

I occasionally receive this non-fatal error in KNXUltimate, but I haven't investigated yet:
```
[error] 03/04/2024, 00:26:45.64 KNXUltimate-KNXEngine: Received KNX packet: TUNNELING: Unexpected Tunnel Ack with seqCounter = 2
Last telegram acknowledge KNXTunnelingAck {
  type: 1057,
  length: 4,
  _header: KNXHeader {
    _headerLength: 6,
    _version: 16,
    service_type: 1057,
    length: 10
  },
  channelID: 1,
  seqCounter: 3,
  status: 0
} true
```

bmalinowsky, 2024-04-04:
> This is what I understand (correct me if I'm wrong): ...

Your points sound about right.

> Why do I need to specify a list of "additional addresses" ...

The additional addresses configure how many client connections are allowed in parallel. Typically, KNX/IP interfaces offer 4 or 8, but you can add as many as you like. The only requirement is that those addresses are not already assigned in the KNX subnet (because this would result in a KNX tunneling client and a KNX device having the same physical address).

> So, any gateway that does not show a list – really supports only one connection?

It depends on the access protocol/HW. You can have a KNX/IP interface which reuses its own address, resulting in max. of 1 connection at a time.

> Should I specify it to be the same as the individual address of the service container? Or set a spare one?

If everything works, there is no need to do anything. Calimero multiplexes multiple tunneling connections over the same source address of the USB stick. I mentioned it because it might show up with mgmt/programming.

> Why this address exists, isn't Calimero giving USB device a complete RAW telegrams to send with any contents it wants?

Certain USB devices reject certain frames, so no, you cannot send arbitrary stuff.

> Why do I have to use a correct KNX physical address as a Bus Coupler to activate routing? 

It's mandated by the KNX spec, to reflect the topology. It's not really necessary from a usability point of view. (You comment out the check, and you can set any other address.)

> As I understand, Calimero is never added to ETS project, and cannot receive a flash update / download via ETS as a real Bus Coupler could. So it is as transparent as LogicMachine.

You cannot add it to an ETS project, because there is no KNX product file.
You can program the properties and certain memory locations of a Calimero device, as you can program the Calimero server (which is a device by definition).

> I thought it could be plugged into the bus anywhere no matter what physical addresses there are?

You can, but the physical addresses have to match the topology. (You can also use arbitrary addresses, you might not like the result, though.)

aleksusklim, 2024-04-04:
Thank you for detailed answer!
So, most of points are "we made this to comply with the spec" (and reduce potential problems), I see. Good to hear that I understand KNX correctly, thanks again.