calimero-project/calimero-server #21: How to use Calimero with KNX Raspberry Pi HAT properly? ETS fails

aleksusklim, 2024-04-13:
I am trying to use https://www.tindie.com/products/cpu20/knx-raspberry-pi-hat/ (without installing `knxd`) for `Raspberry Pi 3 Model B Rev 1.2`

I've changed my [previous](https://github.com/calimero-project/calimero-server/issues/20) config to this:
`<knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet>`

I also followed some steps according to https://hackaday.io/project/171850/instructions

- Free port: `sudo raspi-config` → `3 Interface Options`  → `I6 Serial Port` → Login shell = No, Hardware enabled = Yes;
- Disable Bluetooth: `sudo nano /boot/config.txt` → check `enable_uart=1` + add `dtoverlay=disable-bt`
- Execute: `sudo systemctl stop hciuart.service` + `sudo systemctl disable hciuart.service`

Now Calimero seems to work fine in both tunnel and router modes: I can send reads, I hear responses, and I can write.

But I fail to flush/download a device configuration with ETS!
Previously, with USB connection I had to set-up my Calimero interface as a Coupler between TP and IP (in line properties inside ETS), otherwise only "Full" download worked (with pressing of the programming button); but a normal flushing failed.

Now I'm sure I have set this up correctly, but ETS fails to configure a device anyhow: neither tunnel nor multicast, and cannot complete an individual address change.
But Group/Bus monitors are working!

What have I missed?

After attempts to establish a tunnel in ETS sometimes Calimero stops accepting tunnel requests anymore.
I see in logs
```
[DEBUG  ] io.calimero.serial.tpuart:/dev/ttyAMA0: received TP1 L-Data (length 10): 14 8b bc 11 04 32 01 e1 00 40
[ERROR  ] io.calimero.serial.tpuart:/dev/ttyAMA0: error creating L-Data from TP1 data (length 14 8b bc 11 04 32 01 e1 00 40): java.lang.IllegalArgumentException: fromIndex(0) > toIndex(-2)
```

But I'm not sure that it's the main error here.

<details><summary>More logs:</summary>

```
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
13:40:01.385 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
13:40:01.394 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:40:02.204 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
13:40:02.224 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:40:02.722 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
13:40:02.731 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
13:40:02.737 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
13:40:02.998 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
13:40:03.172 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX' with 1 service container, discovery: listen on [eth0] send on [eth0] interfaces
service container /dev/ttyAMA0:
	server: listen on eth0 (port 3671), KNX IP routing multicast group 224.0.23.12
	Tpuart /dev/ttyAMA0 connection: TP1 medium, device 6.6.0
13:40:03.241 [DEBUG  ] io.calimero.server.knx-server: setup Tpuart subnet {1}
13:40:03.326 [INFO   ] io.calimero.server.gateway.knx-server: activate '/dev/ttyAMA0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
13:40:03.346 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
13:40:03.350 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
13:40:03.354 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
type 'stop' to stop the gateway and shutdown the server
13:40:03.629 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
13:40:03.732 [TRACE  ] io.calimero.server.knx-server: found network interfaces eth0 [192.168.31.22]
13:40:03.738 [INFO   ] io.calimero.server.knx-server: discovery endpoint [eth0] is up and running
13:40:03.748 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 control endpoint bound to 192.168.31.22:3671
13:40:03.760 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on eth0
13:40:03.763 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.serial.provider.jni.TtySerialComProvider
13:40:03.764 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
13:40:03.779 [TRACE  ] io.calimero.serial.provider.jni: check Java library path /usr/java/packages/lib:/lib:/usr/lib
13:40:03.786 [DEBUG  ] io.calimero.serial.provider.jni: opening serial port /dev/ttyAMA0
13:40:03.793 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 KNX IP routing service (eth0 224.0.23.12) is up and running
13:40:03.846 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 secure mgmt/tunneling connections: optional/optional
13:40:03.890 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 tcp service (eth0 192.168.31.22:3671) is up and running
13:40:03.897 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 control endpoint (eth0 192.168.31.22:3671) is up and running
13:40:03.922 [DEBUG  ] io.calimero.serial.ConnectionFactory: io.calimero.serial.provider.jni.TtySerialComProvider@97bbb8 port setup: baudrate 19200, even parity, 8 databits, 1 stopbits, timeouts: read 0 read total 0 constant 100 write total 0 write constant 0
13:40:03.943 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: drained rx queue (0 bytes)
13:40:03.947 [DEBUG  ] io.calimero.serial.tpuart:/dev/ttyAMA0: reset TP-UART controller
13:40:03.952 [DEBUG  ] io.calimero.serial.tpuart:/dev/ttyAMA0: TP-UART reset.ind
13:40:03.958 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: loop time = 2,376 us
[... 317 lines cut ...]
13:40:59.979 [TRACE  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:5559: sending disconnect request for KNX IP LinkLayer 192.168.31.101:5559 (UDP NAT, channel 4, 6.6.4)
13:40:59.981 [TRACE  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:5559: send 192.168.31.22:3671->192.168.31.101:5559 06 10 02 09 00 10 04 00 08 01 00 00 00 00 00 00
13:41:04.332 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: receiver woke from extended idle period of 5,029,608 us
13:41:04.334 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: TP-UART status: Temp. OK
13:41:04.337 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: loop time = 357 us
13:41:09.362 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: receiver woke from extended idle period of 5,029,552 us
13:41:09.364 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: TP-UART status: Temp. OK
13:41:09.367 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: loop time = 2,419 us
13:41:09.984 [INFO   ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:5559: close connection for channel 4 - quit service container /dev/ttyAMA0
13:41:09.985 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 data endpoint 192.168.31.101:5559 closed
13:41:09.985 [DEBUG  ] io.calimero.server.gateway.knx-server: removed connection KNX IP LinkLayer 192.168.31.101:5559 (quit service container /dev/ttyAMA0)
13:41:09.988 [DEBUG  ] io.calimero.server.knx-server: cleanup knx-server control endpoint /dev/ttyAMA0
13:41:09.988 [DEBUG  ] io.calimero.server.knxnetip.KNX IP routing service (eth0 224.0.23.12): close connection - user request
13:41:09.989 [INFO   ] io.calimero.server.knx-server: knx-server control endpoint /dev/ttyAMA0 closed
13:41:09.991 [DEBUG  ] io.calimero.server.knx-server: cleanup knx-server routing service 224.0.23.12
13:41:09.991 [DEBUG  ] io.calimero.server.gateway.knx-server: removed connection KNX IP routing service (eth0 224.0.23.12) (user request)
13:41:09.994 [INFO   ] io.calimero.server.knx-server: knx-server routing service 224.0.23.12 closed
13:41:09.999 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
13:41:10.071 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
13:41:10.086 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
```

</details>

Even longer run: [logs_uart.log](https://github.com/calimero-project/calimero-server/files/14967525/logs_uart.log)

bmalinowsky, 2024-04-14:
Some things I noticed:

* you have a loop somewhere: frame 6.6.3->6.6.10 is repeatedly sent until it is eventually discarded with hop count 0.
* there is this: "4 partial frames discarded, increase max. inter-byte delay ..." which means that the reception of bytes is sometimes delayed. Although Calimero increases the timeout, this takes a while (several frames). And after the second time this has happened, there is not much ETS interaction going on anymore. (So I cannot say if it would have gotten better.)
* connection attempts by ETS: I see this in the log:
    ```
    io.calimero.server.knx-server: address 6.6.2 already assigned
    io.calimero.server.knx-server: address 6.6.3 already assigned
    io.calimero.server.knx-server: assign additional individual address 6.6.4 to channel 32
    ```
   which essentially means that there are moments when all tunneling addresses are in use. Maybe this is why new connection attempts fail?

aleksusklim, 2024-04-14:
I have two new KNX-USB cheap interfaces from China. I will test them (I hope the installed UART HAT won't affect that anyhow, as long as I give a correct XML config)

Then I'll conclude side-by-side logs between UART and USB runs, with the same ETS operations, ensuring restarts of both ETS and Raspberry to provide cleaner logs.
Thank you.

aleksusklim, 2024-04-15:
Surprisingly, both my cheap interfaces are not working with Calimero!

Directly ETS lists and uses them just fine (monitor and flash/download), they both are identified as
`KNX-USB Interface (surface mounter) (Busch-Jaeger Elektro)`
And also ETS shows "Largest telegram length (or smth.) = 15"

But Calimero actually throws
`[ERROR  ] io.calimero.link.connector: initial connection attempt
io.calimero.KNXTimeoutException: waiting for response`
– and no bus communications can be made. Why?

---

Back to the UART HAT: I send via multicast: `read('6/2/1') + write('6/1/1',1,'1.001')`, then `write('6/1/1',0,'1.001') + write('6/2/1',0,'1.001')`

The target relay actuator takes actions, but now I don't get a Response back to KNXUltimate (while in case of GIRA USB it was received) on my read-request.

ETS errors with `Interface not found` after I press `Test` when choosing Calimero.

---

Here are logs of that attempt:
- UART HAT: [logs2_uart_hat.log](https://github.com/calimero-project/calimero-server/files/14977732/logs2_uart_hat.log)
- USB GIRA: [logs2_usb_gira.log](https://github.com/calimero-project/calimero-server/files/14977734/logs2_usb_gira.log)
- USB CHINA: [logs2_usb_china.log](https://github.com/calimero-project/calimero-server/files/14977736/logs2_usb_china.log)

I used the same physical KNX line connector (with firmly attached wires), just plugging it in those three devices, rebooting Raspberry in-between.

aleksusklim, 2024-04-18:
Oh, it would be great if cheap USB ones would work with Calimero!

As for HAT: I am planning to get Raspberry Pi 4 and try it there. Maybe it will be more compatible with newer model.

aleksusklim, 2024-04-19:
It fails with `expected service confirmation msg code 0x4b`
Did I everything correctly?

```
alpi@rpi3:~ $ sudo docker rmi calimeroproject/knxserver
Untagged: calimeroproject/knxserver:latest
Untagged: calimeroproject/knxserver@sha256:5e9fe18cbed5664559fcdd84d5a38ad118c12cc46ff1343506051d56bbe66833
Deleted: sha256:0c6bf6316c03375b8d38c1b15694df826490ec4de7ba74fd8958f0e94402f8ec
Deleted: sha256:057c7e6266130fe2358acb784085bc045bbe604aa05178ecb0a9053ab646253a
Deleted: sha256:495de9cd07eeef5ba86f126986abedb24d71164b2ce085a32a8d2420542a43cb
Deleted: sha256:dc50656af8146e237e8e4d205dbf7bd604dd6e1a29564530de066eb9c761f138
Deleted: sha256:f9a0dbf0f15da4d13b52f909194e5717f8cb060d0a351f8ab049510bdba0cae2
Deleted: sha256:9a0366d2114392dd3c7279f2e53b086701f474cfb7fe1ea000d1b5e9ab44e608
Deleted: sha256:315bd5088587f904ce90b0801a0c8db45648746faf106758d6946f4ef96ae5d0
alpi@rpi3:~ $ sudo docker pull calimeroproject/knxserver
Using default tag: latest
latest: Pulling from calimeroproject/knxserver
622a0779436e: Pull complete 
9a1e23160bca: Pull complete 
c4e788a87cd8: Pull complete 
d2bd141cd659: Pull complete 
5a4b7cae764d: Pull complete 
4f4fb700ef54: Pull complete 
Digest: sha256:fe26f30f15ad7646f621b8edd382749be531bf7761de1a49f053e77fed06f221
Status: Downloaded newer image for calimeroproject/knxserver:latest
docker.io/calimeroproject/knxserver:latest
alpi@rpi3:~ $ sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
11:12:05.934 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
11:12:05.943 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
11:12:06.969 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
11:12:06.994 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
11:12:07.492 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
11:12:07.500 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
11:12:07.506 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
11:12:07.634 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
11:12:07.776 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX' with 1 service container, discovery: listen on [wlan0] send on [wlan0] interfaces
service container Usb-6.6.0:
	server: listen on wlan0 (port 3671), KNX IP routing multicast group 224.0.23.12
	Usb connection: TP1 medium, device 6.6.0
11:12:07.854 [DEBUG  ] io.calimero.server.knx-server: setup Usb subnet 'Usb-6.6.0'
11:12:07.940 [INFO   ] io.calimero.server.gateway.knx-server: activate 'Usb-6.6.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
11:12:07.962 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
11:12:07.965 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
11:12:07.970 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
11:12:08.275 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
11:12:08.364 [TRACE  ] io.calimero.server.knx-server: found network interfaces wlan0 [192.168.31.21]
11:12:08.370 [INFO   ] io.calimero.server.knx-server: discovery endpoint [wlan0] is up and running
11:12:08.445 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
11:12:08.446 [TRACE  ] io.calimero.server.knx-server: Usb-6.6.0 control endpoint bound to 192.168.31.21:3671
11:12:08.494 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on wlan0
11:12:08.500 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
11:12:08.525 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 KNX IP routing service (wlan0 224.0.23.12) is up and running
11:12:08.566 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 secure mgmt/tunneling connections: optional/optional
11:12:08.654 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 tcp service (wlan0 192.168.31.21:3671) is up and running
11:12:08.664 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 control endpoint (wlan0 192.168.31.21:3671) is up and running
11:12:08.997 [DEBUG  ] io.calimero.usb.provider.javax: Attached USB devices:
Root hub
 |--Bus 1 Device 1: ID 1d6b:0002 Linux 6.1.21-v7+ dwc_otg_hcd DWC OTG Controller S/N 3f980000.usb
     High Speed USB
      |--Bus 1 Device 2: ID 0424:9514
          Attached at port 1 (bus:1/1)
          High Speed USB
           |--Bus 1 Device 3: ID 0424:ec00
           |   Attached at port 1 (bus:1/1/1)
           |   High Speed USB
           |--Bus 1 Device 4: ID 145c:1490 KNX USB KNX USB 1518for8C09VG22
               Attached at port 3 (bus:1/1/3)
[... cut ...]
11:12:57.266 [DEBUG  ] io.calimero.server.gateway.knx-server: removed connection KNX IP routing service (wlan0 224.0.23.12) (user request)
11:12:57.273 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
11:12:57.356 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
11:12:57.369 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
alpi@rpi3:~ $ 
```

bmalinowsky, 2024-04-19:
Yes, your steps were completely fine. The stick just trips on the next thing. Because I don't have that particular interface of yours, I can only act on what I see in the log. I again updated the docker image to trace any error when switching the BCU (and subsequently ignore it) and see how far it gets. It would help if you give it a try.

I also find it interesting that the china stick vendor/product ID shows busch-jaeger, that's actually a German manufacturer. AFAIR they weren't that cheap when we used them.

bmalinowsky, 2024-04-19:
No problem, have a nice weekend :)

aleksusklim, 2024-04-22:
Weird.
At first I re-downloaded the docker image to test Chine USB, and it threw errors in console. I tried to connect KNXUltimate in Multicast mode anyway, and it was not working.
But then I pressed some KNX buttons and noticed new log lines.

Then I checked LogicMachine (LM) which was also on my backbone IP segment, and it showed updates for proper group addresses!
So it was working…

I connected KNXUltimate in Tunnel mode and Calimero was indeed working with China USB interface.
And since LM was hearing multicasts, it means my KNXUltimate script somehow broke.
I decided to quickly check GIRA interface – and it flooded the console with a lot of zeros again, on any communication attempt.

I already got a `Raspberry Pi 4B rev 1.2` for testing, so I disassembled the test rig and connected KNX HAT to model PI 4 now, installing the Raspbian 64-bit OS.

Firstly I've tested GIRA again. It worked perfectly in Tunnel mode, in ETS programming (I quickly changed the firmware/settings of a KNX device successfully), and from LM. But KNXUltimate have still not worked in router/multicast; I think this might have something to do with wlan0 wireless interface (I will test more isolated setup)

Secondly I checked China USB – and now it was not working! No tunnel, no ETS, nothing. I think ETS threw "no more connections available" since it failed every single attempt to connect (`Test` button never showed "OK")

Lastly, I tried to use KNX HAT. It worked (but kinda "slowly", compared to GIRA) in Tunnel mode for KNXUltimate and in Multicast for LM, but not in ETS: again, it gave errors like "The device is not responding in time" and "No answer for telegram" (these are not the exact English wording).
Yeah, ETS clearly "sees" the interface itself, but cannot flash/download: it starts, the green bar fills for a tiny bit, and then stuck.

[image]

Also I think ETS could not discover Calimero device on its own (I had to enter IP manually) now. For any setup I've tested! Which is strange because previously it was detecting Calimero quickly, showing it in the interfaces list. (That might be related to other new KNX routers in the same LAN; again I would need a clear environment for testing further)

For reference, this is my config:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX" appData="/usr/app">
  <discovery listenNetIf="wlan0" outgoingNetIf="wlan0" activate="true" />
  <serviceContainer activate="true" routing="true" networkMonitoring="true" udpPort="3671" listenNetIf="wlan0">
    <knxAddress type="individual">6.6.0</knxAddress>
    <routing latencyTolerance="1000">224.0.23.12</routing>
    <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" />
    <!-- knxSubnet type="usb" medium="tp1" knxAddress="6.6.1"></knxSubnet -->
    <knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet>
    <additionalAddresses>
      <knxAddress type="individual">6.6.2</knxAddress>
      <knxAddress type="individual">6.6.3</knxAddress>
      <knxAddress type="individual">6.6.4</knxAddress>
     </additionalAddresses>
  </serviceContainer>
</knxServer>
```

This is the log of Raspberry 3 when China USB was working:
[logs3_china_working.txt](https://github.com/calimero-project/calimero-server/files/15068445/logs3_china_working.txt)

This is the log of Raspberry 4 with GIRA:
[logs3_gira_ets.log](https://github.com/calimero-project/calimero-server/files/15068448/logs3_gira_ets.log)

I did roughly this: tunnel from KNXUltimate, multicast from KNXUltimate, press some KNX buttons, then tried ETS programming.

The same procedure with KNX HAT instead, ETS failed: (the log size is bigger)
[logs3_uart_ets.log](https://github.com/calimero-project/calimero-server/files/15068449/logs3_uart_ets.log)

And for China USB one; it failed badly, so I could not connect neither from KNXUltimate nor from ETS:
[logs3_china_ets1.log](https://github.com/calimero-project/calimero-server/files/15068454/logs3_china_ets1.log)

Also it did not stop the execution when I was mashing Ctrl+C into the console!
[logs3_china_ets2.log](https://github.com/calimero-project/calimero-server/files/15068455/logs3_china_ets2.log)
Flooded with `java.lang.Runtime: Runtime.exit() called with status: 130`, but eventually closed anyway.

---

Say… have you updated your 64-bit build of the docker image? Was I testing the correct version?

Also, I cannot be sure that my previous attempt to run China USB on Raspberry 3 was actually unsuccessful if Multicast in KNXUltimate might have been already broken; I cannot say for sure that I've tested Tunnel mode back then.

bmalinowsky, 2024-04-24:
I only updated the armv7 version -> now I've updated all other platforms, too; your new raspi (the Linux 6.6.20+rpt-rpi-v8 one) probably downloads another docker image. That's why the china stick on that raspi showed the old error for initial connection attempt. -> It should now skip over those exceptions like the armv7 version.

I see in the log (e.g., logs3_china_ets2), that there are several warnings "could not accept new connection (maximum reached)", so all your configured 3 additional addresses are already used up by other established connections. The ETS or any other client won't be able to connect.

When you use the TPUART, can you try to disable routing in the server config? You still have a loop in there, where frames addressed to 6.6.10 get looped around (i.e. the server receives them again from the client side which doesn't make sense). And ETS uses a TCP tunnel so that should be no problem.

> java.lang.Runtime: Runtime.exit() called with status: 130

This comes from the Jvm, meaning the process received SIGTERM (like Ctrl+C), and the server log shows it initiates  shutdown. The reason it takes exactly 30 seconds for shutdown is that you have 3 open connections which don't respond to the disconnect request. Those requests are sent sequentially, each reaching its timeout of 10 seconds (-> 3 * 10 = 30). I probably could do them in parallel, which would avoid the scaling effect.

aleksusklim, 2024-04-24:
> I only updated the armv7 version

Okay, I'll test again, then!

> there are several warnings "could not accept new connection (maximum reached)", so all your configured 3 additional addresses are already used up by other established connections

They are "broken", for me it's something like "ETS/KNXUltimate attempts to connect → fails → I try again → fails → I try again → whoops, no more connections!"

> the TPUART, can you try to disable routing in the server config? You still have a loop in there, where frames addressed to 6.6.10 get looped around

I don't understand where the loop is coming from, but in worst case I could just run another routed Calimero on the same device, pointing it to the gateway of the first one as its TP target, right?

bmalinowsky, 2024-04-24:
Regarding ETS downloads: depending on the ETS version, there is a configuration like _Setting -> Troubleshooting -> Enable extended logging_. (Maybe one has to restart ETS). ETS logs its stuff in the _AppData\Local_ folder (e.g., mine is _C:\Users\b\AppData\Local\KNX\ETS6\Log\ETS6.log_) and its output hasn't much localization. You can take a look there. Or remove that file, then try flashing a device and attach the resulting log file here.

Regarding routing: you said before that you have other KNX routers, so if they use the same multicast group `224.0.23.12`, that would be a cause for looping. There should be only a single multicast forwarder (for a mcast group) for any KNX subnet.

aleksusklim, 2024-04-24:
I've collected logs from Raspberry 3 again, with enabled debugging for ETS5 and with disabled routing in XML.

Raspberry was connected with eth0 directly to my laptop with ETS, making a private network.

Config:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX" appData="/usr/app">
  <discovery listenNetIf="eth0" outgoingNetIf="eth0" activate="true" />
  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="eth0">
    <knxAddress type="individual">6.6.0</knxAddress>
    <!-- routing latencyTolerance="1000">224.0.23.12</routing -->
    <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" />
    <knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet>
    <!-- knxSubnet type="usb" medium="tp1" knxAddress="6.6.1"></knxSubnet -->
    <additionalAddresses>
      <knxAddress type="individual">6.6.2</knxAddress>
      <knxAddress type="individual">6.6.3</knxAddress>
      <knxAddress type="individual">6.6.4</knxAddress>
     </additionalAddresses>
  </serviceContainer>
</knxServer>
```

Setup:
```bash
alpi@rpi3:~ $ sudo docker rmi calimeroproject/knxserver
Untagged: calimeroproject/knxserver:latest
Untagged: calimeroproject/knxserver@sha256:c60b5799f4f3cf9ad3a23eff4aa2567f5dd2828061dfc3526c54c241aac576f6
Deleted: sha256:3c80409804e9d8da210c17e58756a08c96cfcd3d454f47af4ede52d1fabfd429
Deleted: sha256:69f1424c2fe210100b39a839b7f31149b8e5f1ec22793c08f68420238ea4f492
Deleted: sha256:fe86cd882f905af40cf86ab1b5a828d5b55f759d2fe88105d3c6f26c2edad221
Deleted: sha256:209e92ec964c827a5db54afedd4daf9f199781a97289bb91309dd7d003319265
Deleted: sha256:7532c6b90892f90cbe4935235566ae54724f4c8d05e68cf81f2031b1ef4cb9e3
Deleted: sha256:9a0366d2114392dd3c7279f2e53b086701f474cfb7fe1ea000d1b5e9ab44e608
Deleted: sha256:315bd5088587f904ce90b0801a0c8db45648746faf106758d6946f4ef96ae5d0
alpi@rpi3:~ $ sudo docker pull calimeroproject/knxserver
Using default tag: latest
latest: Pulling from calimeroproject/knxserver
622a0779436e: Pull complete 
9a1e23160bca: Pull complete 
ffac431f6856: Pull complete 
711d29ff1936: Pull complete 
39ebe57858ab: Pull complete 
4f4fb700ef54: Pull complete 
Digest: sha256:2d897604a8abe3b2f70d6cbf7faed32ed4d970bfe5e2487c62136b564a9b85bb
Status: Downloaded newer image for calimeroproject/knxserver:latest
docker.io/calimeroproject/knxserver:latest
alpi@rpi3:~ $ sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml | tee Desktop/logs3_x86_gira1.log
```

Two attempts with GIRA, both apparently failed to communicate with the bus properly: for some reason, when I press a KNX button I see
`EP 1 IN is not a KNX USB report (wrong report ID 0x0): 00000000000000000000024dff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000`
– as if receive events were generated, but with bad data.
(Now I am sure no external inference could influence on wires)

ETS failed; actually I couldn't flash at all, not even with china or uart.
Also my ETS crashed several times, probably because of enabled logging.

Logs from Raspberry:
[logs3_x86_gira1.log](https://github.com/calimero-project/calimero-server/files/15093071/logs3_x86_gira1.log)
[logs3_x86_gira2.log](https://github.com/calimero-project/calimero-server/files/15093072/logs3_x86_gira2.log)

Corresponding logs from ETS:
[logs3_ETS5_gira1.log](https://github.com/calimero-project/calimero-server/files/15093073/logs3_ETS5_gira1.log)
[logs3_ETS5_gira2.log](https://github.com/calimero-project/calimero-server/files/15093074/logs3_ETS5_gira2.log)

Then I tried china stick, and it looked like connected, and KNXUltimate showed some events, but threw errors on them.
(By chance, can you recommend a simple software tool that has implemented tunnel and multicast protocols correctly, so we can use it as a reliable diagnostic tool?)

ETS failed flashing, and also couldn't show anything at bus monitor; sometimes it refused to connect with random reasons.

Log from Raspberry:
[logs3_x86_china3.log](https://github.com/calimero-project/calimero-server/files/15093078/logs3_x86_china3.log)

Log from ETS:
[logs3_ETS5_china3.log](https://github.com/calimero-project/calimero-server/files/15093079/logs3_ETS5_china3.log)

Finally, KNX HAT.
It worked initially, as before: KNXUltimate succeeds, sending and receiving telegrams.
ETS bus monitor worked.

But flashing hangs. I tried to use "full download" with pressing the programming button – but even that did not help, the LED didn't even stop blinking.

Log from Raspberry:
[logs3_x86_uart4.log](https://github.com/calimero-project/calimero-server/files/15093083/logs3_x86_uart4.log)

Log from ETS:
[logs3_ETS5_uart4.log](https://github.com/calimero-project/calimero-server/files/15093086/logs3_ETS5_uart4.log)

I'm up to repeat this on Rasberry 4, exactly noting all actions I am doing externally in the process of testing.

bmalinowsky, 2024-04-25:
It's interesting that the Gira stick now also exhibits the problem with init like the china one and doesn't get a response (the several exceptions in the beginning waiting for msg code 0x4b). And it does not get any correct knx frames. This did work in your previous logs.

I hadn't implemented the workaround for busmonitor as for the normal link, so that's why ETS wouldn't show anything. The thing is this does require the init sequence, so at least the china sticks probably won't busmonitor.

The UART does have a lot of "reset receive buffer after ..." lines, meaning that the bytes which are received are delayed (inter-byte delay).

To be continued...

bmalinowsky, 2024-04-25:
Several things:

I updated the docker images to address some tangent issues you reported: 
* the shutdown of connections is now parallel, so the wait time should be shortened
* busmonitor got the same treatment as the normal network link
* I cleaned up the stacktraces a bit

For TP-UART, there are 2 options I can think of:

1) using setserial: check `setserial /dev/ttyAMA0` on your raspi, it probably doesn't have the low latency flag set.
    Use `setserial /dev/ttyAMA0 low_latency`, and see if that makes a difference. This change should also be reflected inside a docker container (you can check that by starting the container with the additional option `--entrypoint ash` (e.g., `docker run --rm -ti --net host -v /home/b/server-config.xml:/usr/app/server-config.xml --privileged --device /dev/ttyAMA0 --entrypoint ash calimeroproject/knxserver)`, and inside install _setserial_ via `apt install setserial` and check the port again).
2) if 1. doesn't help, you can address the inter-byte delay I mentioned before using a command-line option: as that is inside the docker container, too, again use `--entrypoint ash`, then inside the container `export CALIMERO_SERVER_OPTS="-Dio.calimero.serial.tpuart.maxInterByteDelay=20000"`. Then still inside the container start the server `calimero-server-3.0-SNAPSHOT/bin/calimero-server -vvv server-config.xml`.
   In the log you should see an entry like "io.calimero.serial.tpuart: using io.calimero.serial.tpuart.maxInterByteDelay of 20,000 us"

For 1., this is a setting that needs to be done on every restart of the raspi, and for 2. it is specific to a single docker container.  So if you remove the container, you have to redo it. But the goal is to see if any of those options help, then we can take it further.

Regarding tools, since you use Windows, you could try Net'n'Node by Weinzierl, it's nice. (Free, you need to register). If you want to see decoded message traffic between applications, you can install Wireshark (in the display filter type "kip").
(You can also use the knxtools docker, because I know what does work and what not; the disadvantage is that a problem in the core library will affect both the tools and the server.)

aleksusklim, 2024-04-29:
Looks like the cheap sdcard in my Raspberry PI 3 has failed – so I got a Samsung one, reinstalling 32-bit Raspbian again from scratch.

I did everything you told me, but unable to get past last errors. I even tried disabling detailed logs (omit `-vvv`) and disabled "disruption buffer".

I decided to type Enter into the console so that later I can see when I did something. I annotated my logs with that information:

- `// KNXUltimate` – I connect tunnel from KNXUltimate, then it sends one write and one read; then waits and sends two more writes.
- `// BUTTONS` – while KNXUltimate is connected, I press two physical buttons on KNX device, sending two telegrams that I expect to see in KNXUltimate.
- `// STOP` – I send Ctrl+C and wait for the process to stop.

This is KNXUltimate configuration:
```js
knxUltimateClient = new knx.KNXClient({
    ipAddr: "192.168.8.24",
    ipPort: "3671",
    physAddr: "6.6.3",
    suppress_ack_ldatareq: false,
    loglevel: "warn",
    localEchoInTunneling: true,
    hostProtocol: 'TunnelUDP',
    isSecureKNXEnabled: false,
    jKNXSecureKeyring: "",
    localIPAddress: "192.168.8.20",
    KNXEthInterface: "Manual",
});

…

setTimeout(() => {
    knxUltimateClient.read('6/2/1');
    knxUltimateClient.write('6/1/1', 1, "1.001")
}, 5000);
setTimeout(() => {
    knxUltimateClient.write('6/1/1', 0, "1.001")
    knxUltimateClient.write('6/2/1', 0, "1.001")
}, 10000);

```

Calimero config:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX" appData="/usr/app">
  <discovery listenNetIf="enxb827ebeb0c43" outgoingNetIf="enxb827ebeb0c43" activate="true" />
  <serviceContainer activate="true" routing="false" networkMonitoring="true" udpPort="3671" listenNetIf="enxb827ebeb0c43">
    <knxAddress type="individual">6.6.0</knxAddress>
    <!-- routing latencyTolerance="1000">224.0.23.12</routing -->
    <!-- disruptionBuffer expirationTimeout="30" udpPort="5555-5559" / -->
    <!--knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet -->
    <knxSubnet type="usb" medium="tp1" knxAddress="6.6.1"></knxSubnet>
    <additionalAddresses>
      <knxAddress type="individual">6.6.2</knxAddress>
      <knxAddress type="individual">6.6.3</knxAddress>
      <knxAddress type="individual">6.6.4</knxAddress>
     </additionalAddresses>
  </serviceContainer>
</knxServer>
```

The first is the china stick (the second one; previously both two logs were from the first one):
[log4_china.txt](https://github.com/calimero-project/calimero-server/files/15149804/log4_china.txt)
KNXUltimate cannot send to the bus (not effect), cannot read from the bus (not response) but sometimes can hear the bus (events from button presses), but they seem to be delayed sometimes in tunnel mode

The next is Gira:
[log4_gira.txt](https://github.com/calimero-project/calimero-server/files/15149809/log4_gira.txt)
I see zeroes in console on button press again.
Similarly, I can't do any actual bus communication.

Finally, KNX HAT:
[log4_uart.txt](https://github.com/calimero-project/calimero-server/files/15149811/log4_uart.txt)
KNXUltimate is working, as before: relay switches on my write, and I see telegrams on button press.

But this is another attempt for ETS:
[log4_uart_ets.txt](https://github.com/calimero-project/calimero-server/files/15149812/log4_uart_ets.txt)

If fails to do a "full download" of the program (where I need to press a button on the device).
The line `[INFO   ] io.calimero.serial.tpuart: using io.calimero.serial.tpuart.maxInterByteDelay of 20,000 us` is there.

What else?
Why Gira sends nonsense again?
Why China sticks were working once previously?

In multicast mode everything feels to be faster, but telegrams are not sent anyway:
[log4_china_route.txt](https://github.com/calimero-project/calimero-server/files/15149816/log4_china_route.txt)

aleksusklim, 2024-05-02:
For some reason I fail to flash/download a device with ETS via KNX-USB interface (directly connected to the computer) if device's physical address does not have zero in the second position, e.g. `6.6.20` (and thus attached to the third line segment in topology hierarchy).
I even updated ETS to v5.7.7

To make programming work, I had to move the device to the second line in topology, and ETS assigned `6.0.1` to it; then if I change the address to, for example, `6.0.20` – flashing via china USB stick works.

I'm not sure what change in the project caused this (physically I have only one isolated KNX line, with Raspberry being the only bus coupler on it), but I decided to check if this solved the main issue?

Unfortunately not: Calimero (with adjusted `6.0.x` addresses) still fails in ETS with KNX HAT, both in router and in tunnel mode.
Other than in ETS, it looks like KNX HAT works properly, routing normal telegrams in both directions.

Calimero with china USB interface fail not only in ETS (cannot even disable device's programming mode when full downloading), but also cannot route outer telegrams to KNX bus. I can see only messages from KNX gets dispatched to IP.

aleksusklim, 2024-05-02:
Oh, wait!
If Calimero is running on UART – then I cannot program with ETS even if I am connected via KNX-USB from a computer! It somehow floods the bus…
Even with routing disabled. Probably this is why I couldn't flash normally in my latest experiment (because with disconnected everything I seem to be able to use third line hierarchy)

If Calimero is not running – the connected KNX line to HAT does not interfere on its own.

aleksusklim, 2024-05-02:
After testing GIRA with ETS and found no errors whatsoever, I thought that maybe it fails with Raspberry PI 3 because of low power?
For Raspberry PI 4 I have a better power supply, so I updated Calimero there to try x64 version.

Nope, zeros on incoming KNX messages again!

```
alpi@rpi4:/srv/my/dock/app $ sudo docker pull calimeroproject/knxserver
Using default tag: latest
latest: Pulling from calimeroproject/knxserver
579b34f0a95b: Pull complete
5cb4808b3474: Pull complete
58f472f04910: Pull complete
b8e9c10f91ba: Pull complete
8f560a992f44: Pull complete
4f4fb700ef54: Pull complete
Digest: sha256:f070e02c4211ae770461ff99eff9895d2c315254801946663a85969aa695c3d2
Status: Downloaded newer image for calimeroproject/knxserver:latest
docker.io/calimeroproject/knxserver:latest
alpi@rpi4:/srv/my/dock/app $ sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
13:59:45.019 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
13:59:45.022 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:59:45.314 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
13:59:45.322 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
13:59:45.485 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
13:59:45.488 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
13:59:45.490 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
13:59:45.537 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
13:59:45.586 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX' with 1 service container, discovery: listen on [wlan0] send on [wlan0] interfaces
service container Usb-6.6.0:
        server: listen on wlan0 (port 3671), KNX IP routing multicast group 224.0.23.12
        Usb connection: TP1 medium, device 6.6.0
13:59:45.611 [DEBUG  ] io.calimero.server.knx-server: setup Usb subnet 'Usb-6.6.0'
13:59:45.637 [INFO   ] io.calimero.server.gateway.knx-server: activate 'Usb-6.6.0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
13:59:45.644 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
13:59:45.645 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
13:59:45.647 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
13:59:45.778 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
13:59:45.811 [TRACE  ] io.calimero.server.knx-server: found network interfaces wlan0 [192.168.31.21]
13:59:45.814 [INFO   ] io.calimero.server.knx-server: discovery endpoint [wlan0] is up and running
13:59:45.819 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
13:59:45.828 [TRACE  ] io.calimero.server.knx-server: Usb-6.6.0 control endpoint bound to 192.168.31.21:3671
13:59:45.833 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on wlan0
13:59:45.840 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
13:59:45.851 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 KNX IP routing service (wlan0 224.0.23.12) is up and running
13:59:45.855 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 secure mgmt/tunneling connections: optional/optional
13:59:45.875 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 tcp service (wlan0 192.168.31.21:3671) is up and running
13:59:45.878 [INFO   ] io.calimero.server.knx-server: Usb-6.6.0 control endpoint (wlan0 192.168.31.21:3671) is up and running
13:59:46.032 [DEBUG  ] io.calimero.usb.provider.javax: Attached USB devices:
Root hub
 |--Bus 1 Device 1: ID 1d6b:0002 Linux 6.6.20+rpt-rpi-v8 xhci-hcd xHCI Host Controller S/N 0000:01:00.0
 |   High Speed USB
 |    |--Bus 1 Device 2: ID 2109:3431 USB2.0 Hub
 |        Attached at port 1 (bus:1/1)
 |        High Speed USB
 |         |--Bus 1 Device 3: ID 135e:0022 Gira Giersiepen GmbH & Co. KG KNX-USB Data Interface
 |             Attached at port 2 (bus:1/1/2)
 |             Full Speed USB
 |--Bus 2 Device 1: ID 1d6b:0003 Linux 6.6.20+rpt-rpi-v8 xhci-hcd xHCI Host Controller S/N 0000:01:00.0
     Super Speed USB
13:59:46.037 [INFO   ] io.calimero.usb.provider.javax: Found 1 KNX USB devices:
 |--Bus 1 Device 3: ID 135e:0022 Gira Giersiepen GmbH & Co. KG KNX-USB Data Interface
 |   Attached at port 2 (bus:1/1/2)
 |   Full Speed USB
13:59:46.058 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: USB interface 00, setting 0
13:59:46.059 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 2 OUT
13:59:46.060 [TRACE  ] io.calimero.usb.provider.javax.135e:0022: EP 1 IN
[... cut ...]
14:00:04.584 [INFO   ] io.calimero.server.knx-server: knx-server routing service 224.0.23.12 closed
14:00:04.587 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
14:00:04.606 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
14:00:04.613 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
alpi@rpi4:/srv/my/dock/app $
```

bmalinowsky, 2024-05-07:
ETS checks the address in the lines where it wants to download, so that makes sense. But in the end, having assigned the same address to the USB stick as the server address should be OK. Because that is what the server does on its own (not for EMI1 though). I have an implementation of the EMI1 address check on the radar, but since the setup sequence does not work for your stick for some reason, it won't make a difference.

> If Calimero is running on UART – then I cannot program with ETS even if I am connected via KNX-USB from a computer! It somehow floods the bus…

That's interesting, because normally this only happens with routing (a routing loop over Ethernet).

Why the GIRA worked and now doesn't I cannot explain, also because nothing changed in the software wrt that. In the end, I only suppressed the exceptions.

Did you try another tool, like the Weinzierl one, to init the USB sticks? It also shows the used init sequences. One could compare those. Or see if they do anything useful.

aleksusklim, 2024-05-08:
I used 2 computers to connect my 2 china KNX-USB to 2 ETS.
One ETS is used as "Bus monitor" only.
Another ETS is used to flash/download a device, with extended logs enabled.

KNX line is isolated. Raspberry PI 3 connects via Wi-Fi with UART KNX-HAT on it, connected to this KNX bus.
I disabled routing (along with "networkMonitoring") and did NOT connect to Calimero anyhow:

```xml
alpi@rpi3:/srv/my/dock/app $ cat server-config.xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX" appData="/usr/app">
  <discovery listenNetIf="wlan0" outgoingNetIf="wlan0" activate="true" />
  <serviceContainer activate="true" routing="false" networkMonitoring="false" udpPort="3671" listenNetIf="wlan0">
    <knxAddress type="individual">6.0.0</knxAddress>
    <routing latencyTolerance="1000">224.0.23.12</routing>
    <!-- disruptionBuffer expirationTimeout="30" udpPort="5555-5559" / -->
    <knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet>
    <!-- knxSubnet type="usb" medium="tp1" knxAddress="6.0.51"></knxSubnet -->
    <additionalAddresses>
      <knxAddress type="individual">6.0.52</knxAddress>
      <knxAddress type="individual">6.0.53</knxAddress>
      <knxAddress type="individual">6.0.54</knxAddress>
     </additionalAddresses>
  </serviceContainer>
</knxServer>
sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml >../log5_uart.txt
```

I try to program a device ("partial", without pressing a programming button).
When Calimero docker is active – ETS fails. I restart it and close Calimero – is succeeds.

Logs from ETS with Calimero active: [log5_ets_fail.log.txt](https://github.com/calimero-project/calimero-server/files/15251116/log5_ets_fail.log.txt)
Logs from Bus Monitor during this: [log5_bus_fail.xml.txt](https://github.com/calimero-project/calimero-server/files/15251118/log5_bus_fail.xml.txt)
Logs from Calimero itself: [log5_uart.txt](https://github.com/calimero-project/calimero-server/files/15251119/log5_uart.txt)

Logs from restarted ETS after stopping Calimero: [log5_ets_ok.log.txt](https://github.com/calimero-project/calimero-server/files/15251122/log5_ets_ok.log.txt)
Logs from Bus Monitor again: [log5_bus_ok.xml.txt](https://github.com/calimero-project/calimero-server/files/15251124/log5_bus_ok.xml.txt)
(You should probably rename XML files from bus monitor to load them back into ETS)

I hope you can deduce something from this!

bmalinowsky, 2024-05-09:
Thank you! Yes, I found a regression in frame dispatching from a recent refactoring, which I think is responsible for the issue you observe.

I pushed new docker images.

aleksusklim, 2024-05-10:
Yay, it's working!

For KNX HAT, I can use KNXUltimate in Tunnel and Multicast modes to read, write and hear.
I also was able to program with ETS without major errors. Here is what I observed:
1. Until I set `low_latency` + `maxInterByteDelay`, some read-request responses for KNXUltimate were lost. Also, sometimes there was a significant pause between my commands and actual sending data over the bus: for example, I give 2 commands, wait 5 seconds and give 2 more; but the relay quickly turns on/off/on/off executing all four only after yet another 5 seconds! Looks like `setserial` (or `maxInterByteDelay`) fixed this.
2. While I used Bus Monitor in ETS (connected as Tunnel) – I could not connect another Tunnel from KNXUltimate, no matter what "additional address" would I choose. Generally, I find Multicast mode more reliable.
3. Sometimes ETS fails to connect to Calimero gateway in Tunnel mode! It just disappears from the list after I click to select it. During this split-second, ETS shows empty list of "additional addresses" to pick. Adding manually via IP works I think; also the discovery may randomly work too, after restarting both ETS and Calimero.
4. China USB still cannot send to KNX bus, but it is able to hear telegrams back.

Here is a part of Calimero logs when I tried to pick its tunnel in ETS but it disappeared:
```
08:27:03.921 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: TP-UART status: Temp. OK
08:27:03.922 [TRACE  ] io.calimero.serial.tpuart:/dev/ttyAMA0: loop time = 169 us
08:27:05.561 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:44888 06 10 02 03 00 0e 08 01 00 00 00 00 00 00
08:27:05.563 [DEBUG  ] io.calimero.server.knx-server: responses use route-back control endpoint 192.168.31.101:44888
08:27:05.565 [INFO   ] io.calimero.server.knx-server: send KNXnet/IP description to 192.168.31.101:44888: "Calimero KNX", KNX address 6.0.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address b8:27:eb:be:59:16, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 6.0.52, 6.0.53, 6.0.54, KNX manufacturer ID 0x0, data 0x626d323031312020
08:27:05.608 [INFO   ] io.calimero.server.knx-server: accepted /dev/ttyAMA0 tcp connection 192.168.31.101:28133
08:27:05.609 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 02 05 00 1c 08 02 00 00 00 00 00 00 08 02 00 00 00 00 00 00 06 04 02 00 60 35
08:27:05.610 [DEBUG  ] io.calimero.server.knx-server: responses to client use route-back control endpoint 192.168.31.101:28133
08:27:05.610 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0: setup data endpoint (TCP, channel 8) for connection request from 192.168.31.101:28133
08:27:05.611 [DEBUG  ] io.calimero.server.knx-server: responses use route-back data endpoint 192.168.31.101:28133
08:27:05.613 [INFO   ] io.calimero.server.knx-server: assign additional individual address 6.0.53 to channel 8
08:27:05.615 [DEBUG  ] io.calimero.server.knx-server: created socket on 192.168.28.184:52297
08:27:05.620 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 data endpoint KNX IP LinkLayer 192.168.31.101:28133 (TCP, channel 8, 6.0.53) is up and running
08:27:05.621 [DEBUG  ] io.calimero.server.gateway.knx-server: established connection KNX IP LinkLayer 192.168.31.101:28133 (TCP, channel 8, 6.0.53)
08:27:05.630 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 04 24 00 0d 04 08 00 00 08 00 01
08:27:05.632 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: received tunneling-feat.set enable feature info service 01
08:27:05.633 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: respond with tunneling-feat.res enable feature info service 01
08:27:05.647 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 04 22 00 0c 04 08 01 00 06 00
08:27:05.648 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: received tunneling-feat.get individual address
08:27:05.649 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: respond with tunneling-feat.res individual address 6035
08:27:05.669 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 04 22 00 0c 04 08 02 00 07 00
08:27:05.671 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: received tunneling-feat.get max apdu length
08:27:05.672 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: respond with tunneling-feat.res max apdu length 000f
08:27:05.689 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 04 22 00 0c 04 08 03 00 02 00
08:27:05.691 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: received tunneling-feat.get device descriptor type 0
08:27:05.692 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: respond with tunneling-feat.res device descriptor type 0 091a
08:27:05.717 [TRACE  ] io.calimero.server.knx-server: /dev/ttyAMA0 received 192.168.31.101:28133 06 10 02 09 00 10 08 00 08 02 00 00 00 00 00 00
08:27:05.718 [DEBUG  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:28133: close connection for channel 8 - client request
[... cut ...]
09:08:17.435 [TRACE  ] io.calimero.server.knxnetip.KNX IP routing service (wlan0 224.0.23.12): discard multicast loopback cEMI frame: 6.6.4->6/2/1 L_Data.ind, low priority hop count 6, tpdu 00 80
09:08:17.438 [TRACE  ] io.calimero.usb.provider.javax.145c:1490: sending I/O request 0113110008000901030000110c00003201d10080
09:08:17.438 [TRACE  ] io.calimero.server.knxnetip.KNX IP LinkLayer 192.168.31.101:5559: received service ack 3 from 192.168.31.101:5559 (channel 3)
09:08:17.444 [TRACE  ] io.calimero.link.145c:1490: send to 6/2/1 succeeded
```

bmalinowsky, 2024-05-10:
Good to know. I also see that `setserial` does speed up serial timing on my quite old raspberry 2 B. Running in a docker container might incur additional overhead wrt that (but I don't know what docker does internally).

If you run a busmonitor, you cannot connect via tunneling. It's not supported by KNX, the hardware is setup differently for each mode.

I will take a look at the logs...

bmalinowsky, 2024-05-13:
In the logs there is again a dangling client connection, you seem to have a lot of them. Anyway, the server should not delay processing due to such things, so now most client tunneling interaction is asynchronous.

I think I found the culprit with the EMI1 issue, so there should be some progress.

You mentioned multicast being more reliable: if you communicate on the same host (i.e., using a loopback interface), or in your own LAN, a multicasted datagram basically never gets lost. So there is not really a downside in using that type of communication. Just as a side note: the KNX multicasting protocol in itself is an unreliable protocol.

(Docker images updated.)

aleksusklim, 2024-05-13:
I was thinking about using the same host with "reliable" Tunnel connection with Calimero by using your "disruption buffer". So that if my client crashes and restart – it would not miss anything.
But looks like this way would:
- Make reconnection randomly fail if Calimero will think that it has no spare "additional addresses"
- Make clients connected via tunneling randomly hang, potentially affecting my own master connection
- Make ETS unable to use Bus Monitor because my own connection is already used, since you said the monitoring mode cannot be used in parallel with tunneling

So, what are my options?
Scratch "disruption buffer", use localhost Multicast, do not ever crash (this is possible if I'll use another process to separate socket functions and main application logic, effectively making a TCP proxy between me and multicasts to Calimero), but leave Tunnel with a few "additional addresses" just for ETS to optionally use?

bmalinowsky, 2024-05-16:
> * Make reconnection randomly fail if Calimero will think that it has no spare "additional addresses"

In your config you use 3 additional addresses, which is on the low side if you also want ETS to connect, and test things, or have software which quickly connects/disconnects (or forgets to disconnect). 

> * Make clients connected via tunneling randomly hang, potentially affecting my own master connection

Is this a new thing? Or is this related to before with the dangling connection of another client? (Because I did work around that.) 

> * Make ETS unable to use Bus Monitor because my own connection is already used, since you said the monitoring mode cannot be used in parallel with tunneling

Yes, using busmonitor and link-layer at the same time requires two different KNX HW modules. Bus monitoring is a diagnostic tool if you suspect problems with a KNX device or the KNX bus itself. So unless you are interested in the ACKs etc.,  the ETS group monitor or bus activity tab is sufficient, and does work with other clients.

aleksusklim, 2024-05-16:
When I was testing updated docker images with KNX HAT on Raspberry 3, for some reason KNXUltimate was not heard in Calimero: all my sent Multicast messages didn't even show up in console. But KNXUltimate could hear back telegrams from the bus.

But since Logic Machine in the same Wi-Fi could both send and receive routed messages, I concluded either a misconfiguration of KNXUltimate (which I did not update in any way) or my LAN.

Now I decided to test Raspberry 4, and it turned out to be a huge success!!

- UART: everything is working, including discovery and programming with ETS, even without messing with `setserial` or `maxInterByteDelay` (but I haven't tried to flood it with packets)
- GIRA: sending and receiving is working flawlessly! I don't know what was causing those zero dumps; if you did nothing additional to tackle this, then I still suspect a hardware wiring or power supply side-effects
- Both China USB: Receiving is working, but sending is not. Calimero thinks it could send, but relay on the bus cannot hear my commands or read-requests.

I would close this Issue after I'd test PI 3 again.
But what should I do with cheap sticks? I tried to register in Weinzierl to get `Net'n Node 5` working, but they didn't seem to send me an e-mail with the license or anything so far.

What other software can get use of KNX-USB interfaces?
Is it possible to dump ETS communications with USB anyhow?

---

<details><summary>Current Logs:</summary>

```
alpi@rpi4:/srv/my/dock/app $ cat server-config.xml
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX" appData="/usr/app">
  <discovery listenNetIf="wlan0" outgoingNetIf="wlan0" activate="true" />
  <serviceContainer activate="true" routing="true" networkMonitoring="true" udpPort="3671" listenNetIf="wlan0">
    <knxAddress type="individual">6.6.0</knxAddress>
    <routing latencyTolerance="1000">224.0.23.12</routing>
    <knxSubnet type="usb" medium="tp1" knxAddress="6.6.0"></knxSubnet>
    <!-- knxSubnet type="tpuart" medium="tp1">/dev/ttyAMA0</knxSubnet -->
    <additionalAddresses>
      <knxAddress type="individual">6.6.2</knxAddress>
      <knxAddress type="individual">6.6.3</knxAddress>
      <knxAddress type="individual">6.6.4</knxAddress>
     </additionalAddresses>
  </serviceContainer>
</knxServer>
alpi@rpi4:/srv/my/dock/app $ sudo docker rmi calimeroproject/knxserver
Untagged: calimeroproject/knxserver:latest
Untagged: calimeroproject/knxserver@sha256:f070e02c4211ae770461ff99eff9895d2c315254801946663a85969aa695c3d2
Deleted: sha256:135544bd4cb4bfacdd43987c1edaeec7f6a6b34c392b2adbc06288e0a26e26a2
Deleted: sha256:1674f84ed6f60946d91f1d3acea7a7eb8c6715a4e708f4ce275aed54972ee937
Deleted: sha256:28d227b0b54ebec867e6e1a66e1bb16d4e1b8032543b652f4b3f73a7236447ae
Deleted: sha256:ea2b12ec49b76a8f49597f7537e0655d1d90b5e68c5501c0e96aa03220ff7bc6
Deleted: sha256:a0a20e694ed82950f3f488870e62fcb1f5fb5d4ee6df1aee78834524bc3c8e79
Deleted: sha256:b4d8cda4490797039d58f99c05f9c7f56f264e9693a6c24c0eafa0a486341f74
Deleted: sha256:5f4d9fc4d98de91820d2a9c81e501c8cc6429bc8758b43fcb2cd50f4cab9a324
alpi@rpi4:/srv/my/dock/app $ sudo docker pull calimeroproject/knxserver
Using default tag: latest
latest: Pulling from calimeroproject/knxserver
579b34f0a95b: Pull complete
5cb4808b3474: Pull complete
e1e9e719a69e: Pull complete
6d4f0d375038: Pull complete
b3a605520c4f: Pull complete
4f4fb700ef54: Pull complete
Digest: sha256:3a9599f043e4aa39f3fac473d616ea13c0f95cfb690beedfcbc5d63ea7766c0b
Status: Downloaded newer image for calimeroproject/knxserver:latest
docker.io/calimeroproject/knxserver:latest
alpi@rpi4:/srv/my/dock/app $ sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml
```

</details>

<details><summary>China USB:</summary>

```
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
15:09:55.170 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
15:09:55.174 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
[... 884 lines cut ...]
</details>

---

> In your config you use 3 additional addresses, which is on the low side if you also want ETS to connect, and test things, or have software which quickly connects/disconnects (or forgets to disconnect).

I intentionally set a low amount to know about potential problems.
Even if I'd set 100 of them, if there is a permanent leak of addresses – the list would become exhausted anyway!
Do you think I should give, for example, 16 – and then test a lot of simultaneous connections to confirm reliability?
The estimation is "at most this many connections must exist at any given time, including those that were abruptly closed less than 30 seconds ago", right?

> Is this a new thing? Or is this related to before with the dangling connection of another client? (Because I did work around that.)

I think this is older one… When I could not reconnect KNXUltimate back to Calimero (cannot remember the exact scenario)
By the way, can you tell a bit more about the disruption buffer? Specifically, why are you tracking by source porn numbers and not by source IP address?
[... cut ...]
> Yes, using busmonitor and link-layer at the same time requires two different KNX HW modules.
> So unless you are interested in the ACKs etc., the ETS group monitor or bus activity tab is sufficient, and does work with other clients.

Wait, there are `Group Monitor` and `Bus Monitor` under `Monitor` section in the `Bus` tab of ETS.
Is this "Bus Monitor" what we're talking about?

Then why and how it works in Routing mode, when I choose my local adapter rather than IP or USB interfaces to use?
Or ETS just emulates a bus monitoring when multicasting?

aleksusklim, 2024-05-20:
> You can try the new images

I think it is working now!

Both receiving and sending works, with China and HAT.
Both in Tunneling and Routing modes, but.

Strange thing is, my KNXUltimate still fails to get use of sending in multicast mode, only receiving is working. What I have noticed (with Raspberry PI 3) so far:
- When I send via Logic Machine – Calimero sees;
- When I connect with Tunneling in KNXUltimate – Calimero sees;
- When I program with ETS using Wi-Fi adapter (multicasting) – device was flashed successfully;
- I tried a new version of KNXUltimate with no improvements;
- ETS does not see Calimero device discovery anymore! I had to add it by IP manually to use as a gateway (and it worked);
- Bus Monitor in gateway mode in ETS fails for both USB and UART with communication error;
- The strangest thing is that when I send via KNXUltimate multicast – Logic Machine sees my telegram, but Calimero does not.

I suspect a problem on my side, but I don't know where. I checked my Wi-Fi adapter settings and realized I had a wrong netmask assigned (it should be `255.255.240.0` for my particular LAN setup), but fixing this gave nothing.

I will reinstall Raspbian from scratch again, and then collect clean logs as well as perform load tests to check reliability of tunneling connections.

> Both source IP and port are used.

> This is no estimate, this is a strict limit on concurrent connections that qualify for "subscribing" to having frames replayed.

If computer A connects from port 5555, and then computer B connects from port 5555 too – so, for B it would be a different disruption buffer or what? If yes, then it does not look like the port range defines a strict limit.

---

<details><summary> Logs of USB success (writes are sent from Logic Machine):</summary>

```
alpi@rpi3:~ $ sudo docker run --name my_knx --rm --tty --net host --privileged --volume /srv/my/dock/app/server-config.xml:/usr/app/server-config.xml --volume /dev/bus/usb/:/dev/bus/usb/ calimeroproject/knxserver -vvv --no-stdin server-config.xml
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
14:57:31.939 [TRACE  ] io.calimero.xml.XmlResolver: resolve server-config.xml
14:57:31.948 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
14:57:32.969 [TRACE  ] io.calimero.xml.XmlResolver: resolve jar:file:///usr/app/calimero-server-3.0-SNAPSHOT/lib/calimero-core-3.0-SNAPSHOT.jar!/properties.xml
14:57:32.989 [TRACE  ] io.calimero.xml.XmlInputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamReader
14:57:33.298 [DEBUG  ] io.calimero.device.knx-server: loading interface object server from file:///usr/app/knx-server-ios.xml
14:57:33.307 [TRACE  ] io.calimero.xml.XmlResolver: resolve file:///usr/app/knx-server-ios.xml
14:57:33.313 [DEBUG  ] io.calimero.device.knx-server: no interface object server resource, create resource on closing device: /usr/app/knx-server-ios.xml (No such file or directory)
14:57:33.428 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
14:57:33.559 [INFO   ] io.calimero.server.knx-server: use configuration 'Calimero KNX' with 1 service container, discovery: listen on [wlan0] send on [wlan0] interfaces
service container Usb-6.0.0:
        server: listen on wlan0 (port 3671), KNX IP routing multicast group 224.0.23.12
        Usb connection: TP1 medium, device 6.0.0
14:57:33.630 [DEBUG  ] io.calimero.server.knx-server: setup Usb subnet 'Usb-6.0.0'
14:57:33.717 [TRACE  ] io.calimero.device.knx-server: sync Addresstable Object from address 0x116
14:57:33.720 [TRACE  ] io.calimero.device.knx-server: sync Associationtable Object from address 0x1000
14:57:33.725 [TRACE  ] io.calimero.device.knx-server: sync Group Object Table Object from address 0x3000
14:57:33.999 [TRACE  ] io.calimero.server.knx-server: network stack uses IPv4 addresses: unknown
14:57:34.055 [DEBUG  ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: join multicast group 224.0.23.12 on wlan0
14:57:34.062 [TRACE  ] io.calimero.server.knx-server: found network interfaces wlan0 [192.168.28.184]
14:57:34.092 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
14:57:34.095 [TRACE  ] io.calimero.server.knx-server: Usb-6.0.0 control endpoint bound to 192.168.28.184:3671
14:57:34.101 [INFO   ] io.calimero.server.knx-server: discovery endpoint [wlan0] is up and running
14:57:34.105 [TRACE  ] io.calimero.serial.ConnectionFactory: instantiate service provider io.calimero.usb.provider.javax.UsbConnectionProvider
14:57:34.126 [INFO   ] io.calimero.server.knx-server: Usb-6.0.0 KNX IP routing service (wlan0 224.0.23.12) is up and running
14:57:34.197 [INFO   ] io.calimero.server.knx-server: Usb-6.0.0 secure mgmt/tunneling connections: optional/optional
14:57:34.243 [INFO   ] io.calimero.server.knx-server: Usb-6.0.0 tcp service (wlan0 192.168.28.184:3671) is up and running
14:57:34.254 [INFO   ] io.calimero.server.knx-server: Usb-6.0.0 control endpoint (wlan0 192.168.28.184:3671) is up and running
14:57:34.512 [DEBUG  ] io.calimero.usb.provider.javax: Attached USB devices:
Root hub
 |--Bus 1 Device 1: ID 1d6b:0002 Linux 6.6.20+rpt-rpi-v7 dwc_otg_hcd DWC OTG Controller S/N 3f980000.usb
     High Speed USB
      |--Bus 1 Device 2: ID 0424:9514
[... cut ...]
14:58:08.819 [INFO   ] io.calimero.server.knx-server: knx-server routing service 224.0.23.12 closed
14:58:08.823 [DEBUG  ] io.calimero.device.knx-server: saving device memory to file:///usr/app/knx-server-ios.xml
14:58:08.857 [DEBUG  ] io.calimero.device.knx-server: saving interface object server to file:///usr/app/knx-server-ios.xml
14:58:08.870 [TRACE  ] io.calimero.xml.XmlOutputFactory: no StaX implementation found (java.lang.NoClassDefFoundError: javax/xml/stream/XMLStreamException), using internal XMLStreamWriter
alpi@rpi3:~ $
```

</details>

closed 2024-05-20 by aleksusklim

bmalinowsky, 2024-05-26:
One thing I forgot to mention: when the server is running, you can type `stat` which will also list open client connections. So you see which clients are currently connected.