calimero-project/calimero-server #22: Maximum amount of concurrent connections is already used

sametserpil, 2024-04-26:
Hello friends,

I have an office with knx infrastructure and an automation software runs on it. I have 2 knx ip router devices and one of them is broken. I want to use this project as a replacement to that broken knx ip router.  ETS can recognize calimero server but when I test it it says "The interface can not accept the new data connection because its maximum amound of concurrent connections is already used" 

[image]

Here is the console output:

```
Starting a Gradle Daemon (subsequent builds will be faster)
Path for java installation '/usr/lib/jvm/openjdk-17' (Common Linux Locations) does not contain a java executable

> Task :run
08:14:16:188 INFO io.calimero.server.knx-server - Calimero KNX IP server v3.0-SNAPSHOT
08:14:16:212 INFO io.calimero.server.knx-server - use configuration 'Calimero KNX IP Server' with 1 service container, discovery: listen on [ens160] send on [ens160] interfaces
service container 192.168.1.165:3671:
        server: listen on ens160 (port 3671), KNX IP routing disabled
        Udp 192.168.1.165:3671 connection: TP1 medium, device 15.15.0
type 'stop' to stop the gateway and shutdown the server
08:14:16:256 INFO io.calimero.server.knx-server - discovery endpoint [ens160] is up and running
08:14:16:268 INFO io.calimero.server.knx-server - 192.168.1.165:3671 secure mgmt/tunneling connections: optional/optional
08:14:16:271 INFO io.calimero.server.knx-server - 192.168.1.165:3671 tcp service (ens160 192.168.1.165:3671) is up and running
08:14:16:272 INFO io.calimero.server.knx-server - 192.168.1.165:3671 control endpoint (ens160 192.168.1.165:3671) is up and running
08:14:16:272 INFO io.calimero.server.knx-server - 192.168.1.165:3671: setup data endpoint (UDP, channel 1) for connection request from 192.168.1.165:36767
08:14:16:275 INFO io.calimero.server.knx-server - assign additional individual address 15.15.10 to channel 1
08:14:26:267 ERROR io.calimero.knxnetip.KNXnet/IP Tunneling 192.168.1.165:3671 - establishing connection failed, timeout connecting to control endpoint 192.168.1.165:3671
08:14:26:267 ERROR io.calimero.link.connector - initial connection attempt
io.calimero.KNXTimeoutException: timeout connecting to control endpoint 192.168.1.165:3671
        at io.calimero.core/io.calimero.knxnetip.ClientConnection.connect(ClientConnection.java:226)
        at io.calimero.core/io.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:174)
        at io.calimero.core/io.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:166)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.newConnection(KNXNetworkLinkIP.java:586)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.<init>(KNXNetworkLinkIP.java:347)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.newTunnelingLink(KNXNetworkLinkIP.java:163)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$2(SubnetConnector.java:296)
        at io.calimero.core/io.calimero.link.Connector$Link.connect(Connector.java:424)
        at io.calimero.core/io.calimero.link.Connector$Link.<init>(Connector.java:195)
        at io.calimero.core/io.calimero.link.Connector.newLink(Connector.java:151)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:381)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.KnxServerGateway.run(KnxServerGateway.java:729)
        at java.base/java.lang.Thread.run(Thread.java:840)
08:14:26:281 ERROR io.calimero.knxnetip.KNXnet/IP Tunneling 192.168.1.165:3671 - establishing connection failed, timeout connecting to control endpoint 192.168.1.165:3671
08:14:26:281 ERROR io.calimero.link.connector - initial connection attempt
io.calimero.KNXTimeoutException: timeout connecting to control endpoint 192.168.1.165:3671
        at io.calimero.core/io.calimero.knxnetip.ClientConnection.connect(ClientConnection.java:226)
        at io.calimero.core/io.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:174)
        at io.calimero.core/io.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:166)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.newConnection(KNXNetworkLinkIP.java:586)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.<init>(KNXNetworkLinkIP.java:347)
        at io.calimero.core/io.calimero.link.KNXNetworkLinkIP.newTunnelingLink(KNXNetworkLinkIP.java:163)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$2(SubnetConnector.java:296)
        at io.calimero.core/io.calimero.link.Connector$Link.connect(Connector.java:424)
        at io.calimero.core/io.calimero.link.Connector$Link.<init>(Connector.java:195)
        at io.calimero.core/io.calimero.link.Connector.newLink(Connector.java:151)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:381)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.KnxServerGateway$KNXnetIPServerListener.acceptDataConnection(KnxServerGateway.java:264)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.lambda$acceptConnection$14(ControlEndpointService.java:1176)
        at java.base/java.util.stream.MatchOps$1MatchSink.accept(MatchOps.java:90)
        at java.base/java.util.Spliterators$ArraySpliterator.tryAdvance(Spliterators.java:1002)
        at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:129)
        at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:527)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:513)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
        at java.base/java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:230)
        at java.base/java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:196)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.allMatch(ReferencePipeline.java:637)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.acceptConnection(ControlEndpointService.java:1176)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.initNewConnection(ControlEndpointService.java:946)
        at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.acceptControlService(ControlEndpointService.java:407)
[... 24 lines cut ...]
                udpPort="3671" listenNetIf="ens160">
                <knxAddress type="individual">15.15.0</knxAddress>
                <routing/>

                <knxSubnet type="udp" media="knxip">192.168.1.165:3671</knxSubnet>

                <groupAddressFilter>
                        <!-- <knxAddress type="group">2/1/1</knxAddress> -->
                        <!-- <knxAddress type="group">2/1/0</knxAddress> -->
                        <!-- <knxAddress type="group">2/1/20</knxAddress> -->
                </groupAddressFilter>

                <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
                <additionalAddresses>
                        <knxAddress type="individual">15.15.10</knxAddress>
                        <knxAddress type="individual">15.15.11</knxAddress>
                </additionalAddresses>

        </serviceContainer>

        <!-- Add next service container (optional) -->
</knxServer>
```

Could you please help me?

bmalinowsky, 2024-04-26:
In the server configuration, under `<additionalAddresses>`, add some more addresses. This will help. Also, I see in the log output, that you try to connect the subnet link `<knxSubnet ...` to the server itself? Because both the server and the configured subnet connection do have the same IP and port. Which wouldn't make sense.

sametserpil, 2024-04-26:
Oh, I've just figured it out. Thank you. This pproject is awsome.

closed 2024-04-26 by sametserpil