calimero-project/calimero-server #24: Problem running ft12-cemi KNX BAOS MODULE 838 KBERRY

georgegougoudis, 2024-06-11:
I am trying to get Calimero Server to work with this module on a raspberry pi 5 running home assistant and I am facing some issues.

For one, I can get a different interface with TPUART to work properly and even program through ETS without any problems.
I can also run KNXD on the same setup with ft12cemi and it works properly for communication (I could never get KNXD to program with ETS though).

I am building a docker image based on dockerhub image.
The logs are
```
WARNING: Unknown module: io.calimero.serial.provider.rxtx specified to --add-reads
15:07:01.065 [INFO   ] io.calimero.server.knx-server: Calimero KNX IP server v3.0-SNAPSHOT
15:07:01.088 [INFO   ] io.calimero.server.knx-server: use configuration 'KNX IP Server' with 1 service container, discovery: listen on [all] send on [all] interfaces
service container /dev/ttyAMA0:
	server: listen on end0 (port 3671), KNX IP routing multicast group 224.0.23.12
	Ft12 /dev/ttyAMA0 cemi connection: TP1 medium, device 1.1.0
15:07:01.106 [INFO   ] io.calimero.server.gateway.knx-server: activate '/dev/ttyAMA0' disruption buffer on ports [5,555-5,559], disruption timeout 30 s
15:07:01.171 [INFO   ] io.calimero.server.knx-server: discovery endpoint [docker0, hassio, end0, lo] is up and running
15:07:01.173 [INFO   ] io.calimero.knxnetip.KNXnet/IP Routing 224.0.23.12:3671: multicast loopback mode enabled
15:07:01.178 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 secure mgmt/tunneling connections: optional/optional
15:07:01.179 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 KNX IP routing service (end0 224.0.23.12) is up and running
15:07:01.185 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 tcp service (end0 192.168.1.75:3671) is up and running
15:07:01.186 [INFO   ] io.calimero.server.knx-server: /dev/ttyAMA0 control endpoint (end0 192.168.1.75:3671) is up and running
15:07:01.819 [INFO   ] io.calimero.serial.FT12:/dev/ttyAMA0: close serial port /dev/ttyAMA0 - acknowledgment timeout on sending reset
15:07:01.871 [ERROR  ] io.calimero.link.connector: initial connection attempt
io.calimero.KNXAckTimeoutException: resetting BCU failed (no acknowledge reply received)
	at io.calimero.core/io.calimero.serial.FT12Connection.sendReset(FT12Connection.java:476)
	at io.calimero.core/io.calimero.serial.FT12Connection.reset(FT12Connection.java:451)
	at io.calimero.core/io.calimero.serial.FT12Connection.<init>(FT12Connection.java:244)
	at io.calimero.core/io.calimero.serial.FT12Connection.<init>(FT12Connection.java:221)
	at io.calimero.core/io.calimero.link.KNXNetworkLinkFT12.newCemiLink(KNXNetworkLinkFT12.java:68)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$10(SubnetConnector.java:340)
	at io.calimero.core/io.calimero.link.Connector$Link.connect(Connector.java:424)
	at io.calimero.core/io.calimero.link.Connector$Link.<init>(Connector.java:195)
	at io.calimero.core/io.calimero.link.Connector.newLink(Connector.java:151)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:381)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.gateway.KnxServerGateway.run(KnxServerGateway.java:731)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher.run(Launcher.java:655)
	at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher.main(Launcher.java:620)
```

My config (which works perfectly for tpuart subnet) is:
```xml
<?xml version="1.0" encoding="UTF-8"?>

<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="KNX IP Server">
	<!-- KNXnet/IP search & discovery -->
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />

	<!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
	<serviceContainer activate="true" routing="true" networkMonitoring="true"
		udpPort="3671" listenNetIf="end0">
		<knxAddress type="individual">1.1.0</knxAddress>
		<routing/>
		<disruptionBuffer expirationTimeout="30" udpPort="5555-5559" />
		<knxSubnet type="ft12" medium="tp1" format="cemi">/dev/ttyAMA0</knxSubnet>
		<groupAddressFilter>
			<!-- <knxAddress type="group">2/1/1</knxAddress> -->
			<!-- <knxAddress type="group">2/1/0</knxAddress> -->
			<!-- <knxAddress type="group">2/1/20</knxAddress> -->
		</groupAddressFilter>

		<!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
		<additionalAddresses>
			<knxAddress type="individual">1.1.240</knxAddress>
			<knxAddress type="individual">1.1.241</knxAddress>
			<knxAddress type="individual">1.1.242</knxAddress>
			<knxAddress type="individual">1.1.243</knxAddress>
			<knxAddress type="individual">1.1.244</knxAddress>
		</additionalAddresses>
	</serviceContainer>
[... 70 lines cut ...]
15:54:08.021 [WARNING] io.calimero.link./dev/ttyAMA0: negative confirmation of 1/1/3: 2e 00 9d d0 ff ff 09 03 01 00 00
15:54:08.791 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 1.1.101->5/2/2
15:54:08.870 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 1.1.101->5/2/2
15:54:09.202 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->1/1/0
15:54:11.867 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->5/2/3
15:54:12.022 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->5/2/3
15:54:12.309 [WARNING] io.calimero.link./dev/ttyAMA0: negative confirmation of 11/1/20: 2e 00 9d d0 ff ff 59 14 01 00 00
15:54:12.921 [WARNING] io.calimero.link./dev/ttyAMA0: negative confirmation of 1/1/3: 2e 00 9d b0 ff ff 09 03 01 00 00
15:54:13.242 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->5/2/3
15:54:13.449 [WARNING] io.calimero.link./dev/ttyAMA0: negative confirmation of 1/1/2: 2e 00 9d 90 ff ff 09 02 01 00 00
15:54:13.522 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->5/2/3
15:54:13.686 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->1/1/1
15:54:15.729 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 1.1.6->6/2/6
15:54:16.086 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 1.1.101->5/2/3
15:54:16.913 [WARNING] io.calimero.link./dev/ttyAMA0: negative confirmation of 1/1/2: 2e 00 9d b0 ff ff 09 02 01 00 00
15:54:16.915 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->1/1/0
15:54:17.014 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 1.1.101->5/2/3
15:54:18.094 [WARNING] io.calimero.server.gateway.knx-server: hop count 0, discard frame 15.15.255->1/1/0
```

Also I cant figure out how to enable DEBUG/TRACE logging, I tried something like:
```
-Dorg.slf4j.simpleLogger.defaultLogLevel=DEBUG
```
but it doesn't work. I am not very familiar with the JAVA environment so that may be completely wrong.

bmalinowsky, 2024-06-11:
For trace logging, you can try to set `-vvv`, like so: _calimeroproject/knxserver -vvv /path/to/server-config.xml_.

Calimero uses the `System.Logger` façade, the docker image just uses the internal stdout logger. You should also be able to set the level with the `jdk.system.logger.level` system property (INFO, DEBUG, TRACE).
In the image, there are no slf4j libraries inlcuded; if you want to use slf4j, you have to add those libraries into the `lib` folder.

The discarding of frames due to hop count 0 is usually because there is a loop somewhere.

bmalinowsky, 2024-06-13:
Why the reset fails I cannot say from the info you posted, sometimes people mix up ACM and AMA which would lead to the same error.

`-D...` options are for the JVM, so you need to pass it, e.g., via the `export CALIMERO_SERVER_OPTS=` you have in `run.sh`.

Also, put the `-vvv` before `--no-stdin`.

closed 2024-08-19 by bmalinowsky