calimero-project/calimero-server #25: 请求帮助.使用问题

bert1020, 2024-08-19:
我想自己做一个用您这个做一个网关,并且增加一些功能,我现在需要拦截网关接收到的消息做特定的动作,比如我网关里面自定义了一些场景.收到执行14/0/1=1的时候 我去发送15/0/1=1 15/0/2=1 .我该怎么拦截他,有没有哪里可以添加过滤器

bmalinowsky, 2024-08-19:
There is no configuration or similar to specify a filter for what you want to achieve; you have to modify the Java source code.

If your logic is intended for frames from the KNX subnet, use the method `onSubnetFrameReceived` [around here](https://github.com/calimero-project/calimero-server/blob/03afedcd48d9e5383397c112a2b3115be73a52de/src/io/calimero/server/gateway/KnxServerGateway.java#L1331).
If your logic is intended for frames incoming from server clients, use the method `onServerFrameReceived` [around here](https://github.com/calimero-project/calimero-server/blob/03afedcd48d9e5383397c112a2b3115be73a52de/src/io/calimero/server/gateway/KnxServerGateway.java#L1190).

bert1020, 2024-08-20:
[image]
我有这么一个USB设备,但运行之后一直找不到
`
io.calimero.KNXException: no service provider available for io.calimero.serial.usb.spi.UsbConnectionProvider
`
这是我的子网配置  ,如果不写Tokka 也是这样子.
      <knxSubnet type="usb" medium="tp1">Tokka</knxSubnet>
这是我运行的代码 ,直接在idea中运行的
        Launcher launcher = new Launcher("file:///D:/project/knx-server/src/main/resources/config/server-config.xml");
        launcher.run();

bert1020, 2024-08-20:
<?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />
	<serviceContainer activate="true" routing="true" networkMonitoring="true"
					  udpPort="3671" listenNetIf="any">
		<knxAddress type="individual">7.2.0</knxAddress>
		<routing/>
		<knxSubnet type="usb" medium="tp1">Tokka</knxSubnet>
		<groupAddressFilter>
		</groupAddressFilter>
		<additionalAddresses>
			<knxAddress type="individual">7.1.10</knxAddress>
			<knxAddress type="individual">7.1.11</knxAddress>
			<knxAddress type="individual">7.1.12</knxAddress>
			<knxAddress type="individual">7.1.13</knxAddress>
		</additionalAddresses>
	</serviceContainer>
</knxServer>

bert1020, 2024-08-20:
> <?xml version="1.0" encoding="UTF-8"?>
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
	<discovery listenNetIf="all" outgoingNetIf="all" activate="true" />
	<serviceContainer activate="true" routing="true" networkMonitoring="true"
					  udpPort="3671" listenNetIf="any">
		<knxAddress type="individual">7.2.0</knxAddress>
		<routing/>
		<knxSubnet type="usb" medium="tp1">Tokka</knxSubnet>
		<groupAddressFilter>
		</groupAddressFilter>
		<additionalAddresses>
			<knxAddress type="individual">7.1.10</knxAddress>
			<knxAddress type="individual">7.1.11</knxAddress>
			<knxAddress type="individual">7.1.12</knxAddress>
			<knxAddress type="individual">7.1.13</knxAddress>
		</additionalAddresses>
	</serviceContainer>
</knxServer>

bert1020, 2024-08-20:
这是我的日志
``` java -cp "calimero-server-3.0-SNAPSHOT.jar:calimero-core-3.0-SNAPSHOT.jar:calimero-device-3.0-SNAPSHOT.jar" io.calimero.server.Launcher server-config.xml
8月 20, 2024 6:17:47 下午 io.calimero.server.Launcher <init>
信息: Calimero KNX IP server 3.0-SNAPSHOT (03afedc)
8月 20, 2024 6:17:47 下午 io.calimero.server.Launcher <init>
信息: use configuration 'Calimero KNX IP Server' with 1 service container, discovery: listen on [all] send on [all] interfaces
service container Usb-7.1.0:
	server: listen on any (port 3671), KNX IP routing multicast group 224.0.23.12
	Usb connection: TP1 medium, device 7.1.0
type 'stop' to stop the gateway and shutdown the server
8月 20, 2024 6:17:47 下午 io.calimero.link.Connector$Link <init>
严重: initial connection attempt
io.calimero.KNXException: no service provider available for io.calimero.serial.usb.spi.UsbConnectionProvider
	at io.calimero.serial.ConnectionFactory.open(ConnectionFactory.java:92)
	at io.calimero.serial.usb.UsbConnectionFactory.open(UsbConnectionFactory.java:66)
	at io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:117)
	at io.calimero.server.gateway.SubnetConnector.lambda$openNetworkLink$9(SubnetConnector.java:336)
	at io.calimero.link.Connector$Link.connect(Connector.java:424)
	at io.calimero.link.Connector$Link.<init>(Connector.java:195)
	at io.calimero.link.Connector.newLink(Connector.java:151)
	at io.calimero.server.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:381)
	at io.calimero.server.gateway.KnxServerGateway.run(KnxServerGateway.java:737)
	at java.base/java.lang.Thread.run(Thread.java:840)

8月 20, 2024 6:17:47 下午 io.calimero.server.knxnetip.LooperTask run
信息: discovery endpoint [docker0, enp1s0, lo] is up and running
8月 20, 2024 6:17:47 下午 io.calimero.knxnetip.KNXnetIPRouting init
信息: multicast loopback mode enabled
8月 20, 2024 6:17:47 下午 io.calimero.server.knxnetip.LooperTask run
信息: Usb-7.1.0 KNX IP routing service (0.0.0.0 224.0.23.12) is up and running
8月 20, 2024 6:17:47 下午 io.calimero.server.knxnetip.ControlEndpointService <init>
信息: Usb-7.1.0 secure mgmt/tunneling connections: optional/optional
8月 20, 2024 6:17:47 下午 io.calimero.server.knxnetip.LooperTask run
信息: Usb-7.1.0 control endpoint (enp1s0 192.168.123.159:3671) is up and running
8月 20, 2024 6:17:47 下午 io.calimero.server.knxnetip.StreamEndpoint run
信息: Usb-7.1.0 tcp service (enp1s0 tcp:192.168.123.159:3671) is up and running
8月 20, 2024 6:17:52 下午 io.calimero.server.gateway.KnxServerGateway isNetworkLink
警告: cannot dispatch to KNX subnet 7.1.0: no subnet connection
8月 20, 2024 6:17:57 下午 io.calimero.link.Connector$Link lambda$scheduleConnect$1
警告: connection attempt 2: no service provider available for io.calimero.serial.usb.spi.UsbConnectionProvider

closed 2024-08-21 by bert1020