calimero-project/calimero-server #26: unknown mask version 00FA

bert1020, 2024-08-21:
这是这个USB设备 不支持吗?
```信息: Found 1 KNX USB devices:
 |--Bus 1 Device 9: ID 0403:6898 TSY Technology TSY KNX-USB Data Interface S/N 6989BB910B31
 |   Attached at port 1 (bus:1/1/1)
 |   Full Speed USB
8月 21, 2024 1:10:14 下午 com.seawin.knxserver.calimero.gateway.KnxServerGateway run
严重: error opening network link for Usb-7.2.0
io.calimero.KNXIllegalArgumentException: unknown mask version 00FA
	at io.calimero.DeviceDescriptor$DD0.from(DeviceDescriptor.java:199)
	at io.calimero.usb.provider.javax.UsbConnection.deviceDescriptor(UsbConnection.java:419)
	at io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:140)
	at io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:117)
	at com.seawin.knxserver.calimero.gateway.SubnetConnector.lambda$openNetworkLink$9(SubnetConnector.java:317)
	at io.calimero.link.Connector$Link.connect(Connector.java:424)
	at io.calimero.link.Connector$Link.<init>(Connector.java:195)
	at io.calimero.link.Connector.newLink(Connector.java:151)
	at com.seawin.knxserver.calimero.gateway.SubnetConnector.openNetworkLink(SubnetConnector.java:362)
	at com.seawin.knxserver.calimero.gateway.KnxServerGateway.run(KnxServerGateway.java:668)
	at com.seawin.knxserver.calimero.Launcher.run(Launcher.java:634)

bmalinowsky, 2024-08-21:
The reporting is not supported, because the mask version is not specified (or it is a new one). If you skip the particular query, it will probably work. It's these two lines: https://github.com/calimero-project/calimero-core/blob/4520cdd8c94ec499f094ce270d22013489f3ceb3/src/io/calimero/link/KNXNetworkLinkUsb.java#L140

bert1020, 2024-08-22:
现在报这个警告
我理解的是,我启动服务,通过USB接到总线上,应该能收到总线上的消息才对.现在我完全收不到.不知道我这样的理解对吗
``WARN io.calimero.link.0403:6898 -- wrong communication medium setting: using TP1 to access IP medium

bert1020, 2024-08-22:
配置文件直接用的默认的配置文件
```<?xml version="1.0" encoding="UTF-8"?>
<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
    <!-- KNXnet/IP search & discovery -->
    <discovery listenNetIf="all" outgoingNetIf="all" activate="true" />

    <!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
    <serviceContainer activate="true" routing="true" networkMonitoring="true"
                      udpPort="3671" listenNetIf="any">
        <knxAddress type="individual">7.1.0</knxAddress>
        <routing/>
        <!-- latencyTolerance specifies the time window for accepting secure multicasts,
            depends on max. end-to-end network latency (usually [500 .. 8000] ms) -->
        <!-- <routing latencyTolerance="1000">224.0.23.12</routing> -->

        <!-- Enabling the disruption buffer will replay missed frames after reconnecting a KNXnet/IP client link
            using the specific client UDP port range (if caused by a disrupted connection). -->
        <!-- <disruptionBuffer expirationTimeout="30" udpPort="5555-5559" /> -->

        <!-- Specify one KNX subnet type, connecting the KNX network. The subnet connection type is one of
            { "udp", "tcp", "knxip", "ft12", "usb", "tpuart", "user", "virtual", "emulate" }.
            The KNX communication medium is one of { "tp1" (default), "pl110", "knxip", "rf" }.
            In the following are some examples: -->

        <!-- <knxSubnet type="udp">192.168.10.10:3671</knxSubnet> -->
        <!-- two optional attributes are supported for tunneling:
            * netif: specifies the server network interface to the knx subnet
            * useNat="true"|"false": use network address translation (NAT) -->
        <!-- <knxSubnet type="udp" netif="eth0" useNat="true">192.168.10.10:3671</knxSubnet> -->

        <!-- <knxSubnet type="knxip" listenNetIf="en0">224.0.23.12</knxSubnet> -->

        <!-- <knxSubnet type="tcp">192.168.10.80:3671</knxSubnet> -->
        <!-- <knxSubnet type="tcp" host="1.2.0" user="4">192.168.10.80:3671</knxSubnet> -->

        <!-- <knxSubnet type="ft12" medium="pl110" domainAddress="6f">/dev/ttyS0</knxSubnet> -->
        <!-- <knxSubnet type="ft12" medium="tp1" format="cemi">/dev/ttyS0</knxSubnet> -->

        <!-- KNX USB: specify the vendorId:productId or vendor name, e.g., siemens -->
        <!-- <knxSubnet  type="usb" medium="tp1">siemens</knxSubnet> -->
        <!-- if USB device identifier is left empty, the first found KNX USB device is used -->
        <knxSubnet type="usb" medium="tp1"></knxSubnet>

        <!-- <knxSubnet type="tpuart" medium="tp1">/dev/ttyACM0</knxSubnet> -->
        <!-- <knxSubnet type="virtual">virtual-subnet</knxSubnet> -->
        <!-- <knxSubnet type="emulate">emulate-subnet</knxSubnet> -->
        <!-- <knxSubnet type="user" class="my.knx.NetworkLink">o1,i2|i4</knxSubnet> -->

        <!-- Datapoints are only useful if an emulated KNX subnet is used -->
        <!-- <datapoints ref="datapointMap.xml" /> -->

        <!-- KNX group address filter applied by the server for this service container (optional).
             This filter is used both for server->subnet and vice versa. -->
        <groupAddressFilter>
            <!-- <knxAddress type="group">2/1/1</knxAddress> -->
            <!-- <knxAddress type="group">2/1/0</knxAddress> -->
            <!-- <knxAddress type="group">2/1/20</knxAddress> -->
        </groupAddressFilter>

        <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
        <additionalAddresses>
            <knxAddress type="individual">7.1.10</knxAddress>
            <knxAddress type="individual">7.1.11</knxAddress>
            <knxAddress type="individual">7.1.12</knxAddress>
            <knxAddress type="individual">7.1.13</knxAddress>
        </additionalAddresses>

        <!-- Cyclically transmit date (11.001), time (10.001), or date+time (19.001) information on the subnetwork
             and server side, using the default time zone and the specified datapoint(s).
[... 57 lines cut ...]
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth19
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth20
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth21
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth22
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth23
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth24
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface eth25
10:28:40.147 [knx-server discovery endpoint] WARN io.calimero.server.knx-server -- KNXnet/IP discovery join fails with no IP address bound to interface wlan10
[... cut ...]
10:28:43.051 [Usb-7.1.0 tcp connection tcp:192.168.123.233:52766] INFO io.calimero.server.knx-server -- accepted Usb-7.1.0 tcp connection tcp:192.168.123.233:52766
10:28:43.053 [Usb-7.1.0 tcp connection tcp:192.168.123.233:52766] INFO io.calimero.server.knx-server -- Usb-7.1.0: setup data endpoint (channel 1) for connection request from tcp:192.168.123.233:52766
10:28:43.057 [Usb-7.1.0 tcp connection tcp:192.168.123.233:52766] INFO io.calimero.server.knx-server -- assign additional individual address 7.1.10 to channel 1

bmalinowsky, 2024-08-22:
Your usb stick reports an IP medium, which is probably wrong as it should report a TP1 medium. Anyway, this won't prevent the usb stick or the server from working. It's just odd.

You could check the trace log if the stick receives any frames from the knx bus at all.

bert1020, 2024-08-23:
谢谢您的支持.现在有了,其中还有个警告,这是什么意思呢
`10:26:17.318 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/12`
还有我该如何往总线上发送消息呢?
下面是完整日志
```10:26:17.225 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/7 L_Data.ind, low priority hop count 0, tpdu 00 80 3f 80 00 00: T_Group, A_Group.write
10:26:17.225 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/7
10:26:17.269 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/8 L_Data.ind, low priority hop count 0, tpdu 00 80 00 00 00 4a: T_Group, A_Group.write
10:26:17.269 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/8
10:26:17.293 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/6 L_Data.ind, low priority hop count 4, tpdu 00 80 00 00 00 00: T_Group, A_Group.write
10:26:17.293 [knx-server] DEBUG i.c.s.g.knx-server - [dispatchLdataToClients,1597] - dispatch 1.1.3->15/0/6 to all server-side connections
10:26:17.294 [knx-server] TRACE i.c.s.g.knx-server - [dispatchToOtherSubnets,1459] - dispatching to KNX subnets: exclude subnet Usb-7.1.0
10:26:17.294 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,234] - add to multicast loopback frame buffer: 1.1.3->15/0/6 L_Data.ind, low priority hop count 3, tpdu 00 80 00 00 00 00
10:26:17.294 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,238] - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 15 29 00 bc b0 11 03 78 06 05 00 80 00 00 00 00
10:26:17.295 [knx-server routing service 224.0.23.12] TRACE i.c.s.k.K.0.0.0.0.2.12) - [discardLoopbackFrame,649] - discard multicast loopback cEMI frame: 1.1.3->15/0/6 L_Data.ind, low priority hop count 3, tpdu 00 80 00 00 00 00
10:26:17.317 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/12 L_Data.ind, low priority hop count 0, tpdu 00 80 00 00: T_Group, A_Group.write
10:26:17.318 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/12
10:26:17.357 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/13 L_Data.ind, low priority hop count 0, tpdu 00 80 43 5c 00 00: T_Group, A_Group.write
10:26:17.358 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/13
10:26:17.397 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/14 L_Data.ind, low priority hop count 0, tpdu 00 80 00 00 00 00: T_Group, A_Group.write
10:26:17.397 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/14
10:26:17.436 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/15 L_Data.ind, low priority hop count 0, tpdu 00 80 3f 80 00 00: T_Group, A_Group.write
10:26:17.437 [knx-server] WARN  i.c.s.g.knx-server - [adjustHopCount,2226] - hop count 0, discard frame 1.1.3->15/0/15
10:26:17.465 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/7 L_Data.ind, low priority hop count 4, tpdu 00 80 3f 80 00 00: T_Group, A_Group.write
10:26:17.465 [knx-server] DEBUG i.c.s.g.knx-server - [dispatchLdataToClients,1597] - dispatch 1.1.3->15/0/7 to all server-side connections
10:26:17.466 [knx-server] TRACE i.c.s.g.knx-server - [dispatchToOtherSubnets,1459] - dispatching to KNX subnets: exclude subnet Usb-7.1.0
10:26:17.466 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,234] - add to multicast loopback frame buffer: 1.1.3->15/0/7 L_Data.ind, low priority hop count 3, tpdu 00 80 3f 80 00 00
10:26:17.466 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,238] - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 15 29 00 bc b0 11 03 78 07 05 00 80 3f 80 00 00
10:26:17.467 [knx-server routing service 224.0.23.12] TRACE i.c.s.k.K.0.0.0.0.2.12) - [discardLoopbackFrame,649] - discard multicast loopback cEMI frame: 1.1.3->15/0/7 L_Data.ind, low priority hop count 3, tpdu 00 80 3f 80 00 00
10:26:17.512 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/8 L_Data.ind, low priority hop count 4, tpdu 00 80 00 00 00 4a: T_Group, A_Group.write
10:26:17.513 [knx-server] DEBUG i.c.s.g.knx-server - [dispatchLdataToClients,1597] - dispatch 1.1.3->15/0/8 to all server-side connections
10:26:17.513 [knx-server] TRACE i.c.s.g.knx-server - [dispatchToOtherSubnets,1459] - dispatching to KNX subnets: exclude subnet Usb-7.1.0
10:26:17.514 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,234] - add to multicast loopback frame buffer: 1.1.3->15/0/8 L_Data.ind, low priority hop count 3, tpdu 00 80 00 00 00 4a
10:26:17.514 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] TRACE i.c.s.k.K.0.0.0.0.2.12) - [send,238] - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 15 29 00 bc b0 11 03 78 08 05 00 80 00 00 00 4a
[... cut ...] du 00 80 43 5c 00 00
10:26:17.649 [knx-server] TRACE i.c.s.g.knx-server - [onSubnetFrameReceived,1241] - subnet Usb-7.1.0: 1.1.3->15/0/14 L_Data.ind, low priority hop count 4, tpdu 00 80 00 00 00 00: T_Group, A_Group.write
10:26:17.649 [knx-server] DEBUG i.c.s.g.knx-server - [dispatchLdataToClients,1597] - dispatch 1.1.3->15/0/14 to all server-side connections
10:26:17.650 [knx-server] TRACE i.c.s.g.knx-server - [dispatchToOtherSubnets,1459] - dispatching to KNX subnets: exclude subnet Usb-7.1.0

bert1020, 2024-08-29:
谢谢,我表达错误了.我的意思是我该怎么往总线上发送指令?

closed 2024-09-02 by bert1020