calimero-project/calimero-server #29: ETS5读取设备信息 完整下载有问题

bert1020, 2024-12-11:
您好,我用您这个部署了一个knx server
我测试了两种方案 第一种用USB下载器
在ETS5中 读取设备信息,显示设备无响应
[image]
但是编程模式却可以检测到设备
我不知道是不是我哪里设置的不对,下面是我的server-config.xml配置
```
<!-- Calimero server settings (required for startup) -->
<knxServer name="knx-server" friendlyName="Calimero KNX IP Server">
    <!-- KNXnet/IP search & discovery -->
    <discovery listenNetIf="all" outgoingNetIf="all" activate="true" />

    <!-- Provides the KNXnet/IP-side configuration for access to one KNX subnet -->
    <serviceContainer activate="true" routing="true" networkMonitoring="true"
                      udpPort="3671" listenNetIf="any">
        <knxAddress type="individual">1.1.0</knxAddress>
        <routing/>

        <knxSubnet type="usb" medium="tp1"></knxSubnet>

        <!-- Additional KNX individual addresses assigned to client KNXnet/IP connections (optional) -->
        <additionalAddresses>
            <knxAddress type="individual">1.1.15</knxAddress>
        </additionalAddresses>

    </serviceContainer>

</knxServer>
```
下面是我点击读取按钮时产生的日志信息,您帮我分析一下哪里有问题

```
18:15:24.064 [knx-server control endpoint Usb-1.1.0] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received udp:192.168.10.233:57378 06 10 02 03 00 0e 08 01 00 00 00 00 00 00
18:15:24.065 [knx-server control endpoint Usb-1.1.0] DEBUG i.c.s.knx-server - [createResponseAddress,161] - responses use route-back control endpoint udp:192.168.10.233:57378
18:15:24.066 [knx-server control endpoint Usb-1.1.0] INFO  i.c.s.knx-server - [acceptControlService,402] - send KNXnet/IP description to udp:192.168.10.233:57378: "Calimero KNX IP Server", KNX address 1.1.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address 34:29:8f:78:10:ac, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 1.1.15, 7.1.11, 7.1.12, 7.1.13, KNX manufacturer ID 0x0, data 0x626d323031312020
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [run,85] - accepted Usb-1.1.0 tcp connection tcp:192.168.10.233:47175
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 02 05 00 1a 08 02 00 00 00 00 00 00 08 02 00 00 00 00 00 00 04 04 02 00
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.knx-server - [createResponseAddress,170] - responses to client use route-back control endpoint tcp:192.168.10.233:47175
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [acceptControlService,425] - Usb-1.1.0: setup data endpoint (channel 19) for connection request from tcp:192.168.10.233:47175
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.knx-server - [createResponseAddress,161] - responses use route-back data endpoint tcp:192.168.10.233:47175
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [initNewConnection,901] - assign additional individual address 1.1.15 to channel 19
18:15:24.071 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.knx-server - [<init>,66] - created socket udp:192.168.10.233:57379
18:15:24.072 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.g.knx-server - [connectionEstablished,228] - established connection KNX IP LinkLayer tcp:192.168.10.233:47175 (channel 19, 1.1.15)
18:15:24.072 [Usb-1.1.0 data endpoint tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [run,97] - Usb-1.1.0 data endpoint KNX IP LinkLayer tcp:192.168.10.233:47175 (channel 19, 1.1.15) is up and running
18:15:24.078 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] WARN  i.c.s.g.knx-server - [connectionEstablished,238] - skip verifying knx address of 'Usb-1.1.0' subnet interface (unknown mask version 00FA)
18:15:24.078 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 04 24 00 0d 04 13 00 00 08 00 01
18:15:24.078 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [responseForFeature,509] - received tunneling-feat.set enable feature info service 01
18:15:24.078 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [respondToFeature,499] - respond with tunneling-feat.res enable feature info service 01
18:15:24.082 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 04 22 00 0c 04 13 01 00 06 00
18:15:24.082 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [responseForFeature,509] - received tunneling-feat.get individual address 
18:15:24.082 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [respondToFeature,499] - respond with tunneling-feat.res individual address 110f
18:15:24.088 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 04 22 00 0c 04 13 02 00 07 00
18:15:24.088 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [responseForFeature,509] - received tunneling-feat.get max apdu length 
18:15:24.088 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [respondToFeature,499] - respond with tunneling-feat.res max apdu length 00fe
18:15:24.093 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 04 22 00 0c 04 13 03 00 02 00
18:15:24.093 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [responseForFeature,509] - received tunneling-feat.get device descriptor type 0 
18:15:24.094 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [respondToFeature,499] - respond with tunneling-feat.res device descriptor type 0 091a
18:15:24.104 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47175 06 10 02 09 00 10 13 00 08 02 00 00 00 00 00 00
18:15:24.104 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.k.K.1.1.233:47175 - [cleanup,319] - close connection for channel 19 - client request
18:15:24.104 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] DEBUG i.c.s.g.knx-server - [connectionClosed,150] - removed connection KNX IP LinkLayer tcp:192.168.10.233:47175 (client request)
18:15:24.105 [Usb-1.1.0 data endpoint tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [cleanup,130] - Usb-1.1.0 data endpoint tcp:192.168.10.233:47175 closed
18:15:24.110 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47175] INFO  i.c.s.knx-server - [close,166] - close connection to tcp:192.168.10.233:47175
18:15:24.116 [knx-server control endpoint Usb-1.1.0] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received udp:192.168.10.233:57380 06 10 02 03 00 0e 08 01 00 00 00 00 00 00
18:15:24.116 [knx-server control endpoint Usb-1.1.0] DEBUG i.c.s.knx-server - [createResponseAddress,161] - responses use route-back control endpoint udp:192.168.10.233:57380
18:15:24.117 [knx-server control endpoint Usb-1.1.0] INFO  i.c.s.knx-server - [acceptControlService,402] - send KNXnet/IP description to udp:192.168.10.233:57380: "Calimero KNX IP Server", KNX address 1.1.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address 34:29:8f:78:10:ac, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 1.1.15, 7.1.11, 7.1.12, 7.1.13, KNX manufacturer ID 0x0, data 0x626d323031312020
18:15:24.123 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47176] INFO  i.c.s.knx-server - [run,85] - accepted Usb-1.1.0 tcp connection tcp:192.168.10.233:47176
18:15:24.123 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47176] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-1.1.0 received tcp:192.168.10.233:47176 06 10 02 05 00 1a 08 02 00 00 00 00 00 00 08 02 00 00 00 00 00 00 04 04 02 00
18:15:24.123 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47176] DEBUG i.c.s.knx-server - [createResponseAddress,170] - responses to client use route-back control endpoint tcp:192.168.10.233:47176
[... cut ...]
18:15:49.336 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47176] DEBUG i.c.s.g.knx-server - [connectionClosed,150] - removed connection KNX IP LinkLayer tcp:192.168.10.233:47176 (client request)
18:15:49.336 [Usb-1.1.0 data endpoint tcp:192.168.10.233:47176] INFO  i.c.s.knx-server - [cleanup,130] - Usb-1.1.0 data endpoint tcp:192.168.10.233:47176 closed
18:15:49.342 [Usb-1.1.0 tcp connection tcp:192.168.10.233:47176] INFO  i.c.s.knx-server - [close,166] - close connection to tcp:192.168.10.233:47176

```
我也使用了完整下载功能也是下载不了,提示设备在不适当的时候响应
我也在另外一台机器上部署了一套 ,使用了
```<knxSubnet type="tpuart" medium="tp1">/dev/ttyUSB0</knxSubnet>```配置
跟用USB下载器是一样的效果 也是读取不了设备信息 和下载功能

bmalinowsky, 2024-12-11:
You configure a knx subnet of 1.1 (i.e., 1.1.0) but want to address a device with address 15.15.234. The telegram will not be forwarded by the server. The topology is wrong.

bert1020, 2024-12-12:
我把knx server 的 individual 设置成了15.15.0  依旧是设备无响应
麻烦您再帮我分析一下是什么原因
```
14:38:53.855 [knx-server discovery endpoint] DEBUG i.c.s.knx-server - [sendSearchResponse,234] - KNXnet/IP discovery: identify as 'Calimero KNX IP Server' for container Usb-15.15.0 to /192.168.1.102:52987 on [any]
14:38:53.880 [knx-server discovery endpoint] DEBUG i.c.s.knx-server - [sendSearchResponse,234] - KNXnet/IP discovery: identify as 'Calimero KNX IP Server' for container Usb-15.15.0 to /192.168.10.233:52988 on [any]
14:38:53.907 [knx-server discovery endpoint] DEBUG i.c.s.knx-server - [sendSearchResponse,234] - KNXnet/IP discovery: identify as 'Calimero KNX IP Server' for container Usb-15.15.0 to /192.168.10.233:52988 on [any]
14:38:53.934 [knx-server discovery endpoint] DEBUG i.c.s.knx-server - [sendSearchResponse,234] - KNXnet/IP discovery: identify as 'Calimero KNX IP Server' for container Usb-15.15.0 to /192.168.1.102:52987 on [any]
14:38:57.908 [knx-server control endpoint Usb-15.15.0] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-15.15.0 received udp:192.168.10.233:60187 06 10 02 03 00 0e 08 01 00 00 00 00 00 00
14:38:57.909 [knx-server control endpoint Usb-15.15.0] DEBUG i.c.s.knx-server - [createResponseAddress,161] - responses use route-back control endpoint udp:192.168.10.233:60187
14:38:57.909 [knx-server control endpoint Usb-15.15.0] INFO  i.c.s.knx-server - [acceptControlService,402] - send KNXnet/IP description to udp:192.168.10.233:60187: "Calimero KNX IP Server", KNX address 15.15.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address 34:29:8f:78:10:ac, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 15.15.1, 7.1.11, 7.1.12, 7.1.13, KNX manufacturer ID 0x0, data 0x626d323031312020
14:38:57.915 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] INFO  i.c.s.knx-server - [run,85] - accepted Usb-15.15.0 tcp connection tcp:192.168.10.233:1374
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-15.15.0 received tcp:192.168.10.233:1374 06 10 02 05 00 1a 08 02 00 00 00 00 00 00 08 02 00 00 00 00 00 00 04 04 02 00
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.knx-server - [createResponseAddress,170] - responses to client use route-back control endpoint tcp:192.168.10.233:1374
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] INFO  i.c.s.knx-server - [acceptControlService,425] - Usb-15.15.0: setup data endpoint (channel 11) for connection request from tcp:192.168.10.233:1374
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.knx-server - [createResponseAddress,161] - responses use route-back data endpoint tcp:192.168.10.233:1374
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] INFO  i.c.s.knx-server - [initNewConnection,901] - assign additional individual address 15.15.1 to channel 11
14:38:57.916 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.knx-server - [<init>,66] - created socket udp:192.168.10.233:60188
14:38:57.917 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.g.knx-server - [connectionEstablished,228] - established connection KNX IP LinkLayer tcp:192.168.10.233:1374 (channel 11, 15.15.1)
14:38:57.917 [Usb-15.15.0 data endpoint tcp:192.168.10.233:1374] INFO  i.c.s.knx-server - [run,97] - Usb-15.15.0 data endpoint KNX IP LinkLayer tcp:192.168.10.233:1374 (channel 11, 15.15.1) is up and running
14:38:57.926 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] WARN  i.c.s.g.knx-server - [connectionEstablished,238] - skip verifying knx address of 'Usb-15.15.0' subnet interface (unknown mask version 00FA)
14:38:57.926 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-15.15.0 received tcp:192.168.10.233:1374 06 10 04 24 00 0d 04 0b 00 00 08 00 01
14:38:57.926 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [responseForFeature,509] - received tunneling-feat.set enable feature info service 01
14:38:57.927 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [respondToFeature,499] - respond with tunneling-feat.res enable feature info service 01
14:38:57.931 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-15.15.0 received tcp:192.168.10.233:1374 06 10 04 22 00 0c 04 0b 01 00 06 00
14:38:57.931 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [responseForFeature,509] - received tunneling-feat.get individual address 
14:38:57.931 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [respondToFeature,499] - respond with tunneling-feat.res individual address ff01
14:38:57.937 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] TRACE i.c.s.knx-server - [handleServiceType,315] - Usb-15.15.0 received tcp:192.168.10.233:1374 06 10 04 22 00 0c 04 0b 02 00 07 00
14:38:57.937 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [responseForFeature,509] - received tunneling-feat.get max apdu length 
14:38:57.937 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1374] DEBUG i.c.s.k.K.1.1.233:1374 - [respondToFeature,499] - respond with tunneling-feat.res max apdu length 00fe
[... cut ...]
14:39:23.136 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1377] DEBUG i.c.s.g.knx-server - [connectionClosed,150] - removed connection KNX IP LinkLayer tcp:192.168.10.233:1377 (client request)
14:39:23.136 [Usb-15.15.0 data endpoint tcp:192.168.10.233:1377] INFO  i.c.s.knx-server - [cleanup,130] - Usb-15.15.0 data endpoint tcp:192.168.10.233:1377 closed
14:39:23.142 [Usb-15.15.0 tcp connection tcp:192.168.10.233:1377] INFO  i.c.s.knx-server - [close,166] - close connection to tcp:192.168.10.233:1377
```

bert1020, 2024-12-12:
我用另外一个机器,通过tp连上 可以读取设备信息了 ,但是完整下载不行
ETS提示
应用程序无法下载至该设备(必要属性，如PortADDR,的值出错)。

下面是点击完整下载产生的日志
```
17:34:54.304 [knx-server control endpoint /dev/ttyUSB0] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received udp:192.168.10.233:64651 06 10 02 03 00 0e 08 01 00 00 00 00 00 00
17:34:54.304 [knx-server control endpoint /dev/ttyUSB0] DEBUG i.c.s.knx-server - [log,274] - responses use route-back control endpoint udp:192.168.10.233:64651
17:34:54.306 [knx-server control endpoint /dev/ttyUSB0] INFO  i.c.s.knx-server - [log,274] - send KNXnet/IP description to udp:192.168.10.233:64651: "Calimero KNX IP Server", KNX address 15.15.0, KNX medium TP1, Installation 0 - Project 0 (ID 0), KNX IP multicast address 224.0.23.12, MAC address 00:e0:4c:68:00:06, S/N 0000:00000000, Core (v2), Device Management (v2), Tunneling (v2), Routing (v2), 15.15.10, 15.15.11, 15.15.12, 15.15.13, KNX manufacturer ID 0x0, data 0x626d323031312020
17:34:54.311 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] INFO  i.c.s.knx-server - [log,274] - accepted /dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793
17:34:54.311 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 02 05 00 1a 08 02 00 00 00 00 00 00 08 02 00 00 00 00 00 00 04 04 02 00
17:34:54.312 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - responses to client use route-back control endpoint tcp:192.168.10.233:24793
17:34:54.312 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] INFO  i.c.s.knx-server - [log,274] - /dev/ttyUSB0: setup data endpoint (channel 13) for connection request from tcp:192.168.10.233:24793
17:34:54.312 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - responses use route-back data endpoint tcp:192.168.10.233:24793
17:34:54.313 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] INFO  i.c.s.knx-server - [log,274] - assign additional individual address 15.15.10 to channel 13
17:34:54.313 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,230] - created socket udp:192.168.10.200:38056
17:34:54.314 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.g.knx-server - [log,274] - established connection KNX IP LinkLayer tcp:192.168.10.233:24793 (channel 13, 15.15.10)
17:34:54.314 [/dev/ttyUSB0 data endpoint tcp:192.168.10.233:24793] INFO  i.c.s.knx-server - [log,230] - /dev/ttyUSB0 data endpoint KNX IP LinkLayer tcp:192.168.10.233:24793 (channel 13, 15.15.10) is up and running
17:34:54.316 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 04 24 00 0d 04 0d 00 00 08 00 01
17:34:54.317 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - received tunneling-feat.set enable feature info service 01
17:34:54.317 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - respond with tunneling-feat.res enable feature info service 01
17:34:54.322 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 04 22 00 0c 04 0d 01 00 06 00
17:34:54.323 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - received tunneling-feat.get individual address 
17:34:54.323 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - respond with tunneling-feat.res individual address ff0a
17:34:54.328 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 04 22 00 0c 04 0d 02 00 07 00
17:34:54.329 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - received tunneling-feat.get max apdu length 
17:34:54.329 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - respond with tunneling-feat.res max apdu length 000f
17:34:54.334 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 04 22 00 0c 04 0d 03 00 02 00
17:34:54.335 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - received tunneling-feat.get device descriptor type 0 
17:34:54.335 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,274] - respond with tunneling-feat.res device descriptor type 0 091a
17:34:54.348 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.knx-server - [log,274] - /dev/ttyUSB0 received tcp:192.168.10.233:24793 06 10 02 09 00 10 0d 00 08 02 00 00 00 00 00 00
17:34:54.349 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.k.K.1.1.233:24793 - [log,257] - close connection for channel 13 - client request
17:34:54.349 [/dev/ttyUSB0 tcp connection tcp:192.168.10.233:24793] DEBUG i.c.s.g.knx-server - [log,274] - removed connection KNX IP LinkLayer tcp:192.168.10.233:24793 (client request)
17:34:54.349 [/dev/ttyUSB0 data endpoint tcp:192.168.10.233:24793] INFO  i.c.s.knx-server - [log,257] - /dev/ttyUSB0 data endpoint tcp:192.168.10.233:24793 closed
[... cut ...]
17:35:05.745 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] DEBUG i.c.s.k.K.0.0.0.0.2.12) - [log,274] - add to multicast loopback frame buffer: 15.15.234->0/0/4 L_Data.ind, low priority hop count 5, tpdu 00 81
17:35:05.745 [KNX IP routing service (0.0.0.0 224.0.23.12) sender] DEBUG i.c.s.k.K.0.0.0.0.2.12) - [log,274] - sending cEMI frame seq 0, NonBlocking, attempt 1 (channel 0) 06 10 05 30 00 11 29 00 bc d0 ff ea 00 04 01 00 81
17:35:05.745 [knx-server routing service 224.0.23.12] DEBUG i.c.s.k.K.0.0.0.0.2.12) - [log,274] - discard multicast loopback cEMI frame: 15.15.234->0/0/4 L_Data.ind, low priority hop count 5, tpdu 00 81

```

bert1020, 2024-12-12:
通过查询您的代码看到了 PortADDR 是设备的一个属性值 property=17
下面是我用其他相加的ip router查询的,data确实为空
但是我用这个IP路由可以下载成功,是不是代表这个属性不是必须的?
我在哪里可以忽略这个吗?让他可以继续下载
```
<Telegram direction="Incoming" source_address="15.15.234" destination_address="15.12.247" payload="<PropertyValueResponse object_index="0" property_id="17" count="0" start_index="1" data="" />" />
```

bert1020, 2024-12-12:
还是说这个属性可以有个默认值,如果为空我们可以使用某个默认值

bert1020, 2024-12-13:
@bmalinowsky 可以帮我看下吗

closed 2024-12-13 by bert1020