calimero-project/calimero-server #31: Calimero Knx/Net Ip Server Installation on Orange Pi Cm4

Muhammed83, 2025-02-10:
Hello, 
I want to install "Calimero Knx/Net Ip Server" on my Orange Pi Compute Module 4 + Siemens Tp-uart2+ based ic card.
The Siemens Tp-uart2+ based ic card is connected to "/dev/ttyS2" port of Orange Pi Cm4.
I can see Knx datas when i connect it to with minicom.
I tried to create a new script for this installation with ai support.
Please could you check this script and help me for this installation.

[setup_knx_ip_gateway.zip](https://github.com/user-attachments/files/18731256/setup_knx_ip_gateway.zip)

bmalinowsky, 2025-02-10:
What exactly is the problem you would like to solve?

(Btw., there exists an installation script for the server here: https://github.com/Race666/calimero-server)

Muhammed83, 2025-02-10:
Hello bmalinowsky,

Nice to meet you :)
I am trying to install calimero server on Orange pi Cm4.
It is now giving;

JSerialCommSerialProvider loaded via static block.
Initializing SerialPortProvider...
Manually set SerialPortProvider: tuwien.auto.calimero.serial.JSerialCommSerialProvider
Starting KNXNet/IP Server...
Exception in thread "main" tuwien.auto.calimero.KNXException: no serial provider available to open /dev/ttyS2
        at tuwien.auto.calimero.serial.LibraryAdapter.open(LibraryAdapter.java:130)
        at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:186)
        at KnxNetIpServer.main(KnxNetIpServer.java:37)

bmalinowsky, 2025-02-10:
First of all, the script will install and run _calimero-tools_ (i.e., command line tools), not the server. 

Second, regarding your specific problem: if you want to create your own service provider for a serial port (which you do in your script), you need to create a provider-configuration file. This file identifies the provider and is included under _META-INF/services_ in the resulting `.jar` file which contains your provider classes.

Third, whoever or whatever created that script, just created a severe mess.

Muhammed83, 2025-02-10:
Hello Bmalinowsky

Yes i think so, this script is just made by ai of github. So it can be not good as yours.
How can i send you direct message or e-mail ?

bmalinowsky, 2025-02-12:
Did you try the other script I linked?

Otherwise, if you build the server, it actually creates a script for running it. The basic steps are as follows:

```
# Make sure you have a JDK, or install the required toolchain JDK
sudo apt install openjdk-17-jdk-headless

# Install git
sudo apt install git

# Get the server repo
git clone https://github.com/calimero-project/calimero-server.git
cd calimero-server

# Build the server
./gradlew build -x test
tar -xvf build/distributions/calimero-server-3.0-SNAPSHOT.tar

# Run the server script => you have to adjust server-config.xml for your needs!
./calimero-server-3.0-SNAPSHOT/bin/calimero-server resources/server-config.xml 
```

Muhammed83, 2025-02-12:
Hello Bmalinowsky,

Sure, it says at beginning;

root@orangepicm4:~# ./calimero-cm4.sh
Configure support for TPUART
No supported Hardware detected

Muhammed83, 2025-02-12:
[calimero_cm4.zip](https://github.com/user-attachments/files/18770339/calimero_cm4.zip)

Hello Bmalinowsky,

Finally i have used attached script, i modified its hardware detection part.
It run and given following error at end.

"cp: cannot stat './target/nar/serial-native-*-arm-Linux-gpp-jni/lib/arm-Linux-gpp/jni/libserialcom.so': No such file or directory"

bmalinowsky, 2025-02-12:
Please note that what I referred to are two different scripts (the first one was written by @Race666).

The first error (which you fixed yourself) is probably because you use a new Orange PI which is not detected yet. Can you run on the command line `dmesg` and post the output here? I can forward it to the author of the script.

For the current error (it's on line 431 in your script):
Can you find the directory where the project is building (`~/calimero-build`) and walk down the directories to the source of the copy? Like, _serial-native_, _target_, etc. ... to find what's actually in there (_./target/nar/serial-native-*-arm-Linux-gpp-jni/lib/arm-Linux-gpp/jni/libserialcom.so_).

Muhammed83, 2025-02-13:
Hello Bmalinowsky,

Yes i have modified part the hardware detection script of Race666.
Here is "dmesg" command output of my orangepi cm 4

[dmesg_orangepi_cm4.txt](https://github.com/user-attachments/files/18779287/dmesg_orangepi_cm4.txt)

Finally i could run the calimero server and it appears in Ets 5.
But there is a problem, i can't establish the connection over Calimero Knx/Net Ip server.
Please check pictures

[image]

[image]

root@orangepicm4:~# sudo systemctl daemon-reload
root@orangepicm4:~# sudo systemctl restart knx.service
root@orangepicm4:~# sudo systemctl status knx.service
● knx.service - Calimero KNX Daemon
     Loaded: loaded (/etc/systemd/system/knx.service; enabled; preset: enabled)
     Active: active (running) since Thu 2025-02-13 06:33:15 UTC; 1s ago
   Main PID: 2105 (java)
      Tasks: 19 (limit: 4527)
     Memory: 31.3M
        CPU: 3.261s
     CGroup: /system.slice/knx.service
             └─2105 /usr/bin/java --add-reads io.calimero.core=io.calimero.server --add-reads io.calimero.usb.provider.>
Feb 13 06:33:15 orangepicm4 systemd[1]: Started knx.service - Calimero KNX Daemon.

root@orangepicm4:~# sudo journalctl -u knx.service -n 50 --no-pager
Feb 13 06:29:55 orangepicm4 java[3682]: 06:29:55:568 WARN io.calimero.link.connector - connection attempt 2,832: timeout waiting for initial TP-UART state
Feb 13 06:30:06 orangepicm4 java[3682]: 06:30:06:588 WARN io.calimero.link.connector - connection attempt 2,833: timeout waiting for initial TP-UART state
Feb 13 06:30:17 orangepicm4 java[3682]: 06:30:17:608 WARN io.calimero.link.connector - connection attempt 2,834: timeout waiting for initial TP-UART state
Feb 13 06:30:19 orangepicm4 systemd[1]: Stopping knx.service - Calimero KNX Daemon...
Feb 13 06:30:19 orangepicm4 java[3682]: 06:30:19:963 INFO io.calimero.server.gateway.knx-server - knx-server: user request for shutdown
Feb 13 06:30:19 orangepicm4 java[3682]: 06:30:19:983 INFO io.calimero.server.knx-server - knx-server discovery endpoint closed
Feb 13 06:30:19 orangepicm4 java[3682]: 06:30:19:993 INFO io.calimero.server.knx-server - close connection to tcp:192.168.88.183:56197
Feb 13 06:30:19 orangepicm4 java[3682]: 06:30:19:994 INFO io.calimero.server.knx-server - close connection to tcp:192.168.88.183:60343
Feb 13 06:30:20 orangepicm4 java[3682]: 06:30:20:000 INFO io.calimero.server.knx-server - knx-server routing service 224.0.23.12 closed
Feb 13 06:30:20 orangepicm4 java[3682]: 06:30:20:001 INFO io.calimero.server.knx-server - knx-server control endpoint /dev/ttyS2 closed
Feb 13 06:30:20 orangepicm4 systemd[1]: knx.service: Deactivated successfully.
Feb 13 06:30:20 orangepicm4 systemd[1]: Stopped knx.service - Calimero KNX Daemon.
Feb 13 06:30:20 orangepicm4 systemd[1]: knx.service: Consumed 7min 7.480s CPU time.
-- Boot c196c60f4f9e4b3386131b9d5b7a564b --
Feb 13 06:30:52 orangepicm4 systemd[1]: Started knx.service - Calimero KNX Daemon.
Feb 13 06:31:00 orangepicm4 java[1338]: Exception in thread "main" io.calimero.KNXIllegalArgumentException: no network interface found with the specified name 'end1'
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher$XmlConfiguration.getNetIf(Launcher.java:587)
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher$XmlConfiguration.readServiceContainer(Launcher.java:245)
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher$XmlConfiguration.load(Launcher.java:216)
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher$XmlConfiguration.from(Launcher.java:177)
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher.<init>(Launcher.java:654)
Feb 13 06:31:00 orangepicm4 java[1338]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.Launcher.main(Launcher.java:640)
Feb 13 06:31:00 orangepicm4 systemd[1]: knx.service: Main process exited, code=exited, status=1/FAILURE
Feb 13 06:31:00 orangepicm4 systemd[1]: knx.service: Failed with result 'exit-code'.
Feb 13 06:31:00 orangepicm4 systemd[1]: knx.service: Consumed 3.300s CPU time.
Feb 13 06:33:15 orangepicm4 systemd[1]: Started knx.service - Calimero KNX Daemon.
Feb 13 06:33:17 orangepicm4 java[2105]: 06:33:17:556 INFO io.calimero.server.knx-server - Calimero KNX IP server 3.0-SNAPSHOT (64a9571)
Feb 13 06:33:18 orangepicm4 java[2105]: 06:33:18:503 INFO io.calimero.server.knx-server - use configuration 'Calimero KNXnet/IP Server' with 1 service container, discovery: listen on [all] send on [all] interfaces
Feb 13 06:33:18 orangepicm4 java[2105]: service container /dev/ttyS2:
Feb 13 06:33:18 orangepicm4 java[2105]:         server: listen on end1 (port 3671), KNX IP routing multicast group 224.0.23.12
Feb 13 06:33:18 orangepicm4 java[2105]:         Tpuart /dev/ttyS2 connection: TP1 medium, device 15.15.0
Feb 13 06:33:19 orangepicm4 java[2105]: 06:33:19:041 INFO io.calimero.server.knx-server - discovery endpoint [end1, lo] is up and running
Feb 13 06:33:19 orangepicm4 java[2105]: 06:33:19:044 INFO io.calimero.knxnetip.KNX IP Routing 224.0.23.12 - multicast loopback mode enabled
Feb 13 06:33:19 orangepicm4 java[2105]: 06:33:19:065 INFO io.calimero.server.knx-server - /dev/ttyS2 KNX IP Routing end1/224.0.23.12 is up and running
[... cut ...]
Feb 13 06:31:00 orangepicm4 systemd[1]: knx.service: Failed with result 'exit-code'.
Feb 13 06:31:00 orangepicm4 systemd[1]: knx.service: Consumed 3.300s CPU time.
Feb 13 06:33:15 orangepicm4 systemd[1]: Started knx.service - Calimero KNX Daemon.
Feb 13 06:33:17 orangepicm4 java[2105]: 06:33:17:556 INFO io.calimero.server.knx-server - Calimero KNX IP server 3.0-SNAPSHOT (64a9571)
Feb 13 06:33:18 orangepicm4 java[2105]: 06:33:18:503 INFO io.calimero.server.knx-server - use configuratroot@orangepicm4:~#: command not found

Muhammed83, 2025-02-13:
Update, i set all individual addresses at same segment, now it gives following errors

[image]

[image]

bmalinowsky, 2025-02-13:
Thank you for the `dmesg` output, I will take a look what changed.

Regarding the individual addresses: in the script the default addresses are _1.1.0_ and you have 8 client addresses starting with _1.1.129_. The server also complains that _15.15.5_ is not a tunneling address. So unless you changed those addresses in the config file, you have to use those.

Muhammed83, 2025-02-13:
Hello

[server-config.txt](https://github.com/user-attachments/files/18785360/server-config.txt)

Here is my "server-config.xml" file content.
Please check is there any error on it.
I think calimero can't establish connection with Siemens Tp-uart+ chip, but i can see Knx frames when connect to ttyS2 with minicom.
How can i solve it ?

Feb 13 14:39:52 orangepicm4 java[4926]: 14:39:52:416 WARN io.calimero.link.connector - connection attempt 2: timeout waiting for initial TP-UART state
Feb 13 14:40:03 orangepicm4 java[4926]: 14:40:03:448 WARN io.calimero.link.connector - connection attempt 3: timeout waiting for initial TP-UART state
Feb 13 14:40:14 orangepicm4 java[4926]: 14:40:14:493 WARN io.calimero.link.connector - connection attempt 4: timeout waiting for initial TP-UART state
Feb 13 14:40:15 orangepicm4 java[4926]: 14:40:15:384 INFO io.calimero.server.knx-server - close connection to tcp:192.168.88.183:64609
Feb 13 14:40:15 orangepicm4 java[4926]: 14:40:15:391 INFO io.calimero.server.knx-server - accepted /dev/ttyS2 connection tcp:192.168.88.183:64614
Feb 13 14:40:15 orangepicm4 java[4926]: 14:40:15:397 INFO io.calimero.server.knx-server - /dev/ttyS2: setup data endpoint (channel 11) for connection request from tcp:192.168.88.183:64614
Feb 13 14:40:15 orangepicm4 java[4926]: 14:40:15:402 WARN io.calimero.server.knx-server - /dev/ttyS2: tunneling on busmonitor-layer currently not allowed (active connections for tunneling on link-layer)
Feb 13 14:40:15 orangepicm4 java[4926]:         currently connected: [KNX IP LinkLayer tcp:192.168.88.183:64613 (channel 10, 15.15.129)]
Feb 13 14:40:15 orangepicm4 java[4926]: 14:40:15:405 WARN io.calimero.server.knx-server - no data endpoint for remote endpoint tcp:192.168.88.183:64614, could not accept new connection (maximum reached)
Feb 13 14:40:25 orangepicm4 java[4926]: 14:40:25:523 WARN io.calimero.link.connector - connection attempt 5: timeout waiting for initial TP-UART state
Feb 13 14:40:36 orangepicm4 java[4926]: 14:40:36:596 WARN io.calimero.link.connector - connection attempt 6: timeout waiting for initial TP-UART state

bmalinowsky, 2025-02-14:
It doesn't receive the initial state. So I'm not sure if it can even communicate.
The install script sets an UART overlay for "Orange", which you changed to "OrangeCM4". On line 655 in your script. Maybe it's that? That block would not have been executed.

Muhammed83, 2025-02-14:
Hello bmalinowsky

I have two type Siemens Tp-Uart based board, one of them is Siemens Tp-uart2 and other is Tp-uart2+.
I was testing it with Tp-uart2+ based board and it has not worked.
Today i tested the Tp-uart2, not "+" one and it worked, i surprised, because these chips are same, difference is just speed of "+" one.
It has "115200" baud rate only. Other things are same.
How can we run Tp-uart2+ board with calimero ?
Is there any possibility ?

bmalinowsky, 2025-02-14:
Calimero sets the Baud rate only to 19.200 Bd, but I think the tp-uart 2+ chip should also support that baud rate.

Muhammed83, 2025-02-15:
Tried it several times but i didn't work.
Is there any possibility to set calimero to support tp-uart2+ ?

Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.MatchOps$1MatchSink.accept(MatchOps.java:90)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.Spliterators$ArraySpliterator.tryAdvance(Spliterators.java:1002)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:129)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:527)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:513)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:230)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:196)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.util.stream.ReferencePipeline.allMatch(ReferencePipeline.java:637)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.acceptConnection(ControlEndpointService.java:1201)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.initNewConnection(ControlEndpointService.java:971)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.acceptControlService(ControlEndpointService.java:428)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.ControlEndpointService.handleServiceType(ControlEndpointService.java:350)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.StreamLooper.onReceive(StreamLooper.java:198)
Şub 15 13:49:02 orangepicm4 java[2147]:         at io.calimero.server@3.0-SNAPSHOT/io.calimero.server.knxnetip.StreamLooper.run(StreamLooper.java:98)
Şub 15 13:49:02 orangepicm4 java[2147]:         at java.base/java.lang.Thread.run(Thread.java:840)
Şub 15 13:49:02 orangepicm4 java[2147]: 13:49:02:486 INFO io.calimero.server.knx-server - /dev/ttyS2 data endpoint KNX IP LinkLayer tcp:192.168.88.183:64960 (channel 8, 0.0.224) is up and running
Şub 15 13:49:02 orangepicm4 java[2147]: 13:49:02:540 INFO io.calimero.server.knx-server - /dev/ttyS2 data endpoint tcp:192.168.88.183:64960 closed
Şub 15 13:49:02 orangepicm4 java[2147]: 13:49:02:544 INFO io.calimero.server.knx-server - close connection to tcp:192.168.88.183:64960
Şub 15 13:49:13 orangepicm4 java[2147]: 13:49:13:505 WARN io.calimero.link.connector - connection attempt 2: timeout waiting for initial TP-UART state
Şub 15 13:49:24 orangepicm4 java[2147]: 13:49:24:539 WARN io.calimero.link.connector - connection attempt 3: timeout waiting for initial TP-UART state

bmalinowsky, 2025-02-16:
> Is there any possibility to set calimero to support tp-uart2+ ?

I don't know right now, because the chips should be compatible and I can't say what's the difference that causes it.

Can you run the software with trace log enabled when using the tp-uart2+ chip?
For that, either in _/etc/systemd/system/knx.service_, at line `ExecStart=`, insert `-vvv` between `io.calimero.server` and `--no-stdin`, i.e., it should read `... io.calimero.server -vvv --no-stdin ...`.

Or run something like `knxtools -vvv groupmon --tpuart /dev/ttyS2`.

Muhammed83, 2025-02-17:
Hello,

My "knx.service" file seems as following.

[image]

There is no "--no-stdin" word there.
"ExecStart=/usr/bin/java --add-reads io.calimero.core=io.calimero.server --add-reads io.calimero.usb.provider.javax=ALL-UNNAMED --limit-modules io.calimero.server,io.calimero.serial.provide>"

bmalinowsky, 2025-02-17:
It's a long line, move your cursor to the right.

Muhammed83, 2025-02-17:
The complete one;

[Unit]
Description=Calimero KNX Daemon
After=network.target

[Service]
WorkingDirectory=/home/knx
# Wait for all interfaces, systemd-networkd-wait-online.service must be enabled
#ExecStartPre=/lib/systemd/systemd-networkd-wait-online --timeout=60
# Wait for a specific interface
#ExecStartPre=/lib/systemd/systemd-networkd-wait-online --timeout=60 --interface=end1
ExecStart=/usr/bin/java --add-reads io.calimero.core=io.calimero.server --add-reads io.calimero.usb.provider.javax=ALL-UNNAMED --limit-modules io.calimero.server,io.calimero.serial.provide>Type=simple
User=knx
Group=knx
#TimeoutStartSec=60
SuccessExitStatus=0 143

[Install]
WantedBy=multi-user.target network-online.target

bmalinowsky, 2025-02-17:
Please provide the output using 
`cat /etc/systemd/system/knx.service`

Muhammed83, 2025-02-17:
Here is,
I think it was changed.

[Unit]
Description=Calimero KNX Daemon
After=network-online.target
Wants=network-online.target

[Service]
WorkingDirectory=/home/knx
EnvironmentFile=/etc/default/knx
# Wait for network interfaces to be fully available
ExecStartPre=/bin/sleep 30
ExecStart=/usr/bin/java $DEF_SERVER_OPTS io.calimero.server.Launcher --no-stdin /etc/calimero/server-config.xml
User=knx
Group=knx
Restart=always
RestartSec=30
SuccessExitStatus=0 143

[Install]
WantedBy=multi-user.target network-online.target

Muhammed83, 2025-02-18:
In here should i put the "--vvv" ?

"io.calimero.server.Launcher --vvv --no-stdin /etc/calimero/server-config.xml"

bmalinowsky, 2025-02-18:
Yes.

Also note that in the output you posted, `$DEF_SERVER_OPTS` is an unexpanded variable. That shouldn't be.

Muhammed83, 2025-02-18:
Ok.

> Yes.
> 
> Also note that in the output you posted, `$DEF_SERVER_OPTS` is an unexpanded variable. That shouldn't be.

Yes this variable suggested by "artifical intelligence".because of knx doesn't start at start up.
I think i removed something by mistake in content of knx.service and so it wasn't work.
And then "ai" suggested this variable, i surprised it works well with this variable.

Normally which parameters should include in knx.service file ? 
I will try to work knx.service without this variable.
Could you send me a example knx.service file ?

Thanks

bmalinowsky, 2025-02-18:
```
[Unit]
Description=Calimero KNX Daemon
After=network.target

[Service]
WorkingDirectory=/home/knx
# Wait for all interfaces, systemd-networkd-wait-online.service must be enabled
#ExecStartPre=/lib/systemd/systemd-networkd-wait-online --timeout=60
# Wait for a specific interface
#ExecStartPre=/lib/systemd/systemd-networkd-wait-online --timeout=60 --interface=eth0
ExecStart=/usr/bin/java --add-reads io.calimero.core=io.calimero.server --add-reads io.calimero.usb.provider.javax=ALL-UNNAMED --limit-modules io.calimero.server,io.calimero.serial.provider.jni,io.calimero.usb.provider.javax -p /etc/calimero:/opt/calimero-server -cp "/etc/calimero:/opt/calimero-server/*" --module io.calimero.server -vvv --no-stdin /etc/calimero/server-config.xml
Type=simple
User=knx
Group=knx
#TimeoutStartSec=60
SuccessExitStatus=0 143

[Install]
WantedBy=multi-user.target network-online.target
```

Muhammed83, 2025-02-19:
Hello Bmalinowsky,

Here is the log of my knx.service with **"-vvv"** in current situation (Knx.service file with $DEF_SERVER_OPTS variable)
I will try also run knx.service file without this variable option, with above content which you sent.
Please check this log now, why tpuart2+ doesn't work.

[knx-service-log-with-tpuart2+.txt](https://github.com/user-attachments/files/18864525/knx-service-log-with-tpuart2%2B.txt)

Thanks

bmalinowsky, 2025-02-21:
Can you try the attached library?

1. In _/opt/calimero-server/_, delete _calimero-core-3.0-SNAPSHOT.jar_
2. Download the attached file, and remove the `.txt` extension from its name; the resulting name should be _calimero-core-3.0-SNAPSHOT.jar_
3. Put the new `.jar` file into _/opt/calimero-server/_
4. Rerun the server
5. Post the trace log

[calimero-core-3.0-SNAPSHOT.jar.txt](https://github.com/user-attachments/files/18908315/calimero-core-3.0-SNAPSHOT.jar.txt)

Muhammed83, 2025-02-23:
Hello Bmalinowsky,

I did what you said, put this file to "/opt/calimero-server/" and renamed it as ."jar" again.
I also used the "knx.service" file which you sent before and it is also working without variable.
Attached the log, please check.

[knx-service-log-with-tpuart2+23022025.txt](https://github.com/user-attachments/files/18932441/knx-service-log-with-tpuart2%2B23022025.txt)

bmalinowsky, 2025-02-26:
The tpuart connection doesn't really receive a lot from the tpuart 2+ chip. Sending seems to works.
I attached another modified library, where I unconditionally set the speed to 115200 Bd, which is supported by the 2+ chips.
Can you please try as before?

[calimero-core-3.0-SNAPSHOT.jar.txt](https://github.com/user-attachments/files/18989736/calimero-core-3.0-SNAPSHOT.jar.txt)

Muhammed83, 2025-02-26:
Hello,

I did it, but service is giving error, i think there is a error in .jar.
Please check the log.

[knx-service-log-with-tpuart2+26022025.txt](https://github.com/user-attachments/files/18991385/knx-service-log-with-tpuart2%2B26022025.txt)

bmalinowsky, 2025-02-26:
OK, it probably didn't upload correctly. I attached it again.

[calimero-core-3.0-SNAPSHOT.jar.txt](https://github.com/user-attachments/files/18993363/calimero-core-3.0-SNAPSHOT.jar.txt)

Muhammed83, 2025-02-26:
Hello 

It seems working with this new file.
But there is some error i think.
For instance when i give turn on command over ets5, commands seem to be going out twice in group monitor.
Please check new log attached.

[knx-service-log-with-tpuart2+26022025-1.txt](https://github.com/user-attachments/files/18994213/knx-service-log-with-tpuart2%2B26022025-1.txt)

bmalinowsky, 2025-02-27:
Can you check if the reason for the duplicated entries is whether the group monitor just shows the second frame because it is generated for KNXnet/IP routing?
The calimero server has IP routing enabled by default, so every frame also shows up in the IP network.

You can change it in _/etc/calimero/server-config.xml_. Within the element `serviceContainer` there is an attribute `routing="true"`. You can set it to `routing="false"`, and restart the server.

Muhammed83, 2025-02-27:
Hello,

I did which you said.
But it is still same.

[knx-service-log-with-tpuart2+27022025.txt](https://github.com/user-attachments/files/19016191/knx-service-log-with-tpuart2%2B27022025.txt)

Muhammed83, 2025-03-03:
Hello Bmalinowsky

Is there any progress about this issue ?

Thanks

bmalinowsky, 2025-03-05:
Hello,

from the trace I cannot detect duplicated frames. You have 2 tcp connections open to the same host, so each connection gets served for each event, but that's ok.

bmalinowsky, 2025-03-05:
Yes, attach a longer ETS log. Maybe I can observe something.

Muhammed83, 2025-03-06:
Ok. Please check the attached logs.
0.0.220 source was address of Homeassistant.

[Ets_log_06032025.zip](https://github.com/user-attachments/files/19103155/Ets_log_06032025.zip)

bmalinowsky, 2025-03-14:
After looking at the logs, I don't think the frames are repeated on the bus itself. What you see in the group monitor is the initial command sent to the bus (hop count 6), and afterwards the indication on the bus which is fed back to the ETS client connection (hop count 4).

Muhammed83, 2025-03-16:
Hello
Ok. Understood you.
It already works well, seems no problem.
I installed fresh copy of calimero, but forgot to test it with original "calimero core 3.0 snapshot", replaced it with your modified one.
Could you send me non modified "calimero core 3.0 snapshot" ?

bmalinowsky, 2025-03-20:
Can you clarify which version you want again? If you use the install script, it will install the newest version, which should be fine.

Two important thing I forgot to mention: 
* the install script got updated to fix the error you had with "Orange Pi"
* there is now a property _io.calimero.serial.tpuart.uartBaudRate_, which allows you to set the UART speed. In your case, the property would be `-Dio.calimero.serial.tpuart.uartBaudRate=115200`. For example, in the install script you can add that property like this (search for the line):
    - Before: `ExecStart=/usr/bin/java $DEF_SERVER_OPTS --module io.calimero.server --no-stdin $CALIMERO_CONFIG_PATH/server-config.xml`
    - After: `ExecStart=/usr/bin/java -Dio.calimero.serial.tpuart.uartBaudRate=115200 $DEF_SERVER_OPTS --module io.calimero.server --no-stdin $CALIMERO_CONFIG_PATH/server-config.xml`

Muhammed83, 2025-03-23:
Hello Bmalinowsky

I have used my old modified by ai script.
I don't know which version is installed, how can i learn which version is installed?

It is good to modify script, i already know it.
But i think you should add also aarch64 cpu architecture.

bmalinowsky, 2025-03-24:
When you start the server, it shows a log entry like this:

`INFO io.calimero.server.knx-server - Calimero KNX IP server 3.0-SNAPSHOT (d37bca7)`

First part is the general version (3.0-SNAPSHOT), the part in brackets is the commit hash of the commit the library was compiled from (the commits you see on github).

I will see what's up with the aarch64 port.

Muhammed83, 2025-03-24:
The calimero already starts automaticaly on system start.
How can i do start it manually and see this log ?

Please check also following logs of knx.service, i think there is a warn about hop which you mentioned above before.

● knx.service - Calimero KNX Daemon
     Loaded: loaded (/etc/systemd/system/knx.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-03-18 09:59:00 UTC; 6 days ago
    Process: 1063 ExecStartPre=/bin/sleep 30 (code=exited, status=0/SUCCESS)
   Main PID: 1541 (java)
      Tasks: 34 (limit: 4527)
     Memory: 139.5M
     CGroup: /system.slice/knx.service
             └─1541 /usr/bin/java --add-reads io.calimero.core=io.calimero.server --add-reads io.calimero.usb.provider.javax=ALL-UNNAMED --limit-modules io.calimero.server,io.calimero.serial.provider.jni,io>

Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:002 INFO io.calimero.server.knx-server - assign additional individual address 0.0.129 to channel 1
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:007 INFO io.calimero.server.knx-server - /dev/ttyS2 data endpoint KNX IP LinkLayer tcp:192.168.88.182:45804 (channel 1, 0.0.129) is up and running
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:046 INFO io.calimero.server.knx-server - send KNXnet/IP description to tcp:192.168.88.182:45804: "Calimero KNXnet/IP Server", KNX address 0.0.0, KNX mediu>
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:080 INFO io.calimero.server.knx-server - accepted /dev/ttyS2 connection tcp:192.168.88.187:39690
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:091 INFO io.calimero.server.knx-server - /dev/ttyS2: setup data endpoint (channel 2) for connection request from tcp:192.168.88.187:39690
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:094 INFO io.calimero.server.knx-server - assign additional individual address 0.0.130 to channel 2
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:098 INFO io.calimero.server.knx-server - /dev/ttyS2 data endpoint KNX IP LinkLayer tcp:192.168.88.187:39690 (channel 2, 0.0.130) is up and running
Mar 24 06:00:57 homeassistant1 java[1541]: 06:00:57:156 INFO io.calimero.server.knx-server - send KNXnet/IP description to tcp:192.168.88.187:39690: "Calimero KNXnet/IP Server", KNX address 0.0.0, KNX mediu>
Mar 24 08:02:46 homeassistant1 java[1541]: 08:02:46:179 WARN io.calimero.server.gateway.knx-server - hop count 0, discard frame 15.15.255->0.0.6
Mar 24 08:02:46 homeassistant1 java[1541]: 08:02:46:276 WARN io.calimero.server.gateway.knx-server - 

[knx_service_log_24032025.txt](https://github.com/user-attachments/files/19428187/knx_service_log_24032025.txt)

 0, discard frame 15.15.255->0.0.6

bmalinowsky, 2025-03-24:
You can use this line:

`journalctl -u knx -o short-precise | grep "Calimero KNX IP"`

> Please check also following logs of knx.service, i think there is a warn about hop which you mentioned above before.

Yes, it means there is a loop somewhere, usually with more than one IP router. So a message ends up being received & sent again in a circle.

Muhammed83, 2025-03-24:
I used you command, but there is no any output when gave it.

root@homeassistant1:~# sudo journalctl -u knx -o short-precise | grep "Calimero KNX IP"
root@homeassistant1:~#

> Yes, it means there is a loop somewhere, usually with more than one IP router. So a message ends up being received & sent again in a circle.

Yes there is also other ip interface/router in same line, s the problem caused by this ?

bmalinowsky, 2025-03-24:
I looked at your server-config, you have a different server name. Try this instead:

`journalctl -u knx -o short-precise | grep "Calimero KNXnet/IP Server"`

> Yes there is also other ip interface/router in same line, s the problem caused by this ?

If they have both the same IP multicast address, probably yes.

Muhammed83, 2025-03-24:
root@homeassistant1:~# journalctl -u knx -o short-precise | grep "Calimero KNXnet/IP Server"
root@homeassistant1:~#

still same.

bmalinowsky, 2025-03-24:
Hm, what if you look only for "Calimero"? There should be plenty of lines.

`journalctl -u knx -o short-precise | grep "Calimero"`

Muhammed83, 2025-03-24:
It is strange, still no output.

root@homeassistant1:~# journalctl -u knx -o short-precise | grep "Calimero"
root@homeassistant1:~#

Muhammed83, 2025-03-24:
> If they have both the same IP multicast address, probably yes.

I deactivated other Ip routers, but still there is repeat in telegrams.

bmalinowsky, 2025-03-24:
You can also get the version otherwise, that's maybe easier:

`unzip /opt/calimero-server/calimero-server-3.0-SNAPSHOT.jar -d /tmp`
`cat /tmp/META-INF/MANIFEST.MF `

bmalinowsky, 2025-03-24:
Ok, now I understand. The newest jar file for the server you built with the install script is initially stored here:

_/root/calimero-build/calimero-server/build/libs/calimero-server-3.0-SNAPSHOT.jar_

You can just copy it.

Muhammed83, 2025-03-24:
Sorry i think there is misunderstanding, i was asking the original "calimero-core-3.0-SNAPSHOT.jar"
I had replaced the original "calimero-core-3.0-SNAPSHOT.jar" with your modified one which you sent me here.

> OK, it probably didn't upload correctly. I attached it again.
> 
> [calimero-core-3.0-SNAPSHOT.jar.txt](https://github.com/user-attachments/files/18993363/calimero-core-3.0-SNAPSHOT.jar.txt)

So, i think the original " "calimero-core-3.0-SNAPSHOT.jar" must be here, "/calimero-build/calimero-core/build/libs/calimero-core-3.0-SNAPSHOT.jar" is that so ?

bmalinowsky, 2025-03-24:
Yes correct, my mistake.

Muhammed83, 2025-03-24:
Ok. I will copy it and test it.
Thanks

Muhammed83, 2025-03-25:
I have test the original "core-3.0-SNAPSHOT.jar"
But it is not working again, your modified one is working.
Is this "loop-repeating telegram" event is problem ?

bmalinowsky, 2025-03-25:
Did you change the baudrate to 115200?

There should be a log entry
_using io.calimero.serial.tpuart.uartBaudRate of 115,200 Bd/s_

> Is this "loop-repeating telegram" event is problem ?

Repeated telegrams put more load on the knx bus, so it eventually can become a problem.

Do you need routing with the Calimero server? If not, you can disable it in the _server-config.xml_
There is a line
`<serviceContainer activate="true" routing="true" ...`
Set routing to false like so
`<serviceContainer activate="true" routing="false" ...`
 
and restart the server. If the telegram loop disappears, you know there is another routing device. Otherwise, the loop has another cause. Anyway, it would be good to find the root cause.

Muhammed83, 2025-03-25:
> Did you change the baudrate to 115200?
> 
> There should be a log entry _using io.calimero.serial.tpuart.uartBaudRate of 115,200 Bd/s_

No i used it as it is.
How can i change the speed in the original "calimero-core-3.0-SNAPSHOT.jar"

[21 more replies]